Koenkk/zigbee2mqtt #3006: Reached 20 devices. Cant add more even with a lot of routers. How to procced

YamiFrankc, 2020-02-26:
Hello, 
I am trying to add a couple more light bulbs to my network, however, I noticed that I already have 20 devices and they dont pair. Is there a way to make these  bulbs(routers) connect to other routers instead of the coordinator?
I tried pairing them far away from the coordinator but the result is the same.

Version of Zigbee2Mqtt: 1.11.0
Coordinator version: 20190425
Coordinator device: CC2531

Koenkk, 2020-02-26:
Reflashing the CC2531 should free up the device table (doesn't need to require repairing of existing devices).

YamiFrankc, 2020-02-27:
Thanks, that worked!
I also flashed it with a newer version... a new problem that appeared it that some of the devices(light bulbs) seem to take longer to respond, especially when triggering them with a zigbee remote(which makes me think its more of a problem with the remote).
But good news is i can add the 8 bulbs i bought!

closed 2020-05-06 by stale[bot]