Koenkk/zigbee2mqtt #3007: dev branch: error with "npm ci" on armv6l

mahowi, 2020-02-26:
# Bug Report
## What happened
I tried to switch to dev branch following the instructions at https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html
`npm ci` fails with
`npm WARN prepare removing existing node_modules/ before installation`
`npm ERR! code EAI_AGAIN`
`npm ERR! errno EAI_AGAIN`
`npm ERR! request to https://registry.npmjs.org/run-async/-/run-async-2.4.0.tgz failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443`

`npm ERR! A complete log of this run can be found in:`
`npm ERR!     /opt/zigbee2mqtt/.npm/_logs/2020-02-26T20_19_58_853Z-debug.log`

## What did you expect to happen
Switching to dev branch.

## How to reproduce it (minimal and precise)
Follow the instructions at https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

## Debug info
`zigbee@pihole:~$ cat /sys/firmware/devicetree/base/model;echo`
`Raspberry Pi Model B Plus Rev 1.2`
`zigbee@pihole:~$ uname -a`
`Linux pihole 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux`

`zigbee@pihole:~$ npm --version`
`6.7.0`
`zigbee@pihole:~$ node --version`
`v11.15.0`
This is the latest version available for armv6l.

`npm ci` works on master branch.

Koenkk, 2020-02-27:
`getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443` indicates an internet connection issue, can you verify this?

mahowi, 2020-02-27:
I can download the file on this Pi via `wget` without any problem. So it shouldn't be a connection problem. And on master branch all packages get loaded.
I can remember there has been a similar problem on armv6l a while ago.

Koenkk, 2020-02-27:
Does `npm install` work?

mahowi, 2020-02-28:
> Does `npm install` work?

Yes.

`info  2020-02-28 08:33:26: Starting zigbee2mqtt version 1.11.0-dev (commit #342d4c1)`

Koenkk, 2020-02-28:
Does `wget https://registry.npmjs.org/run-async/-/run-async-2.4.0.tgz` work on the Pi?

Can you also try with the latest dev branch?

mahowi, 2020-02-28:
Yes, downloading the file via what works. I'll try the latest dev branch tomorrow.

mahowi, 2020-02-29:
Sorry, I'm away from home this weekend and my network seems to be down. So I can't update zigbee2mqtt via remote access.

mahowi, 2020-03-02:
I have access again to my network.
With the latest dev branch `npm ci` installs all packages again. :+1:

closed 2020-03-02 by Koenkk