Koenkk/zigbee2mqtt #3008: Sensors stopped working and wouldn't repair

mecko, 2020-02-26:
# Bug Report
## What happened
Everything worked fine up until a couple of days ago when some Xiaomi sensors stopped responding. I tried to repair one of them and there was absolutely nothing in the logs. I took one that was still working, tried to repair it and now it can't pair either.

I tried to run Z2M with DEBUG=* npm start, and the only thing that _might_ be relevant is this:
```
serialport/stream _read reading { start: 603, toRead: 64933 } +31ms
serialport/binding-abstract read +11s
serialport/bindings/unixRead Starting read +32ms
serialport/bindings/unixRead read error { [Error: EAGAIN: resource temporarily unavailable, read] errno: -11, code: 'EAGAIN', syscall: 'read' } +1ms
serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
serialport/bindings/poller Polling for "readable" +34ms
```

I followed the FAQ and tried to reset the CC2530 using the button that's closest to the USB port. That didn't help either. I also tried to move the CC2530 to a different USB port through a hub. LED works on the sensors when I click the button.

When nothing else worked, I upgraded Z2M to the latest version (as of a couple of minutes ago) and tried all of the above all over again.

## What did you expect to happen
The sensors shouldn't have stopped working unexpectedly. The sensors should have repaired when I tried to.

## How to reproduce it (minimal and precise)
Be as unlucky as I am.

## Debug Info
zigbee2mqtt version: 31e56785e46fe745dea0aa248cf27e2529938d04
CC253X firmware version: ```Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}}'```

mecko, 2020-02-27:
After a while and for an unexplained reason, some sensors started working again as if nothing has ever happened. Some still don't.

Koenkk, 2020-02-27:
Your firmware is outdated, did you already try updating to the latest? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

Koenkk, 2020-02-28:
Hard to say but always a good first attempt to fix issues as the firmware stability has increased.

When running with the DEBUG=* and clicking the button do you absolutely see nothing in the log?

mecko, 2020-02-28:
Just the serialport thing as mentioned in the original issue post. Other than that, no.

Koenkk, 2020-02-28:
Do you have any chance to sniff the network? For now this seems to be the only way to gain more insight in why this happens. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

mecko, 2020-02-29:
That requires the hardware to flash the CC2531 with different firmware, and I don't have that nearby. Unless there's anything else to check or test, this will have to way a few days.

mecko, 2020-02-29:
@Koenkk 
After playing with it some more, I realized that the CC2531 itself is having issues as it started to flash the red LED and mixed green and red and Z2M started complaining about timeouts (these symptoms are new).

I moved it to yet another USB port (USB3 this time) and had to repair all of the sensors but they're working now. None of this makes any sense though, because I tried 2 different ports (USB2 fwiw), and even tried to plug it through a USB hub, and neither worked properly.

I'd close this issue for now with a request for better debug outputs in the future. Also, I'd suggest that the disable_led option will be turned off automatically when there's an error/issue, or at least mention that in the docs.

closed 2020-02-29 by mecko