Koenkk/zigbee2mqtt #3009: [feature] want ability to filter out values from mqtt payload/attrib publishing

sjorge, 2020-02-26:
It would be nice to specify a per device list of properties to remove form the json payload published over mqtt (and the alternative that publishes under multiple topics which name escapes me)

I’ll take a look at this myself.

sjorge, 2020-02-29:
```patch
diff --git a/lib/controller.js b/lib/controller.js
index f0fe0b7..647be92 100644
--- a/lib/controller.js
+++ b/lib/controller.js
@@ -285,17 +285,17 @@ class Controller {
             attributes.forEach((a) => messagePayload.device[a] = device[a]);
         }

+        // filter mqtt message attributes
+        if (entity.settings.mqtt_attribute_filter) {
+            entity.settings.mqtt_attribute_filter.forEach((a) => delete messagePayload[a])
+        }
+
         if (Object.entries(messagePayload).length) {
-            if (settings.get().experimental.output === 'attribute_and_json') {
+            if ((settings.get().experimental.output === 'attribute_and_json') || (settings.get().experimental.output === 'json')) {
                 await this.mqtt.publish(entity.name, JSON.stringify(messagePayload), options);
+            }
+            if ((settings.get().experimental.output === 'attribute_and_json') || (settings.get().experimental.output === 'attribute')) {
                 await this.iteratePayloadAttributeOutput(`${entity.name}/`, messagePayload, options);
-            } else if (settings.get().experimental.output === 'json') {
-                await this.mqtt.publish(entity.name, JSON.stringify(messagePayload), options);
-            } else {
-                /* istanbul ignore else */
-                if (settings.get().experimental.output === 'attribute') {
-                    await this.iteratePayloadAttributeOutput(`${entity.name}/`, messagePayload, options);
-                }
             }
         }
     }
diff --git a/lib/util/settings.js b/lib/util/settings.js
index abd4dca..3d60c9f 100644
--- a/lib/util/settings.js
+++ b/lib/util/settings.js
@@ -225,6 +225,7 @@ const schema = {
                         friendly_name: {type: 'string'},
                         retain: {type: 'boolean'},
                         qos: {type: 'number'},
+                        mqtt_attribute_filter: {type: 'array', items: {type: 'string'}},
                     },
                     required: ['friendly_name'],
                 },
```

@Koenkk I have this but it is not working, but it looks fine to me?

sjorge, 2020-02-29:
Edit it actually works fine! For both devices an groups. I'm having issues with running the tests now though: #3031

Even with just a clean checkout of dev

Koenkk, 2020-03-01:
Can you make a PR? Would be interesting to see if tests also fail on azure pipelines

sjorge, 2020-03-01:
Done, I created PR #3037 for this

sjorge, 2020-03-03:
> #3037 So filter out both from device_options.filtered_attributes and devices.ID.filtered_attributes ?

I'll re-open this and do a follow up, because it does indeed seem to be ignore. Probably next weekend though.

closed 2020-03-03 by sjorge