Koenkk/zigbee2mqtt #3010: OTA Update check

khaeljy, 2020-02-27:
# Bug Report
## What happened
Device updates his firmware.

## What did you expect to happen
Only check if update is available.

## How to reproduce it (minimal and precise)
Send DEVICE_FRIENDLY_NAME payload to the MQTT topic zigbee2mqtt/bridge/ota_update/check

## Debug Info
zigbee2mqtt version: 1.11.0
CC2531 firmware version: CC2531_SOURCE_ROUTING_20190619

nldroid, 2020-02-27:
This is not really helpfull.

What device are you trying to update? Do you get errors? Are you sure there should be an update? If yes, why?

khaeljy, 2020-02-27:
All my devices (Ikea and Osram).

When I send a message to the topic `zigbee2mqtt/bridge/ota_update/check`, I see in logs `Update of 'DEVICE_NAME' at 0%`. And device starts update of his firmware.

Koenkk, 2020-02-27:
Could you provide the debug logging? https://www.zigbee2mqtt.io/information/debug.html#zigbee2mqtt-debug-logging

khaeljy, 2020-02-27:
I have tried with a button (it's my latest device who as an update available), but I have an error during the update.
Yesterday, when I have made the same test on my bulbs, they are successfully updating.

As you can see receiving an MQTT message on `/zigbee2mqtt/bridge/ota_update/check` topic trigger an update to the latest firmware.
```
debug 2020-02-27 19:20:24: Received MQTT message on '/zigbee2mqtt/bridge/ota_update/check' with data 'button_4'
info  2020-02-27 19:20:24: Updating 'button_4' to latest firmware
info  2020-02-27 19:20:24: MQTT publish: topic '/zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Updating 'button_4' to latest firmware","meta":{"status":"update_in_progress","device":"button_4"}}'
debug 2020-02-27 19:20:25: Received Zigbee message from 'button_4', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190410","swBuildId":"2.2.008"}' from endpoint 1 with groupID 0
debug 2020-02-27 19:20:25: No converter available for 'E1743' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20190410","swBuildId":"2.2.008"}'
debug 2020-02-27 19:20:25: Updating to latest '0x14b457fffe4dbbcb' (TRADFRI on/off switch)
debug 2020-02-27 19:20:25: Using endpoint '1'
debug 2020-02-27 19:20:27: Received Zigbee message from 'button_4', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":570459697}' from endpoint 1 with groupID 0
debug 2020-02-27 19:20:27: No converter available for 'E1743' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":570459697}'
debug 2020-02-27 19:20:27: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":570459697}'
debug 2020-02-27 19:20:28: Got new image for '0x14b457fffe4dbbcb'
debug 2020-02-27 19:20:28: Starting upgrade
debug 2020-02-27 19:20:29: Received Zigbee message from 'button_4', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":570459697}' from endpoint 1 with groupID 0
debug 2020-02-27 19:20:29: No converter available for 'E1743' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":1,"manufacturerCode":4476,"imageType":4549,"fileVersion":570459697}'
debug 2020-02-27 19:20:38: OTA update at 0%, remaining Infinity seconds
info  2020-02-27 19:20:38: Update of 'button_4' at 0%
info  2020-02-27 19:20:38: MQTT publish: topic '/zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update of 'button_4' at 0%","meta":{"status":"update_progress","device":"button_4","progress":0}}'
debug 2020-02-27 19:20:38: Image block response failed (CommandResponse 0x14b457fffe4dbbcb/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":4549,"fileVersion":570492465,"fileOffset":0,"dataSize":63,"data":{"type":"Buffer","data":[30,241,238,11,0,1,56,0,0,0,124,17,197,17,49,6,1,34,2,0,69,66,76,32,116,114,97,100,102,114,105,95,111,110,111,102,102,95,115,104,111,114,116,99,117,116,95,99,111,110,116,114,126,190,2,0,0,0,64,190,2,0,0]}}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":50686,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":18,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,39,5,0,124,17,197,17,49,6,1,34,0,0,0,0,63,30,241,238,11,0,1,56,0,0,0,124,17,197,17,49,6,1,34,2,0,69,66,76,32,116,114,97,100,102,114,105,95,111,110,111,102,102,95,115,104,111,114,116,99,117,116,95,99,111,110,116,114,126,190,2,0,0,0,64,190,2,0,0]}}' failed with status '16' (expected '0')))
debug 2020-02-27 19:20:43: Image block response failed (CommandResponse 0x14b457fffe4dbbcb/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":4549,"fileVersion":570492465,"fileOffset":0,"dataSize":63,"data":{"type":"Buffer","data":[30,241,238,11,0,1,56,0,0,0,124,17,197,17,49,6,1,34,2,0,69,66,76,32,116,114,97,100,102,114,105,95,111,110,111,102,102,95,115,104,111,114,116,99,117,116,95,99,111,110,116,114,126,190,2,0,0,0,64,190,2,0,0]}}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":50686,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":20,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,40,5,0,124,17,197,17,49,6,1,34,0,0,0,0,63,30,241,238,11,0,1,56,0,0,0,124,17,197,17,49,6,1,34,2,0,69,66,76,32,116,114,97,100,102,114,105,95,111,110,111,102,102,95,115,104,111,114,116,99,117,116,95,99,111,110,116,114,126,190,2,0,0,0,64,190,2,0,0]}}' failed with status '16' (expected '0')))
debug 2020-02-27 19:20:50: Image block response failed (CommandResponse 0x14b457fffe4dbbcb/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":4549,"fileVersion":570492465,"fileOffset":0,"dataSize":63,"data":{"type":"Buffer","data":[30,241,238,11,0,1,56,0,0,0,124,17,197,17,49,6,1,34,2,0,69,66,76,32,116,114,97,100,102,114,105,95,111,110,111,102,102,95,115,104,111,114,116,99,117,116,95,99,111,110,116,114,126,190,2,0,0,0,64,190,2,0,0]}}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":50686,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":28,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,41,5,0,1 [... cut ...] ,114,105,95,111,110,111,102,102,95,115,104,111,114,116,99,117,116,95,99,111,110,116,114,126,190,2,0,0,0,64,190,2,0,0]}}' failed with status '16' (expected '0')))
debug 2020-02-27 19:21:29: Got upgrade end request for '0x14b457fffe4dbbcb': {"status":149,"manufacturerCode":4476,"imageType":4549,"fileVersion":570492465}
debug 2020-02-27 19:21:29: Update failed with reason: 'aborted by device'
error 2020-02-27 19:21:29: Update of 'button_4' failed (Update failed with reason: 'aborted by device')
```

Koenkk, 2020-02-27:
Are you sure this is a clean install of Zigbee2mqtt 1.11.0? I think the problem is caused by the `/` at the beginning, `zigbee2mqtt/bridge/ota_update/check` should work. However I cannot produce this locally (no Zigbee2mqtt logging is shown at all when I send this).

jeromelaban, 2020-03-06:
Update - I got the device to accept the update by bringing it closer to the coordinator (that's the only thing I changed).

carldebilly, 2020-03-06:
@jeromelaban Your network is probaly too weak at some places...

nldroid, 2020-03-06:
This bugreport is for the bug that the OTA started while only doing a check if there was an update.

Bugs for the OTA process itself go here:

https://github.com/Koenkk/zigbee2mqtt/issues/2921

Koenkk, 2020-03-06:
I will close this as the issue of the OP has been fixed in the dev branch

closed 2020-03-06 by Koenkk