Koenkk/zigbee2mqtt #3011: Once I get more than 20 devices, some of them just loose connection

mr2c12, 2020-02-27:
I had a quite stable setup:
- coordinatore cc2530+ccc2591, fw 1.2 NOT source routing
- one cc2530 router
- mixed osram, ikea and philips bulbs
- one innr smart plug
- one xiaomi smart plug
- a bunch of battery powered end devices
- z2m 1.9 running on a raspberry

It was pretty stable when I used to have 15-18 devices.
Once I added a bunch of new sensors (xiaomi temperature sensors, motion sensor but also a cc2531 router and an ikea repeater) and got more than 20 devices, a lot of problems appeared. It seems that some devices often loose connection.

So yesterday I even re-flashed my coordinator with 1.2 WITH source routing, but nothing really changed.

Attached a picture of my network:

[image]

I read that my coordinator should be enough for ~30-40 devices.
Can I fix my situation?
Should I switch to a more powerful coordinator (like CC26X2R1)? Does it really fix it for sure?

Last thing: when z2m starts, the log says:
 Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'

It seems to me that it is the same output log that I used to get with fw without source routing. I am sure I flashed to correct w/sr fw, how can I be realy sure?

sv87411, 2020-02-28:
Hi @mr2c12 

I can't guarantee it will help, but I also had lots of issues with devices dropping off my network with a CC2531 as coordinator and a lot of (about 5) router devices like Innr plugs and Innr lights. I had just 4 Xiaomi sensors at the time. 

I was getting very frustrated and noticed I had a lot of sub 100 link quality connections and a lot below 50. It seems you do too. I bit the bullet and ordered a CC26X2R1, flashed it with the Z 3.0 stack and it's made a world of difference. The majority of my links now are 100+ with some 200+. 

I'm migrating from SmartThings and I've now grown my Z2M network to 16 router devices (Innr, Ikea, SmartThings) and 11 sensors. Mostly Xiaomi temp and motion, but also Samsung buttons and contact sensors.

My network has been rock solid since I introduced the CC26X2R1 and it has sat on my desk with no need to move it and using the on PCB antenna.

It was certainly worth it for me to move to the CC26X2R1!

Stuart

mr2c12, 2020-02-28:
Thanks @sv87411,
your feedback is really useful. I am seriously considering upgrading to CC26X2R1.

At the moment I would like to understand whether the "20 devices" limit is some kind of limit on the cc2530+cc2591. It's driving me crazy. It was solid stable until I got to that number of devices.
Now it's a pain :)

sv87411, 2020-02-28:
I'm afraid my only experience was with the CC2531 USB dongle. As I understand it a CC2531 running as coordinator it can only have 15 directly connected devices and each of those can be endpoint or router. Then if you run a second CC2531 as router that can have 21 devices and so on. But I found the CC2531 to be very under powered and got poor link quality even with an external antenna mod. 

Reading the data sheet for the CC2530 that you are using, those chips were designed for low powered zigbee node devices; extract from datasheet "_The CC2530 is a true system-on-chip (SoC) solution for IEEE 802.15.4, Zigbee and RF4CE applications. It enables robust network **nodes** to be built with very low total bill-of-material costs_". My emphasis on **nodes**. It only has 8KB of RAM and I suspect that, along with a low powered CPU, is the limiting factor regarding how many devices can be handled in a stable environment when running it as a coordinator. Even though the chip is paired with the CC2591 for increased RF range, I think the CC2530 is the problem. 

@Koenkk may know more about the details of how the various stacks use the SoC resources and how they limit the number of connected devices.

The CC2652 SoC that's on the LaunchXL CC26X2R1 board is a far more powerful SoC with 80K RAM and a much more powerful core CPU and RF section. I know I sound like I do, but I don't work for Texas Instruments :smiley: . But now I have the CC26X2R1 in place I am not spending hours of my time removing and re-pairing devices and wondering why things aren't working. They just now work.

My time is valuable to me and I found I was spending a lot of time trying to get the cheaper devices working that could be better spent on setting up my smart home. The CC26X2R1 has allowed me this time and I now have 28 router devices and 22 endpoints in my network with the CC26X2R1 as coordinator and it is very stable with no drop outs. I still have about 6 devices to migrate over, but I have no concerns about this process now and know it will work. I now spend my time configuring Home Assistant and setting up automations with Node Red which I find far more interesting than banging my head against a wall and constantly pressing small buttons on Xiaomi motion sensors :laughing: 

If you can afford to go down the CC26x2 route I would seriously consider it because you will save yourself a lot of time and headache.

Stuart

mr2c12, 2020-02-29:
Dear @sv87411 ,
your time is valuable and you spent a lot of effort on this issue.
Thank you very much.
Last question for you: it will probably be cheaper to me the CC1352P-2 respect the CC26X2R1 (the former is 51€ no shipping fees, the latter is 40€ + 20€ shipping fees). From the documentation they are the same. I won't use an external antenna. Can I go for the CC1352P-2?

Koenkk, 2020-02-29:
@mr2c12 just go for the cheapest one, it's unlikely that you will see any difference between both (especially if you don't use an external antenna).

mr2c12, 2020-02-29:
Thanks. I will consider one of them (funny enough, the most expensive board will cost less given the free shipping fees).
After two days of hard work, I isolated the problem. Will now continue on a more specific issue on that:
https://github.com/Koenkk/zigbee2mqtt/issues/3030

closed 2020-02-29 by mr2c12

sv87411, 2020-02-29:
> Dear @sv87411 ,
> your time is valuable and you spent a lot of effort on this issue.
> Thank you very much.

Happy to help. I'd much rather spend my time here trying to improve the whole Z2M experience than pairing devices.

For me in the UK, I ordered the CC26x2R1 through [Mouser](https://mouser.co.uk) and including sales tax it cost me £48. The CC1352P on Mouser is £60 including sales tax. My CC26x2R1 was shipped directly from Texas USA, delivery was very fast (4 days) via FedEx and free. All import duty was covered too. All in all a very good experience. Mouser are international so might be worth checking them out if you haven't already.

Re Koen's comment about an external antenna mine has been running fine with the on-board PCB antenna so I won't be going external any time soon. Also mine is situated in my study which is at the front corner of my house so it's not even central and it has a direct connection to a Xioami temperature sensor that's in my garden at the back of my house! It's impressive.

Good luck!

mr2c12, 2020-03-07:
Dear @sv87411 ,
thanks again for your help.
I finally got my new CC26x2R1, repaired all my device and.. I will wait for any problem *not* to occur!

Just a note: my hope was to get a stronger signal especially downstair. Actually, I got very similar strenght reading (according to network map). It seems that my previous coordinator (cc2530+cc2591) had a similar power signal to the new CC26x2R1.
Not a big problem (I just had one bulb with serious connection/signal problems). As per this issue, my goal is to have a coordinator that is able to handle >20 devices!

Thank you!