Koenkk/zigbee2mqtt #3012: Support for Develco Smoke Alarm SMSZB-120

BlizzWave, 2020-02-27:
The original issue has gone stale, but is still not resolved. 
#1809 

Help with getting this smoke alarm to work with zigbee2mqtt.

Link to log
[https://pastebin.com/rsHx5uzA](url)

Koenkk, 2020-02-27:
- First make sure you are running zigbee2mqtt 1.11.0 (contains interview improvements)
- Let's then focus on getting the smoke detection to work. Could you provide the logging when triggering the smoke alarm?

BlizzWave, 2020-03-02:
I have made some progress here. 
Smoke detection works and battery related stuff works. 

I still don't get a successful interview though.
And i don't know how to trigger the siren with command. 

```
{
"smoke":false,"
battery_low":false,
"linkquality":97,
"battery":"100.00",
"voltage":3500,
"battery_alarm_state":0,
"temperature":25.12,
"supervision_reports":true,
"restore_reports":true,
"test":false
}
```

Added this to my devices.js file

```

    {
       zigbeeModel: ['SMSZB-120'],
       model: 'SMSZB-120',
       vendor: 'Develco Products A/S',
       description: 'Smoke detector with siren',
       supports: 'smoke, warning, temperature',
       fromZigbee: [fz.temperature, fz.bitron_battery_att_report, fz.develco_smoke, fz.ignore_basic_report, fz.ignore_genOta],
       toZigbee: [tz.warning],
       meta: {configureKey: 1},
       configure: async (device, coordinatorEndpoint) => {
	   const endpoint = device.getEndpoint(35);
	   await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasZone', 'genBasic']);
	   await configureReporting.batteryVoltage(endpoint);
           const endpoint2 = device.getEndpoint(38);
           await bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
  	   await configureReporting.temperature(endpoint2);
       },
    },

```
And added this to my fromZigbee.js file

```
    develco_smoke: {
        cluster: 'ssIasZone',
        type: ['attributeReport', 'readResponse', 'commandStatusChangeNotification'],
        convert: (model, msg, publish, options, meta) => {
            const zoneState = msg.data.zoneState;
            const zoneType = msg.data.zoneType;
            const zoneStatus = msg.data.zonestatus;
            const iasCieAddr = msg.data.iasCieAddr;
            const zoneId = msg.data.zoneId;
        
        return {
            sensor_type: zoneType,
            enrolled: zoneState,
            smoke: (zoneStatus & 1) > 0, // Bit 0 = Alarm: Smoke
            IASCIEAddress: iasCieAddr,
            zoneID: zoneId,
            battery_low: (zoneStatus & 1 << 3) > 0, // Bit 3 = Battery LOW indicator
	    supervision_reports: (zoneStatus & 1 << 4) > 0,
	    restore_reports: (zoneStatus & 1 << 5) > 0,
            test: (zoneStatus & 1 << 8) > 0,

            };    
        },
    },

```

Koenkk, 2020-03-02:
- How did you got the device working now?
- Instead of `develco_smoke` I would recommend to use the already existing converter `ias_smoke_alarm_1`
- For the siren try to add the following `toZigbee` converter: `tz.warning` and now send to `zigbee2mqtt/NEO Sirene/set` with payload `{"warning": {"mode": "emergency"}}`

BlizzWave, 2020-03-02:
> * How did you got the device working now?

I thought that a device needed a successful interview to work. Therefor I thought that I could not troubleshoot.

> * Instead of `develco_smoke` I would recommend to use the already existing converter `ias_smoke_alarm_1`

They are not the same. 
The Develco Smoke alarm does not have tamper, trouble or ac_status.
Ref. Tech Manual

> * For the siren try to add the following `toZigbee` converter: `tz.warning` and now send to `zigbee2mqtt/NEO Sirene/set` with payload `{"warning": {"mode": "emergency"}}`

Nothing happens. 

```
zigbee2mqtt:debug 2020-03-02 19:31:21: Received MQTT message on 'zigbee2mqtt2/0x0015bc0031004688/set' with data '{"warning": {"mode": "emergency"}}'
zigbee2mqtt:debug 2020-03-02 19:31:21: Publishing 'set' 'warning' to '0x0015bc0031004688'
zigbee2mqtt:error 2020-03-02 19:31:36: Publish 'set' 'warning' to '0x0015bc0031004688' failed: 'Error: Command 0x0015bc0031004688/1 ssIasWd.startWarning({"startwarninginfo":53,"warningduration":10}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 51981 - 1 - 2 - 1282 - 11 after 6000ms)'
zigbee2mqtt:debug 2020-03-02 19:31:36: Error: Command 0x0015bc0031004688/1 ssIasWd.startWarning({"startwarninginfo":53,"warningduration":10}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 51981 - 1 - 2 - 1282 - 11 after 6000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-03-02 19:31:36: MQTT publish: topic 'zigbee2mqtt2/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'warning' to '0x0015bc0031004688' failed: 'Error: Command 0x0015bc0031004688/1 ssIasWd.startWarning({\"startwarninginfo\":53,\"warningduration\":10}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 51981 - 1 - 2 - 1282 - 11 after 6000ms)'","meta":{"friendly_name":"0x0015bc0031004688"}}'
zigbee2mqtt:debug 2020-03-02 19:34:25: Received Zigbee message from '0x0015bc0031004688', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":48,"extendedstatus":0}' from endpoint 35 with groupID 0
```

Koenkk, 2020-03-03:
Can you provide the `database.db` entry of this device? And are you sure it's possible to manually activate the alarm?

BlizzWave, 2020-03-03:
Here you go:

`{"id":2,"type":"EndDevice","ieeeAddr":"0x0015bc0031004688","nwkAddr":51981,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Battery","modelId":"SMSZB-120","epList":[1,35,38],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[]},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,15,32,1280,1282],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"SMSZB-120","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":1,"dateCode":"2018-11-26 06:22"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00193e5093","zoneState":0}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryAlarmState":0}}},"binds":[]},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2287}}},"binds":[]}},"dateCode":"2018-11-26 06:22","zclVersion":1,"interviewCompleted":false,"meta":{"configured":1},"lastSeen":1583231344532}`

According to the technical manual you should be able to trigger the alarm. 
Currently I'm testing if they trigger each other when there is smoke.  ( I have multiple Develco smoke alarms)

Koenkk, 2020-03-04:
The `tz.warning` uses the IAS WD cluster. Maybe the value is not correct, please try with different combinations:

So to `zigbee2mqtt/NEO Sirene/set` with payload `{"warning": {"mode": "MODE", "level": "LEVEL"}}`

- Where`MODE` can be: stop, burglar, fire, emergency, police_panic, fire_panic, emergency_panic
- And `LEVEL` can be: low, medium, high, very_high

BlizzWave, 2020-03-04:
I tried every combination you can think of now, still no go.
On thing though, I cannot get the enroll to work, do you think that hos something to do with it?

Koenkk, 2020-03-05:
@BlizzWave I'm not sure if it has something to do with it. Are you sure triggering the alarm actually works via Deconz?

Koenkk, 2020-03-06:
Can you try with `zigbee2mqtt/NEO Sirene/set` with payload `{"warning": {"mode": "burglar", "level": "high", "strobe": false, "duration": 300}}`

BlizzWave, 2020-03-06:
Still nothing

```
Mar 06 08:01:32 raspberrypi npm[9343]: zigbee2mqtt:debug 2020-03-06 08:01:32: Received MQTT message on 'zigbee2mqtt2/0x0015bc0031004688/set' with data '{"warning": {"mode": "burglar", "level": "high", "strobe": false, "duration": 300}}'
Mar 06 08:01:32 raspberrypi npm[9343]: zigbee2mqtt:debug 2020-03-06 08:01:32: Publishing 'set' 'warning' to '0x0015bc0031004688'
Mar 06 08:01:47 raspberrypi npm[9343]: zigbee2mqtt:error 2020-03-06 08:01:47: Publish 'set' 'warning' to '0x0015bc0031004688' failed: 'Error: Command 0x0015bc0031004688/1 ssIasWd.startWarning({"startwarninginfo":18,"warningduration":300}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 14844 - 1 - 25 - 1282 - 11 after 6000ms)'
Mar 06 08:01:47 raspberrypi npm[9343]: zigbee2mqtt:debug 2020-03-06 08:01:47: Error: Command 0x0015bc0031004688/1 ssIasWd.startWarning({"startwarninginfo":18,"warningduration":300}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 14844 - 1 - 25 - 1282 - 11 after 6000ms)
Mar 06 08:01:47 raspberrypi npm[9343]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
Mar 06 08:01:47 raspberrypi npm[9343]:     at Generator.throw (<anonymous>)
Mar 06 08:01:47 raspberrypi npm[9343]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Mar 06 08:01:47 raspberrypi npm[9343]: zigbee2mqtt:info  2020-03-06 08:01:47: MQTT publish: topic 'zigbee2mqtt2/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'warning' to '0x0015bc0031004688' failed: 'Error: Command 0x0015bc0031004688/1 ssIasWd.startWarning({\"startwarninginfo\":18,\"warningduration\":300}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 14844 - 1 - 25 - 1282 - 11 after 6000ms)'","meta":{"friendly_name":"0x0015bc0031004688"}}'
```

SwoopX, 2020-03-06:
Would it help if I provide you guys with a traffic sniff while triggering the alert?

BlizzWave, 2020-03-06:
> Would it help if I provide you guys with a traffic sniff while triggering the alert?

If you wouldn't mind that would be very helpful indeed!

SwoopX, 2020-03-06:
Can do that in the evening.

BlizzWave, 2020-03-06:
So i decided to make CC2531 sniffer because I was wondering what was going in with the enrollment not working. 
Some funny stuff here also. 
Error from Wireshark:

`ZCL IAS Zone: Zone Status Change Notification, Seq: 4[Malformed Packet]`

SwoopX, 2020-03-06:
Here you go

[image]

BlizzWave, 2020-03-06:
Thank you!

Here is how it looks when zigbee2mqtt is sending the command.
But these seem different
`Acknowlegment Request`
`Disable Default Resonse `

[image]

SwoopX, 2020-03-06:
Hm, if I recall correctly, the default response thingy could cause hickups... Also, I had real difficulties to get the device enrolled back in the days. Not sure if that has been sorted out by now.

BlizzWave, 2020-03-06:
I have just modified line 384 in endpoint.js so that I would get a "true" on the Disable Default Response.

`return Object.assign({ timeout: 6000, manufacturerCode: null, disableDefaultResponse: true }, providedOptions);`

This did not help.

Also the `Frame Control Field: Cluster -specific` is different.

I'm hoping @Koenkk will figure something out.

Koenkk, 2020-03-08:
@SwoopX sniffer log shows that the message is send to endpoint 35, while yours is send to 1. 

Try with:

```js
{
    zigbeeModel: ['SMSZB-120'],
    model: 'SMSZB-120',
    vendor: 'Develco Products A/S',
    description: 'Smoke detector with siren',
    supports: 'smoke, warning, temperature',
    fromZigbee: [fz.temperature, fz.bitron_battery_att_report, fz.develco_smoke, fz.ignore_basic_report, fz.ignore_genOta],
    toZigbee: [tz.warning],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(35);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasZone', 'genBasic']);
        await configureReporting.batteryVoltage(endpoint);
        const endpoint2 = device.getEndpoint(38);
        await bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
        await configureReporting.temperature(endpoint2);
    },
    endpoint: (device) => {
        return {'default': 35};
    },
    },
```

Also, please remove the `disableDefaultResponse` that you've added.

SwoopX, 2020-03-08:
@Koenkk good catch. Haven't thought about it since I don't need to worry with deconz. Of course it must go to 35, 1 is the manufacturer specific endpoint.

BlizzWave, 2020-03-08:
Well I must be blind or something. Did not see that.

**BUT**
It works now!

Thanks for the help @Koenkk and @SwoopX 

To get the siren to start
`{"warning": {"mode": "burglar", "level": "high", "strobe": false, "duration": 300}}`

To stop it
`{"warning": {"mode": "stop", "level": "low", "strobe": false, "duration": 300}}`

It gives a error when starting the siren, but it works.

```
Mar 08 19:32:44 raspberrypi npm[856]: zigbee2mqtt:error 2020-03-08 19:32:44: Publish 'set' 'warning' to '0x0015bc0031004688' failed: 'Error: Command 0x0015bc0031004688/35 ssIasWd.startWarning({"startwarninginfo":18,"warningduration":3}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: AREQ - AF - dataConfirm after 3000ms)'
Mar 08 19:32:44 raspberrypi npm[856]: zigbee2mqtt:debug 2020-03-08 19:32:44: Error: Command 0x0015bc0031004688/35 ssIasWd.startWarning({"startwarninginfo":18,"warningduration":3}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: AREQ - AF - dataConfirm after 3000ms)
Mar 08 19:32:44 raspberrypi npm[856]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
Mar 08 19:32:44 raspberrypi npm[856]:     at Generator.throw (<anonymous>)
Mar 08 19:32:44 raspberrypi npm[856]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Mar 08 19:32:44 raspberrypi npm[856]: zigbee2mqtt:info  2020-03-08 19:32:44: MQTT publish: topic 'zigbee2mqtt2/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'warning' to '0x0015bc0031004688' failed: 'Error: Command 0x0015bc0031004688/35 ssIasWd.startWarning({\"startwarninginfo\":18,\"warningduration\":3}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: AREQ - AF - dataConfirm after 3000ms)'","meta":{"friendly_name":"0x0015bc0031004688"}}'

```

Koenkk, 2020-03-08:
Can you modify the toZigbee warning converter to add a timeout like done here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L844 ? (check if it works with a timeout of `10000`).

SwoopX, 2020-03-08:
One additional thing though, as I'm pretty sure it does not work with deconz. Not sure if it has been taken into consideration here. It's about clearing the alarm state after the smoke alarm has been triggered.

[image]

BlizzWave, 2020-03-08:
> Can you modify the toZigbee warning converter to add a timeout like done here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L844 ? (check if it works with a timeout of `10000`).

Like this?

```
const info = (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level]);
const opts = {timeout: 10000}; //Test

await entity.command(
    'ssIasWd',
    'startWarning',
    {startwarninginfo: info, warningduration: values.duration, opts},
    getOptions(meta),
```

Koenkk, 2020-03-09:
Great, once you feel it's finished, could you make the PRs?

BlizzWave, 2020-03-10:
Yes, I will do that.
They way I modified the warning config effects all devices that use tz.warning. 
Should I create a separate config for the Develco ex. tz.warning_develco or 
should we have exception inside the current tz.warning?
The latter I don't know how to do. I'm not good at programming! 

Next thing is to look at is the enrollment issue. 
For a fire alarm to be really good it needs to be able to start other fire alarms without the need to go to the coordinator first. 
And if I understand the technical manual it should be able to do that.

@SwoopX Does enrollment work on Deconz?

Koenkk, 2020-03-10:
Just create the PRs and I will do the cleanups.

SwoopX, 2020-03-10:
@BlizzWave yes, it's working flawlessly. But as I said, I had trouble with zigbee2mqtt back in the days. Could that also be related to the endpoint? IAS stuff is endpoint 35...

Koenkk, 2020-03-10:
@BlizzWave to check please sniff the traffic when pairing the device with zigbee2mqtt.

BlizzWave, 2020-03-10:
Here's the log from Wireshark.

[develco_fire_alarm_sniffing.zip](https://github.com/Koenkk/zigbee2mqtt/files/4314773/develco_fire_alarm_sniffing.zip)

Koenkk, 2020-03-11:
I'm not sure this was just an incident, but the status change notification, where zigbee-herdsman is waiting on to confirm is has been enrolled correctly, is malformed:

[image]

Could you sniff the traffic one more and confirm that the package is malformed again? If yes, I guess that the device is enrolled just fine (as it also triggers on smoke).

BlizzWave, 2020-03-11:
Its not random. I get that every time.
The smoke alarm does not set the enroll to true either.
I have not tested with smoke to see if it triggers the other fire alarms yet though.

I quote myself from ealier:
> Some funny stuff here also.
> Error from Wireshark:
> 
> `ZCL IAS Zone: Zone Status Change Notification, Seq: 4[Malformed Packet]`

Koenkk, 2020-03-11:
But the alarm is triggered when there is smoke, right?

Sorry missed the malformed packet part, but in that case it's expected that interview fails, if the device works as expected this can be ignored though (in that case add it to the docs).

BlizzWave, 2020-03-11:
Yes, it triggers.  (Will do test with multiple alarms to check if they trigger each other later)
The interview fails sometimes and you can try again and it works. 
I can remember reading something about this in another post, that the smoke alarm needs a couple of tries to actually interview correctly.

Koenkk, 2020-03-11:
Good, as we cannot the interview fail issue (device issue due to the malformed packet), could you make a PR?

BlizzWave, 2020-03-11:
Well I'm wondering if this happens on Deconz also, if its the device then Deconz should have the same error

@SwoopX could I bother you once more to do sniff traffic again while the sensor is being interviewed?

SwoopX, 2020-03-11:
What do you guys need exactly? Enrollment is working well on deconz side

[image]

However, I also see malformed packets, but typically directly after a valid one. Afaik, the malformed has an increased counter by 2.

[image]

BlizzWave, 2020-03-11:
Could post the whole log like I did?

SwoopX, 2020-03-11:
[Develco_join_neu3.zip](https://github.com/Koenkk/zigbee2mqtt/files/4320553/Develco_join_neu3.zip)

BlizzWave, 2020-03-11:
@SwoopX That can't be correct. Contains no real Zigbee traffic

BlizzWave, 2020-03-11:
I also get the Enroll request and Enroll response. But it keeps on asking for enroll even though it has gotten a successful response. 
And i get a malformed packet after a while, same as @SwoopX .
I have also tested with smoke to see if they trigger each other. They dont **NOT**.

@SwoopX your enroll works, but if one alarm triggers from smoke does it start the other alarms?
I'm assuming you have multiple. haha

SwoopX, 2020-03-11:
> That can't be correct. Contains no real Zigbee traffic

It is, Dedicated test gateway with only the sensor in it.

> your enroll works, but if one alarm triggers from smoke does it start the other alarms?

Haven't tested that one, I currently don't dare it. Got 4 of them operational, two more lying around.

BlizzWave, 2020-03-12:
> > That can't be correct. Contains no real Zigbee traffic
> 
> It is, Dedicated test gateway with only the sensor in it.
Okay, I can't see it. If I sort by protocol I only get beacons from the Deconz. 
No joining or reporting of any sensor.

Might be asking alot, but the ones you have laying around, could you join them to your test network and see if they trigger each other from smoke?

Because if they dont then there is a issue with the firmware in the smoke alarm and I will send a email to Develco Support.

SwoopX, 2020-03-12:
I knew this request would come... :P

I'll see what I can do on the weekend. Regarding the firmware, I know there are, well, "bugs" or inconsistencies. Was also thinking of reaching out trying to get access to their support forum. Not sure if some observations are known and eventually resolved by any updates. Maybe we can team up then...

BlizzWave, 2020-03-12:
> I knew this request would come... :P
>
> I'll see what I can do on the weekend. Regarding the firmware, I know there are, well, "bugs" or inconsistencies. Was also thinking of reaching out trying to [...]

Haha! I had to ask. Thanks for your help so far!

Btw I have contacted Develco. Hopefully I get some response. I'll l post what I get here. 
If (really big if here) theres some kind of firmware update for this issue. Then we stuble on a new issue, how would we update the smoke alarm?

BlizzWave, 2020-03-18:
@SwoopX Did you get to test it?

A little update, I have been in contact with Develco about the enrollment issue. They are looking into it.
Btw, to get access to their support forum you need to pay a fee.

SwoopX, 2020-03-19:
> A little update, I have been in contact with Develco about the enrollment issue. They are looking into it.
> Btw, to get access to their support forum you need to pay a fee.

Man, I just wanna have the firmware updates and the "bugs" fixed they have...

Haven't had the time to check it unfortunately. The meltdown out there made life a bit more complicated...

BlizzWave, 2020-03-20:
Got a response from Develco today. 
@Koenkk what do you think? 

> Issue is that your Gateway is taking to long to answer the Enrollrequest.
> 
> Our sensors gives the TIMEOUT 0x94 (The exchange was aborted due to excessive response time.) after 15 seconds.
> 
>  
> 
> ------------------------------------
> Lasse Pedersen
> Senior Technical Adviser
> Phone: +45 30 54 30 46
> Develco Products A/S

Koenkk, 2020-03-20:
I investigated your sniff a bit more and think we can one more thing.

- First update to zigbee2mqtt 1.12.0
- Now replace `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js` with the following https://gist.github.com/Koenkk/6cc1d83b631775ff67c5da9039c9ca33
- Pair the device again (factory reset it) and provide both the sniff log and zigbee herdsman debug log

BlizzWave, 2020-03-20:
Followed your instructions. 
The sensor will now enroll, but only for a couple of seconds.
Also the timeout 0x94 is still there.

But have a look in the logs

Attached sniff and debug. 
[sniff_and_debug.zip](https://github.com/Koenkk/zigbee2mqtt/files/4362030/sniff_and_debug.zip)

SwoopX, 2020-03-20:
@BlizzWave check out https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2608#issuecomment-601749756 regarding the group alert.

Koenkk, 2020-03-21:
In addition to the controller.js from https://github.com/Koenkk/zigbee2mqtt/issues/3012#issuecomment-601871251 can you also update the `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js` with https://gist.github.com/Koenkk/6999e024118f454df7303a453c7017a8

Interview should succeed after this.

Btw: where do you see the 0x94 timeout error in that sniff?

BlizzWave, 2020-03-21:
> @BlizzWave check out [dresden-elektronik/deconz-rest-plugin#2608 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2608#issuecomment-601749756) regarding the group alert.
> 
Will have a look. Thanks

> Btw, have you sent your request to [info@develcoproducts.com](mailto:info@develcoproducts.com)? Only regarding the enrollment issue or also regarding the malformed packet?

Yes, I have. We have so far talked about the enrollment issue. But I will start asking more about the malformed packet now since this must be a firmware issue on their side.
 
> Would like to engage them for a node descriptor issue as well (and hopefully get a firmware update ;)).

@Koenkk 

I think you got the enroll to work now. Bit for enroll is set to 1.
I thought that triggering one of the sensors should start the other one when properly enrolled.
But the second sensor does not react. 
Will do some sniffing to see what actually happens. 

> In addition to the controller.js from [#3012 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/3012#issuecomment-601871251) can you also update the `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js` with https://gist.github.com/Koenkk/6999e024118f454df7303a453c7017a8
> 
> Interview should succeed after this.
> 
> Btw: where do you see the 0x94 timeout error in that sniff?

Look at no 157 in the Wireshark sniff.

BlizzWave, 2020-03-21:
Sensor being triggered by smoke.
Reports smoke but also the famous malformed packet comes in.
I just sent a email to Develco about this, but this must be firmware related.

[image]

Malformed Packet

[image]

Koenkk, 2020-03-22:
Good, these changes are available in the latest dev branch now so the device should pair fine without any changes. Could you make a PR to zigbee-herdsman-converters to add support for this device?

Regarding auto triggering more smoke alarms: I didn't see this yet within Zigbee, atm it makes sense that it doesn't work as the smoke sensor only sends messages to the coordinator.

BlizzWave, 2020-03-22:
So to get this to work is to make a "automation" in whatever home automation you might use and manually trigger the other sensors with the command from earlier in the thread.

> Regarding auto triggering more smoke alarms: I didn't see this yet within Zigbee, atm it makes sense that it doesn't work as the smoke sensor only sends messages to the coordinator.

BlizzWave, 2020-03-29:
Response from Develco stopped so I just did the PR's for this. If theres any response in the future I will post here and update whats needed

SwoopX, 2020-03-29:
Maybe I just kept them too busy with my request on the node descriptor 😜

closed 2020-06-06 by stale[bot]

BlizzWave, 2020-08-05:
So I have finally integrated the smoke alarms from Develco in my Zigbee2MQTT / Home Assistant system.

There one small annoyance though.
The sensor does report voltage, but it seems Zigbee2MQTT does not convert it to battery percentage when reporting to Home Assistant.

[24 more replies]