Koenkk/zigbee2mqtt #3013: Xiaomi ZNCZ02LM

kwetnico, 2020-02-27:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
Hello, I have an installation with zigbee2mqtt, a raspberry pi 4, and several xiaomi peripherals in majority. I bought a Xiaomi ZNCZ02LM socket which is integrated into my jeedom v3 live (without xiaomi wc gateway), it serves as a relay on my zigbee network. Under jeedom I installed it with the xiaomihome plugin but I cannot control it on or off. I go back a lot of information: state, consumption, signal etc ... But not a command or I can control it. Do you have a solution?

zigbee2mqtt version: 1.11.0 
CC2531 firmware version: 06/2019

Koenkk, 2020-02-28:
Please provide the log when trying to control it.

kwetnico, 2020-02-28:
Hello,

I do not have a log because I do not launch any command, the command to activate or deactivate set grayed out we just read the state.
I wish I could give it the ON or OFF order.
[image]
[image]

Koenkk, 2020-02-28:
I'm not familiar with Jeedom, but when sending via MQTT to `zigbee2mqtt/Xiaomi_Prise/set` with payload `TOGGLE` does the switch toggle? If yes, this would indicate it's not a Zigbee2mqtt problem.

kwetnico, 2020-03-03:
> Je ne connais pas Jeedom, mais lors de l'envoi via MQTT vers `zigbee2mqtt/Xiaomi_Prise/set`avec la charge utile, `TOGGLE`le commutateur bascule-t-il? Si oui, cela indiquerait que ce n'est pas un problème Zigbee2mqtt.

No, i find with this command it's ok : 
mosquitto_pub -t zigbee2mqtt/Xiaomi_Prise/set -m " {\"state\":\"OFF\"}"

Koenkk, 2020-03-03:
Good, that means zigbee2mqtt is working correctly. Please open an issue at Jeedom or their forums

closed 2020-03-03 by Koenkk