Koenkk/zigbee2mqtt #3015: Suggestion in regards to IKEA E1757 pairing

gimlichael, 2020-02-27:
On this page, https://www.zigbee2mqtt.io/devices/E1757.html, you suggest to use the sniffer as the source of the roller blind.

Unfortunately, this will not work (at least on my network), if the sniffer is more than 10 m. away from the rollers after the setup is done.

What i did on my setup was to pair the roller with the IKEA E1746 that is in the same room as the blinds.

After i did this, i could actually control my curtains and no longer received either the LQ error or no network route error.

Anyway, this is just my findings - consider if it is something you want to suggest to others.

Keep up the good work!

Koenkk, 2020-02-28:
> What i did on my setup was to pair the roller with the IKEA E1746 that is in the same room as the blinds.

Could you explain a bit more how you did this?

gimlichael, 2020-02-28:
> > What i did on my setup was to pair the roller with the IKEA E1746 that is in the same room as the blinds.
> 
> Could you explain a bit more how you did this?

The same procedure as written in the article I link to; just with the IKEA repeater instead of the sniffer. Hope that make sense :-)

Koenkk, 2020-02-29:
Just to confirm, the signal repeater is included when you buy these blinds, right?

gimlichael, 2020-02-29:
Yes, that is correct. And per default, the blinds are paired to the included repeater.

That said, the article clearly states, that you should pair it with the sniffer.
Problem is, that when you do that - the blinds only works when (for my instance) the RasberryPI (with the sniffer) is within range. The moment you move the sniffer to your "snif-room(TM)", then the blinds cannot longer be reached and you will encounter LQ and/or network route error.

Having tried that twice, I decided to op-in for the IKEA approach; pair the blinds with the repeater.
This solved the issue, and why I made this suggestion (as i think people have a tendency to follow your guides 1-1) :-)

Have a great day - and thank you for your continued efforts with this wonderful project.

Koenkk, 2020-02-29:
Thanks, it indeed makes much more sense to pair it to the signal repeater, Ive updated the pairing instructions accordingly (https://www.zigbee2mqtt.io/devices/E1757.html#pairing).

Thanks for you contribution, I assume this can be closed now.

closed 2020-02-29 by Koenkk

station7, 2020-03-19:
Sorry, I do not get it. Do I have to pair the TRADFRI signal repeater to zigbee2mqtt and the blinds to the signal repeater to be able to control the blinds via MQTT?

gimlichael, 2020-03-19:
> Sorry, I do not get it. Do I have to pair the TRADFRI signal repeater to zigbee2mqtt and the blinds to the signal repeater to be able to control the blinds via MQTT?

Yes, that is correct. The reason is, that Zigbee in the blinds has an extreme limited range .. so if your Zigbee Coordinator is more than 5-10 m. away, this is the approach you need to use. It took a little trial-and-error from my side, as the first documentation suggested direct link with the ZC.

Also, your repeater must be placed in the same room as the blinds - otherwise you are asking for problems :-)

Let me know if you experiencing problems - the IKEA setup is not for the faint hearted.

station7, 2020-03-19:
Michael, thanks for your response and your offer for help. As I learned, the blinds, the repeater and the button come pre-paired and I assume that this pairing is lost when I pair the repeater to zigbee2mqtt. Can you still use the button?

gimlichael, 2020-03-19:
Yes, they come pre-paired.

If you want to automate the setup, you need to do the setup we already talked about, and then add the button as well.

I have setup one button to control two blinds - and it works instant.

station7, 2020-04-08:
Michael, 

I got my FYRTUR blinds working since a few days, I did just like you proposed. Today, I realized that the signal repeater is no longer paired with zigbee2mqtt, but the blinds are still working, so I removed the signal repeater and the blinds still work with z2m. So far, so good. However, I miss the battery information in zigbee2mqtt. How can I check the battery level of my blinds?

gimlichael, 2020-04-09:
Hmm .. yeah, sometime it takes time for the devices to sync to your controller. I have learned, that patience is everything (not only from my parents - but in particular with Zigbee and Z-Wave).

What version of z2mqtt are you running? I am using version 1.11 in Docker.

jdtsmith, 2020-04-16:
> I have setup one button to control two blinds - and it works instant.

The up/down remote button is added/bound to z2m in your setup, right?  Does your button allow you to halt and reverse direction on a group of blinds in motion?