Koenkk/zigbee2mqtt #3018: Support for Wyze Lock

theupmachine, 2020-02-28:
# Bug Report
Support for  [Wyze Lock](https://wyze.com/wyze-lock)
## What happened
Device seemed to pair fine, but it did make a beep that sounded as though an error occurred about a minute after pairing. Messages are still being passed to zigbee2mqtt though. Also, the lock is still controllable though the wyze app via bluetooth.
## How to reproduce it
Pair device and observe output when locking/unlocking manually.
## Debug Info

This was added to the devices.js
```
    {
        zigbeeModel: ['Ford'],
        model: 'Ford',
        vendor: 'Yunding',
        description: 'Wyze Lock',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
    },
```

This is an example of the data that is received when manually locking and unlocking the door:
```
zigbee2mqtt:debug 2020-02-27 20:34:51: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,31,0,79,171,64,0,69,138,94,128,111,0,1,6,96,62,223,66,164,58,52,87,77,182,218,219,0,74,143,249,219,56,12,108,216,225,176,201,176,163,161,72,185,155,238,33,195,226,135,58,238,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,227,168,2,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-27 20:34:51: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,31,0,79,171,64,0,69,138,94,128,111,0,1,6,96,62,223,66,164,58,52,87,77,182,218,219,0,74,143,249,219,56,12,108,216,225,176,201,176,163,161,72,185,155,238,33,195,226,135,58,238,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,227,168,2,207,157,111,32,191,188,222,68]}'
zigbee2mqtt:debug 2020-02-27 20:34:53: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,32,0,79,171,64,0,69,192,131,128,112,0,1,6,96,62,223,66,164,58,52,80,77,182,218,219,0,74,143,249,219,56,12,108,216,225,176,201,176,163,161,72,185,155,238,33,195,226,135,58,232,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,227,168,4,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-27 20:34:53: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,32,0,79,171,64,0,69,192,131,128,112,0,1,6,96,62,223,66,164,58,52,80,77,182,218,219,0,74,143,249,219,56,12,108,216,225,176,201,176,163,161,72,185,155,238,33,195,226,135,58,232,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,227,168,4,207,157,111,32,191,188,222,68]}'
zigbee2mqtt:debug 2020-02-27 20:34:55: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,33,0,79,171,64,0,69,120,102,128,113,0,1,6,96,62,223,66,164,58,52,81,77,182,218,219,0,74,143,249,219,56,12,108,216,225,176,201,176,163,161,72,185,155,238,33,195,226,135,58,234,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,227,168,6,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-27 20:34:55: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,33,0,79,171,64,0,69,120,102,128,113,0,1,6,96,62,223,66,164,58,52,81,77,182,218,219,0,74,143,249,219,56,12,108,216,225,176,201,176,163,161,72,185,155,238,33,195,226,135,58,234,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,227,168,6,207,157,111,32,191,188,222,68]}'
zigbee2mqtt:debug 2020-02-27 20:34:56: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,34,0,117,171,64,0,107,179,172,128,114,0,1,6,96,62,223,66,164,58,52,82,77,182,218,219,0,74,137,249,219,56,14,236,217,225,248,193,176,235,161,72,185,156,238,33,195,226,135,33,166,78,180,29,37,192,222,135,88,54,179,149,218,47,89,51,172,191,243,212,180,207,212,73,71,235,110,35,56,67,170,70,165,44,52,28,36,232,134,226,82,66,198,148,36,76,217,49,87,30,222,179,177,189,84,72,188,195,54,70,202,200,223,71,172,76,180,30,32,182]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-27 20:34:56: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,34,0,117,171,64,0,107,179,172,128,114,0,1,6,96,62,223,66,164,58,52,82,77,182,218,219,0,74,137,249,219,56,14,236,217,225,248,193,176,235,161,72,185,156,238,33,195,226,135,33,166,78,180,29,37,192,222,135,88,54,179,149,218,47,89,51,172,191,243,212,180,207,212,73,71,235,110,35,56,67,170,70,165,44,52,28,36,232,134,226,82,66,198,148,36,76,217,49,87,30,222,179,177,189,84,72,188,195,54,70,202,200,223,71,172,76,180,30,32,182]}'
```
zigbee2mqtt version: 1.10.0
CC253X firmware version: 20190608

Koenkk, 2020-02-28:
Can you try with the snippet below and control it through: `zigbee2mqtt/0x680ae2fffe6c5324/set` payload `{"state": "LOCK"}` and `{"state": "UNLOCK"}`

```js
{
    zigbeeModel: ['Ford'],
    model: 'Ford',
    vendor: 'Yunding',
    description: 'Wyze Lock',
    supports: 'TODO',
    fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery_200],
    toZigbee: [tz.generic_lock],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await configureReporting.lockState(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

dmulcahey, 2020-02-29:
The lock needs to be calibrated before it will do anything. The normal lock cluster works right OOTB after the device is calibrated. I have to take some time to see what Wyze is doing during calibration but from what it makes you do it looks like it could be persisting the locked and unlocked position of the actuator and it looks like it may be recording gyro readings for open, closed and ajar positions for the door itself. They make you lock the lock, unlock the lock, open the door, close the door and then slightly open the door...

theupmachine, 2020-03-01:
I can confirm that this does work, but causes a timeout error.

```
debug 2020-03-01 14:53:10: Received MQTT message on 'zigbee2mqtt/0x680ae2fffe6c5324/set' with data '{"state": "UNLOCK"}'
debug 2020-03-01 14:53:10: Publishing 'set' 'state' to '0x680ae2fffe6c5324'
debug 2020-03-01 14:53:11: Received Zigbee message from '0x680ae2fffe6c5324', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
info  2020-03-01 14:53:11: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6c5324', payload '{"state":"UNLOCK","linkquality":28,"battery":100}'
debug 2020-03-01 14:53:21: Received Zigbee message from '0x680ae2fffe6c5324', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
info  2020-03-01 14:53:21: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6c5324', payload '{"state":"UNLOCK","linkquality":26,"battery":100}'
error 2020-03-01 14:53:25: Publish 'set' 'state' to '0x680ae2fffe6c5324' failed: 'Error: Command 0x680ae2fffe6c5324/1 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 48024 - 1 - 192 - 257 - 11 after 6000ms)'
debug 2020-03-01 14:53:25: Error: Command 0x680ae2fffe6c5324/1 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 48024 - 1 - 192 - 257 - 11 after 6000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
debug 2020-03-01 14:55:01: Received MQTT message on 'zigbee2mqtt/0x680ae2fffe6c5324/set' with data '{"state": "LOCK"}'
debug 2020-03-01 14:55:01: Publishing 'set' 'state' to '0x680ae2fffe6c5324'
debug 2020-03-01 14:55:03: Received Zigbee message from '0x680ae2fffe6c5324', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
info  2020-03-01 14:55:03: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6c5324', payload '{"state":"LOCK","linkquality":28,"battery":100}'
debug 2020-03-01 14:55:12: Received Zigbee message from '0x680ae2fffe6c5324', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
info  2020-03-01 14:55:12: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6c5324', payload '{"state":"LOCK","linkquality":28,"battery":100}'
error 2020-03-01 14:55:16: Publish 'set' 'state' to '0x680ae2fffe6c5324' failed: 'Error: Command 0x680ae2fffe6c5324/1 closuresDoorLock.lockDoor({"pincodevalue":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 48024 - 1 - 193 - 257 - 11 after 6000ms)'
debug 2020-03-01 14:55:16: Error: Command 0x680ae2fffe6c5324/1 closuresDoorLock.lockDoor({"pincodevalue":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 48024 - 1 - 193 - 257 - 11 after 6000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:344:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
```

Reporting from manually locking and unlocking the door is also not working
Unlocking:
```
debug 2020-03-01 15:05:25: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,72,0,79,171,64,0,69,18,152,128,67,0,1,6,96,62,223,66,164,58,52,88,77,182,218,219,4,21,87,249,219,56,103,20,216,225,176,201,176,163,161,72,185,156,238,33,195,226,131,101,60,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,231,247,208,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
debug 2020-03-01 15:05:25: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,72,0,79,171,64,0,69,18,152,128,67,0,1,6,96,62,223,66,164,58,52,88,77,182,218,219,4,21,87,249,219,56,103,20,216,225,176,201,176,163,161,72,185,156,238,33,195,226,131,101,60,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,231,247,208,207,157,111,35,191,188,222,68]}'
```
Locking:
```
debug 2020-03-01 15:06:12: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,73,0,79,171,64,0,69,185,30,128,68,0,1,6,96,62,223,66,164,58,52,89,77,182,218,219,4,21,7,249,219,56,91,20,216,225,176,201,176,163,161,72,185,155,238,33,195,226,131,101,99,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,231,247,143,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
debug 2020-03-01 15:06:12: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,73,0,79,171,64,0,69,185,30,128,68,0,1,6,96,62,223,66,164,58,52,89,77,182,218,219,4,21,7,249,219,56,91,20,216,225,176,201,176,163,161,72,185,155,238,33,195,226,131,101,99,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,231,247,143,207,157,111,32,191,188,222,68]}'
```
Similar messages are generated when opening and closing the door.
```
[... cut ...] 111,32,191,188,222,68]}' from endpoint 1 with groupID 0
debug 2020-03-01 15:15:00: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,79,0,79,171,64,0,69,85,107,128,74,0,1,6,96,62,223,66,164,58,52,87,77,182,218,219,4,19,23,249,219,56,55,22,216,225,176,201,176,163,161,72,185,155,238,33,195,226,131,99,115,78,180,29,40,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,231,241,159,207,157,111,32,191,188,222,68]}'
```

Koenkk, 2020-03-02:
Timeout errors should be fixed with:

```js
{
    zigbeeModel: ['Ford'],
    model: 'Ford',
    vendor: 'Yunding',
    description: 'Wyze Lock',
    supports: 'TODO',
    fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery_200],
    toZigbee: [tz.generic_lock],
    meta: {configureKey: 1, options: {disableDefaultResponse: true}},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await configureReporting.lockState(endpoint);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

Regarding the non-standard payload send when unlocking/locking manually, I need to figure out what bytes represent the lock/unlock status.

Could you provide the log of 5 manual locks and unlocks?

theupmachine, 2020-03-02:
Here is a log from several manual locks and unlocks

```
(UNLOCK) debug 2020-03-02 12:16:55: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,45,0,79,171,64,0,69,127,160,128,54,0,1,6,96,62,223,66,164,58,52,43,77,182,218,219,5,123,85,249,219,56,6,232,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,11,51,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,223,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
(LOCK) debug 2020-03-02 12:17:03: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,44,0,79,171,64,0,69,7,40,128,53,0,1,6,96,62,223,66,164,58,52,42,77,182,218,219,5,123,85,249,219,56,6,232,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,11,50,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,222,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
(UNLOCK) debug 2020-03-02 12:17:08: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,46,0,79,171,64,0,69,146,121,128,55,0,1,6,96,62,223,66,164,58,52,36,77,182,218,219,5,123,89,249,219,56,3,232,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,11,1,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,237,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
(LOCK) debug 2020-03-02 12:17:11: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,47,0,79,171,64,0,69,254,59,128,56,0,1,6,96,62,223,66,164,58,52,37,77,182,218,219,5,123,99,249,219,56,16,104,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,11,3,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,239,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
(UNLOCK) debug 2020-03-02 12:17:16: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,48,0,79,171,64,0,69,159,49,128,57,0,1,6,96,62,223,66,164,58,52,38,77,182,218,219,5,123,99,249,219,56,16,104,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,11,8,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,228,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
(LOCK) debug 2020-03-02 12:17:20: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,49,0,79,171,64,0,69,32,33,128,58,0,1,6,96,62,223,66,164,58,52,39,77,182,218,219,5,123,109,249,219,56,212,233,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,11,20,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,248,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
(UNLOCK) debug 2020-03-02 12:17:23: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,50,0,79,171,64,0,69,183,158,128,59,0,1,6,96,62,223,66,164,58,52,32,77,182,218,219,5,123,119,249,219,56,31,104,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,11,23,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,251,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
(LOCK) debug 2020-03-02 12:17:27: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,51,0,79,171,64,0,69,3,42,128,60,0,1,6,96,62,223,66,164,58,52,33,77,182,218,219,5,123,119,249,219,56,31,104,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,11,19,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,255,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
(UNLOCK) debug 2020-03-02 12:17:37: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,52,0,79,171,64,0,69,25,199,128,61,0,1,6,96,62,223,66,164,58,52,34,77,182,218,219,5,123,123,249,219,56,26,104,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,11,101,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,137,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
(LOCK) debug 2020-03-02 12:17:48: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,53,0,79,171,64,0,69,160,33,128,62,0,1,6,96,62,223,66,164,58,52,35,77,182,218,219,5,123,15,249,219,56,61,104,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,11,104,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,132,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
(UNLOCK) debug 2020-03-02 12:17:59: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,54,0,79,171,64,0,69,173,41,128,63,0,1,6,96,62,223,66,164,58,52,92,77,182,218,219,5,123,19,249,219,56,244,105,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,11,115,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,159,207,157,111,35,191,188,222,68]}' from endpoint 1 with groupID 0
[... cut ...] groupID 0
(LOCK) debug 2020-03-02 12:18:31: Received Zigbee message from '0x680ae2fffe6c5324', type 'raw', cluster 'manuSpecificPhilips', data '{"type":"Buffer","data":[5,90,18,59,0,79,171,64,0,69,252,158,128,68,0,1,6,96,62,223,66,164,58,52,89,77,182,218,219,5,123,53,249,219,56,46,232,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,11,83,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,153,191,207,157,111,32,191,188,222,68]}' from endpoint 1 with groupID 0
```

Koenkk, 2020-03-02:
Thanks, unfortunately I couldn't find which byte indicated the lock/unlock so probably it's inside the bits. Either somebody has to reverse engineer it (this will be very hard I think) or Wyze should provide the specification for this Wyze specific cluster. Perhaps you can try contacting them and mention if they do that this lock can be supported by Zigbee2mqtt (and probably also Deconz and ZHA).

theupmachine, 2020-03-02:
The 5th from the last number appears to be 35 when unlocked and 32 when locked which I have confirmed with repeat testing. This is conserved when controlling via bluetooth and when the door automatically locks when closed.

This does not agree with the first two lines of the logs I posted, but I think they may have gotten out of order as the 4th number appears to just count up and has them reversed.

Similarly, the last value is associated with the open/close state of the door, and is 68 when the door is closed and 71 when the door is open.

Here are two repeats of this sequence which seems to line up with my analysis:
1. The door is manually unlocked and closed
2. unlocked and open
3. unlocked and closed
4. automatically locked and closed
5-8. repeat

```
debug 2020-03-02 17:19:47: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,98,0,79,171,64,0,69,231,170,128,118,0,1,6,96,62,223,66,164,58,52,123,77,182,218,219,5,162,87,249,219,56,87,50,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,210,55,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,64,219,207,157,111,35,191,188,222,68]}'
debug 2020-03-02 17:19:54: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,99,0,79,171,64,0,69,107,70,128,119,0,1,6,96,62,223,66,164,58,52,116,77,182,218,219,5,162,81,249,219,56,85,178,216,225,176,201,176,163,161,72,185,136,238,33,195,226,130,210,62,78,180,29,223,192,222,135,167,201,14,149,218,59,192,51,167,97,171,183,238,230,64,210,207,157,111,35,191,188,222,71]}'
debug 2020-03-02 17:20:08: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,100,0,79,171,64,0,69,156,107,128,120,0,1,6,96,62,223,66,164,58,52,117,77,182,218,219,5,162,101,249,219,56,130,179,216,225,176,201,176,163,161,72,185,139,238,33,195,226,130,210,12,78,180,29,223,192,222,135,167,201,14,149,218,59,192,51,167,97,171,183,238,230,64,224,207,157,111,35,191,188,222,68]}'
debug 2020-03-02 17:20:11: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,101,0,79,171,64,0,69,168,51,128,121,0,1,6,96,62,223,66,164,58,52,118,77,182,218,219,5,162,111,249,219,56,133,51,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,210,15,78,180,29,40,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,64,227,207,157,111,32,191,188,222,68]}'
debug 2020-03-02 17:20:19: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,102,0,79,171,64,0,69,34,178,128,122,0,1,6,96,62,223,66,164,58,52,119,77,182,218,219,5,162,105,249,219,56,135,179,216,225,176,201,176,163,161,72,185,156,238,33,195,226,130,210,22,78,180,29,37,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,64,250,207,157,111,35,191,188,222,68]}'
debug 2020-03-02 17:20:28: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,103,0,79,171,64,0,69,69,43,128,123,0,1,6,96,62,223,66,164,58,52,112,77,182,218,219,5,162,125,249,219,56,136,179,216,225,176,201,176,163,161,72,185,136,238,33,195,226,130,210,31,78,180,29,223,192,222,135,167,201,14,149,218,59,192,51,167,97,171,183,238,230,64,243,207,157,111,35,191,188,222,71]}'
debug 2020-03-02 17:20:38: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,104,0,79,171,64,0,69,32,162,128,124,0,1,6,96,62,223,66,164,58,52,113,77,182,218,219,5,162,7,249,219,56,107,50,216,225,176,201,176,163,161,72,185,139,238,33,195,226,130,210,97,78,180,29,223,192,222,135,167,201,14,149,218,59,192,51,167,97,171,183,238,230,64,141,207,157,111,35,191,188,222,68]}'
debug 2020-03-02 17:20:41: No converter available for 'Ford' with cluster 'manuSpecificPhilips' and type 'raw' and data '{"type":"Buffer","data":[5,90,18,105,0,79,171,64,0,69,41,82,128,125,0,1,6,96,62,223,66,164,58,52,114,77,182,218,219,5,162,1,249,219,56,105,178,216,225,176,201,176,163,161,72,185,155,238,33,195,226,130,210,108,78,180,29,40,192,222,135,88,54,14,149,218,59,192,51,167,97,171,183,238,230,64,128,207,157,111,32,191,188,222,68]}'
```

dmulcahey, 2020-03-02:
I sent Wyze a request for the Spec. We’ll see what they say

dmulcahey, 2020-04-06:
Massive hack atm: https://github.com/zigpy/zha-device-handlers/blob/db1d0b4b88f3e4d3490a19aac4b0fbe9b2595f52/zhaquirks/wyze/lock.py#L80

not sure how accurate this is yet. It all seems to work though. Here are the logs I used to capture this:

```
OPEN / CLOSE

OPEN
79,171,64,0,69,225,11,128,119,0,1,6,96,181,225,73,166,106,45,143,216,208,93,161,36,241,207,0,31,71,226,105,192,66,3,194,196,138,118,202,254,177,163,247,18,105,107,22,158,30,172,79,74,166,89,253,81,85,52,108,30,68,159,157,191,194,24,188,154,24,88,242,136,164,34
79,171,64,0,69,186,76,128,121,0,1,6,96,181,225,73,166,106,45,141,216,208,93,161,36,241,75,0,31,71,129,105,192,66,3,194,196,138,118,202,254,177,163,247,18,105,107,22,29,30,172,79,74,166,89,253,81,85,52,108,30,68,159,157,191,194,24,188,154,24,88,113,136,164,34
79,171,64,0,69,80,102,128,123,0,1,6,96,181,225,73,166,106,45,139,216,208,93,161,36,246,225,0,31,71,206,235,192,66,3,194,196,138,118,202,254,177,163,247,18,105,107,17,135,30,172,79,74,166,89,253,81,85,52,108,30,68,159,157,191,194,24,188,154,24,95,235,136,164,34

CLOSED
79,171,64,0,69,44,154,128,120,0,1,6,96,181,225,73,166,106,45,140,216,208,93,161,36,241,171,0,31,71,9,104,192,66,3,194,196,138,118,202,254,178,163,247,18,105,107,22,252,30,172,79,74,166,89,253,81,85,52,108,30,68,159,157,191,194,24,188,154,24,88,144,136,164,34
79,171,64,0,69,76,83,128,122,0,1,6,96,181,225,73,166,106,45,138,216,208,93,161,36,241,15,0,31,71,178,105,192,66,3,194,196,138,118,202,254,178,163,247,18,105,107,22,94,30,172,79,74,166,89,253,81,85,52,108,30,68,159,157,191,194,24,188,154,24,88,50,136,164,34
79,171,64,0,69,235,189,128,124,0,1,6,96,181,225,73,166,106,45,136,216,208,93,161,36,246,175,0,31,71,250,107,192,66,3,194,196,138,118,202,254,178,163,247,18,105,107,17,242,30,172,79,74,166,89,253,81,85,52,108,30,68,159,157,191,194,24,188,154,24,95,158,136,164,34

LOCK/UNLOCK - APP
LOCKED
79,171,64,0,69,179,138,128,170,0,1,6,96,181,225,73,166,106,45,164,216,208,93,161,36,203,51,0,31,71,3,123,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,44,112,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,98,28,136,164,34
79,171,64,0,69,143,100,128,176,0,1,6,96,181,225,73,166,106,45,162,216,208,93,161,36,200,187,0,31,71,149,123,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,47,234,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,97,134,136,164,34
79,171,64,0,69,43,179,128,188,0,1,6,96,181,225,73,166,106,45,190,216,208,93,161,36,206,197,0,31,71,21,248,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,41,154,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,103,246,136,164,34
79,171,64,0,69,85,225,128,200,0,1,6,96,181,225,73,166,106,45,186,216,208,93,161,36,204,109,0,31,71,203,248,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,43,49,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,101,93,136,164,34

UNLOCKED
79,171,64,0,69,139,120,128,173,0,1,6,96,181,225,73,166,106,45,165,216,208,93,161,36,200,213,0,31,71,121,250,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,47,137,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,97,229,136,164,34
79,171,64,0,69,85,5,128,179,0,1,6,96,181,225,73,166,106,45,163,216,208,93,161,36,200,153,0,31,71,140,251,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,47,193,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,97,173,136,164,34
79,171,64,0,69,33,41,128,185,0,1,6,96,181,225,73,166,106,45,161,216,208,93,161,36,200,127,0,31,71,6,122,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,47,35,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,97,79,136,164,34
79,171,64,0,69,65,22,128,4,0,1,6,96,181,225,73,166,106,45,185,216,208,93,161,36,204,13,0,31,71,227,248,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,43,80,30,172,79,180,166,89,253,173,67,52,108,30,68,159,157,191,194,24,188,154,24,101,60,136,164,34

LOCK/UNLOCK - MANUAL

LOCK
79,171,64,0,69,172,89,128,5,0,1,6,96,181,225,73,166,106,45,182,216,208,93,161,36,194,189,0,31,71,55,253,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,37,224,30,172,79,176,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,107,140,136,164,34
79,171,64,0,69,81,137,128,8,0,1,6,96,181,225,73,166,106,45,181,216,208,93,161,36,194,105,0,31,71,104,253,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,37,58,30,172,79,176,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,107,86,136,164,34
79,171,64,0,69,50,59,128,10,0,1,6,96,181,225,73,166,106,45,179,216,208,93,161,36,194,9,0,31,71,64,253,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,37,95,30,172,79,176,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,107,51,136,164,34

UNLOCK
79,171,64,0,69,98,242,128,6,0,1,6,96,181,225,73,166,106,45,183,216,208,93,161,36,194,129,0,31,71,38,253,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,37,37,30,172,79,176,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,107,73,136,164,34
79,171,64,0,69,186,32,128,9,0,1,6,96,181,225,73,166,106,45,178,216,208,93,161,36,194,67,0,31,71,183,124,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,37,103,30,172,79,176,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,107,11,136,164,34
79,171,64,0,69,202,26,128,11,0,1,6,96,181,225,73,166,106,45,176,216,208,93,161,36,195,229,0,31,71,93,253,192,66,3,194,196,138,118,202,254,165,163,247,18,105,107,36,186,30,172,79,176,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,106,214,136,164,34

AUTO LOCK
79,171,64,0,69,44,174,128,36,0,1,6,96,181,225,73,166,106,46,66,216,208,93,161,36,199,39,0,31,71,12,126,192,66,3,194,196,138,118,202,254,162,163,247,18,105,107,32,120,30,172,79,189,166,89,253,174,170,52,108,30,68,159,157,191,194,24,188,154,24,110,20,136,164,34
```

dmulcahey, 2020-04-06:
It appears that each new pairing of the lock may change the data that is sent. I am working with a few others to confirm this.

dmulcahey, 2020-04-06:
https://forums.wyzecam.com/t/wyze-lock-traditional-zigbee-integration/91876/4 Here is the latest plea to them to get the spec.

dmulcahey, 2020-04-06:
[image]

OK from what I can tell... what we need to do is record the values that are in this position after calibration. Will be different each time the lock is joined to an account but they are stable after that. Make note of what is in that position for each action and you are good to go for the actions outlined above.

closed 2020-06-12 by stale[bot]

jscherry, 2020-06-12:
I found another repository shows the locks status but they haven’t gotten the lock unlock features working yet. I think it’s called wyze api?

dmulcahey, 2020-06-12:
I have all the actions outlined above “working” it’s just painful and requires calibration with the wyze app first.

jscherry, 2020-06-12:
https://github.com/JoshuaMulliken/ha-wyzeapi/releases. Can you get it to lock and unlock?

dmulcahey, 2020-06-12:
Yeah, that project is API based. I have it working with ZHA and HA. I’ll get it going again and I’ll post a video this weekend.

jscherry, 2020-06-13:
Excellent I’ll keep up with it. I’d like to try it out. Is it set up by adding the code above to config file or do you add it to the zigbee2mqtt config file?

mike1627, 2020-09-24:
Any updates to this?  I tried installing the wyze quirk but it still won't update the status of the lock if it's been manually opened/closed

mike1627, 2020-09-27:
> [image]
> 
> OK from what I can tell... what we need to do is record the values that are in this (denoted by the arrow directly to the left of the 2 highlighted bytes) position after calibration. Will be different each time the lock is joined to an account but they are stable after that. Make note of what is in that position for each action and you are good to go for the actions outlined above.

What are you using to obtain this data?

auntphil, 2022-02-20:
Was there ever any progress on this integration? I have two locks I want to move over to zigbee2mqtt. I can get it paired with zigbee2mqtt and performed the calibration afterwards but it still shows unsupported and does not pass the device on to the broker.

theupmachine, 2022-06-24:
I'm not sure if it was a firmware update to the device, or an update to zigbee2mqtt, but this lock seems to be working correctly now. The lock state is correctly reported after manually (or bluetooth) toggling the door lock.
The open/closed state of the door is not working however.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['Ford'],
    model: 'Ford',
    vendor: 'Yunding',
    description: 'Wyze Lock',
    fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
    toZigbee: [tz.lock],
    meta: {configureKey: 1, options: {disableDefaultResponse: true}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.endpoints[0];
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await reporting.lockState(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.lock(), e.battery()],
};

module.exports = definition;
```

segdy, 2022-06-30:
After long back and forth I am on the same status as you: I can finally see it and use Lock and Unlock via zigbee2mqtt. Sadly lock status does not work and battery neither (shows wrongfully 100%).

Worse, the export to Home Assistant now only shows the "Lock" button (probably because lock status does not work). 

Any hope to get this to work?

theupmachine, 2022-06-30:
@segdy Sorry, if it wasn't clear in my previous post. The code I posted is for an external converter that is working for me. Is that what you are using?

segdy, 2022-06-30:
What I did is:

I copied [your code](https://github.com/Koenkk/zigbee2mqtt/issues/3018#issuecomment-1165982160) into `/opt/zigbee2mqtt/data/YundingFord.js`, then added the following line to my `configuration.yaml`:

```
 external_converters:
  - YundingFord.js
```

Then it took a few re-pairs. I am not sure if it was required but after pairing I also enabled door status, calibrated and disabled door status again. Everything works now except the lock status. For example, when I lock, the status stays unlocked or when I unlock, the status stays locked. There is a "Refresh" button in z2m and only after I click this, the lock status gets updated. Is this the case for you too? Can this be automated such that the status is automatically updated?

theupmachine, 2022-07-01:
No, for me it works as you would expect. The next step for you is probably to enable debug logging and look at the logs when locking or unlocking the door. 
Also, what firmware version do you have for the lock within the wyze app? Mine is 2.1.22.0

If the logs don't show anything useful and your firmware is the same version as mine, then it will probably take someone more knowledgeable than me to help.

segdy, 2022-07-01:
Yes, firmware is the same, 2.1.22.0. I added to zigbee2mqtt configuration.yaml:

```
advanced:
  log_level: debug
```

Interestingly no zigbee messages, only mqtt. Do I miss something? Anyway, if I lock, I get only this message:

```
Zigbee2MQTT:debug 2022-06-30 17:15:38: Received MQTT message on 'zigbee2mqtt/WyzeLock/set' with data '{"state":"LOCK"}'
Zigbee2MQTT:debug 2022-06-30 17:15:38: Publishing 'set' 'state' to 'WyzeLock'
```

Nothing else, and the status does not update. Only after I click the refresh button, I get:

```
Zigbee2MQTT:debug 2022-06-30 17:16:21: Received MQTT message on 'zigbee2mqtt/WyzeLock/get' with data '{"state":""}'
Zigbee2MQTT:debug 2022-06-30 17:16:21: Publishing get 'get' 'state' to 'WyzeLock'
Zigbee2MQTT:info  2022-06-30 17:16:23: MQTT publish: topic 'zigbee2mqtt/WyzeLock', payload '{"battery":100,"linkquality":255,"lock_state":"locked","state":"LOCK"}'
```

Does any of this look off to you?

theupmachine, 2022-07-01:
That does help I think. For some reason zigbee2mqtt is not seeing the "attributeReport" messages after sending the command to the lock. This what it should look like:

```
Debug 2022-06-30 19:23:01Received MQTT message on 'zigbee2mqtt/0x680ae2fffe6c5324/set' with data '{"state":"UNLOCK"}'
Debug 2022-06-30 19:23:01Publishing 'set' 'state' to '0x680ae2fffe6c5324'
Debug 2022-06-30 19:23:01Received Zigbee message from '0x680ae2fffe6c5324', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Info 2022-06-30 19:23:01MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6c5324', payload '{"battery":9.5,"linkquality":47,"lock_state":"unlocked","state":"UNLOCK"}'
```

Could you try re-enabling the door position setting in the app? That is the only setting I have set differently than you.

segdy, 2022-09-27:
Hi @theupmachine I think it was the battery! I tried everything, I even replaced my ConnBee II with the (well praised) Sonoff Zigbee stick. Same. Got new batteries (which were rechargeable though). Many iterations of removing, repairing.

Finally I got actual new batteries. Now it seems to work :-)

jonnylangefeld, 2023-05-21:
Hmm I'm running into the same issue. The lock is connected and I can click lock/unlock, but the state is not reported right after. No `attributeReport` for the lock in the debug logs. When I click the refresh button manually it gets the correct state. My firmware version is also `2.1.22.0`. I deleted and reconnected a bunch of times. Also deleted from the Wyze app and reconfigured. I also put in brand new non-chargeable batteries. 

No luck so far 😔 

@segdy besides putting in new batteries, is there anything else that you did right around the same time?

segdy, 2023-07-01:
It is a little buggy for me and I need to remove and repair a couple of times for success. but new batteries always did wonder so far.
I don’t use it for my main door since it’s so crappy.

BrianHenryIE, 2023-07-15:
Is it possible to disable auto-lock through Z2M?

Elderwolf1700, 2023-07-15:
@theupmachine Does it still work for you? I just tried to add it to zigbee2mqtt and the lock status dosen't update unless it's through HA. I tried following all the guides of adding stuff like the YundingFord.js code but nothing seems to work. I haven't done much with zigbee2mqtt code editing so it may be my fault.

segdy, 2024-10-17:
I’m back having the issue that the lock status doesn’t update when manually operated :-( 

is there a single person here for who it works? Just so I can rule out any configuration issue?

segdy, 2024-10-17:
@dmulcahey once I got the values (177, 178 etc in your screenshot), how do I update the converter?

the converter seems to be extremely straight forward and doesn’t use any numerical values:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/wyze.ts

gmallapp, 2024-12-20:
@segdy,
I've had to do this quite differently. After calibration, I'm looking at the 46th & 57th index of the message and this is how it looks like for me
[image]
I've masked the other indexes, because I don't understand the security implications of posting them for everyone to see.
and here's the code I'm using. Every 30 or 60mins, the lock sends out a message that is smaller or longer. So the code discards any msg that is smaller than 70 or longer than 90 values:

Hope this helps..
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const logger = require('zigbee-herdsman-converters/lib/logger');
//const extend = require('zigbee-herdsman-converters/lib/extend');

const e = exposes.presets;

const fzLocal = {
    c4_lock_operation_event: {
        cluster: 'manuSpecificAssaDoorLock',
        type: ['raw'],
        convert: async (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.length <70 or > 90){
                return result;
            }
            if (msg.data[57] == 56 && msg.data[46] == 218){
                logger.logger.warning("the lock is unlocked via the app")
                result.state = 'UNLOCK'
                result.lock_state = 'unlocked'
            }

            else if (msg.data[57] == 56 && msg.data[46] == 221){
                logger.logger.warning("the lock is locked via the app")
                result.state = 'LOCK'
                result.lock_state = 'locked'
            }

            else if (msg.data[57] == 60 && msg.data[46] == 218){
                logger.logger.warning("the lock is unlocked manually")
                result.state = 'UNLOCK'
                result.lock_state = 'unlocked'
            }

            else if (msg.data[57] == 60 && msg.data[46] == 221){
                logger.logger.warning("the lock is locked manually")
                result.state = 'LOCK'
                result.lock_state = 'locked'
            }

            else if (msg.data[57] == 49 && msg.data[46] == 221){
                logger.logger.warning("the lock is locked via auto lock")
                result.state = 'LOCK'
                result.lock_state = 'locked'
            }

            if (msg.data[57] == 198 && msg.data[46] == 206){
                logger.logger.warning("the door is open")
                result.door_state = 'open'
            }

            else if (msg.data[57] == 198 && msg.data[46] == 205){
                logger.logger.warning("the door is closed")
                result.door_state = 'closed'
            }
            return result;

        },
    },
};

const definition =
    {
        zigbeeModel: ['Ford'],
        model: 'WLCKG1',
        vendor: 'Wyze',
        description: 'wyzeee Lock',
        fromZigbee: [fz.lock, fz.lock_operation_event, fzLocal.c4_lock_operation_event, fz.battery],
        toZigbee: [tz.lock],
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.endpoints[0];
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg', 'manuSpecificAssaDoorLock']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.lock(), e.battery()],
    }

module.exports = definition;
```

segdy, 2024-12-21:
Thank you @gmallapp , that’s awesome! Just to make sure I understand … the log messages you get when z2m is running? Does it need to be paired already or shouldn’t it?

and then you do manually lock/unlock via app, manually, auto lock and read out what’s on offsets 46 and 57? The numbers (192, 49, 60 etc) May be different for me?

gopal87-home, 2024-12-21:
1. Pair via Bluetooth on your app and finish calibration. 
2. Pair via z2m.
3. In config, set logging to debug, and tail the logs, now unlock/ open/close/lock manually, via app etc and note the message for each action.
4. Replace the if else in the code with your values for the offsets. The numbers might be different for you.

Elderwolf1700, 2024-12-22:
I may have done something wrong, but when I go for the values in the logs all the output codes are the same resulting in the same 46th and 57th value having the same value for every action. Are they supposed to be different?

Elderwolf1700, 2024-12-28:
@gopal87-home Is there something else that can be done or should be done. I keep trying as per my last comment and keep getting the same values. Ive restarted everything, tried re-pairing 100s of times and still nothings different.

duartefq, 2024-12-28:
Maybe you should look at different values. [Take this implementation](https://github.com/zigpy/zha-device-handlers/blob/db1d0b4b88f3e4d3490a19aac4b0fbe9b2595f52/zhaquirks/wyze/lock.py#L91) for example, the lock/door state is at index 41, and the source (app or manual) is at index 52. 

Unfortunately this is all trial and error. In my case, I'm using offsets 57 for source and 46 for state.

```js
// Offset Position of the event type. 
// See SourceMessage for values found.
const EVENT_TYPE = 57;

// Offset Position of the event state. 
// For doors, it's either "open" or "closed" (see DoorState), 
// For locks, it's either "locked" or "unlocked" (see LockState).
const EVENT_STATE = 46;

// Values found for event types.
// AUTO means auto locked, and APP is lock via the official app.
const SourceMessage = {
  DOOR: 242,
  MANUAL: 8,
  AUTO: 5,
  APP: 12,
};

// Values found for Door State.
const DoorState = {
  OPEN: 91,
  CLOSED: 88,
};

// Values found for Lock State.
const LockState = {
  LOCKED: 72,
  UNLOCKED: 79,
};
```

Elderwolf1700, 2024-12-29:
I keep trying and can't get it to work even after trying multiple mumbers. Is it supposed to fully pair or is it supposed to fail multiple times during the paring process to only "pair" after the lock has timed out? I'm going mad trying to get this to work, it's been like a year and I don't know what's wrong.

gopal87-home, 2024-12-31:
@Elderwolf1700 , I would remove the external_converters: config and just make sure the lock is fully paired first. You should be able to LOCK/UNLOCK from the Z2M GUI. locking and unlocking from the Wyze App using the bluetooth LE should also reflect the LOCK/UNLOCK state in Z2M. This functionality works well without the external_converter customization.
You did say earlier that you were seeing the debug messages, so your lock should have paired with Z2M.

Can you post the entire message string for the 4 states:
Open-manually:
Door-Open:
Door-Closed:
Close-manually:

Elderwolf1700, 2025-01-01:
@gopal87-home, First thing i did was make sure there was no Converters.

I paired the lock to Wyze and then tried to pair it to Z2M. It did it's slowly flashed green while it tries to pair and kept pairing, failing and re-pairing. After 2 minutes it does the failed to pair beep and has a red light but looks liked it paired (maybe). During those 2 minutes it's giving me the following notifications https://imgur.com/a/HmK8BDX.

I can unlock and lock from Z2M just fine but if I do it from Wyze, it dosent reflect in Z2M.

Here is the entire message string from the 4 states (minus some jumble from other zigbee devices)

**Open manually:**

[2024-12-31 22:29:52] debug: 	z2m: No converter available for 'WLCKG1' with cluster 'manuSpecificAssaDoorLock' and type 'raw' and data '{"data":[5,90,18,12,0,79,171,64,0,69,106,70,128,153,0,13,6,96,124,223,81,181,235,228,247,40,179,236,12,81,9,241,99,8,210,173,65,141,245,35,134,39,146,65,84,140,91,55,15,148,153,171,145,124,159,100,108,64,197,232,105,37,205,251,15,9,209,116,155,242,117,56,248,64,254,240,157,250,90,182,13,232,254,222,87],"type":"Buffer"}'
 

**Door Open:**

2024-12-31 23:01:54] debug: 	zh:zstack:unpi:parser: <-- [254,105,68,129,0,0,0,252,12,20,1,1,0,174,0,145,124,126,0,0,85,5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146]
[2024-12-31 23:01:54] debug: 	zh:zstack:unpi:parser: --- parseNext [254,105,68,129,0,0,0,252,12,20,1,1,0,174,0,145,124,126,0,0,85,5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146]
[2024-12-31 23:01:54] debug: 	zh:zstack:unpi:parser: <-- [65,84,140,79,55,15,148,153,171,153,255,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,30,250,90,182,13,232,254,222,84,12,20,0,107]
[2024-12-31 23:01:54] debug: 	zh:zstack:unpi:parser: --- parseNext [254,105,68,129,0,0,0,252,12,20,1,1,0,174,0,145,124,126,0,0,85,5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146,65,84,140,79,55,15,148,153,171,153,255,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,30,250,90,182,13,232,254,222,84,12,20,0,107]
[2024-12-31 23:01:54] debug: 	zh:zstack:unpi:parser: --> parsed 105 - 2 - 4 - 129 - [0,0,0,252,12,20,1,1,0,174,0,145,124,126,0,0,85,5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146,65,84,140,79,55,15,148,153,171,153,255,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,30,250,90,182,13,232,254,222,84,12,20,0] - 107
[2024-12-31 23:01:54] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":64512,"srcaddr":5132,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":174,"securityuse":0,"timestamp":8289425,"transseqnumber":0,"len":85,"data":{"type":"Buffer","data":[5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146,65,84,140,79,55,15,148,153,171,153,255,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,30,250,90,182,13,232,254,222,84]}}
[2024-12-31 23:01:54] debug: 	zh:controller: Failed to parse frame: Error: Cluster 'manuSpecificAssaDoorLock' has no command '0'
[2024-12-31 23:01:54] debug: 	zh:controller: Received payload: clusterID=64512, address=5132, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=174, frame=undefined
[2024-12-31 23:01:54] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-31 23:01:54] debug: 	z2m: Received Zigbee message from '0x003c84fffe2dd2de', type 'raw', cluster 'manuSpecificAssaDoorLock', data '{"data":[5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146,65,84,140,79,55,15,148,153,171,153,255,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,30,250,90,182,13,232,254,222,84],"type":"Buffer"}' from endpoint 1 with groupID 0
[2024-12-31 23:01:54] debug: 	z2m: No converter available for 'WLCKG1' with cluster 'manuSpecificAssaDoorLock' and type 'raw' and data '{"data":[5,90,18,107,0,79,171,64,0,69,192,161,128,55,0,13,6,96,124,223,81,181,235,228,148,40,179,236,12,81,1,113,99,8,210,205,71,141,245,35,134,39,146,65,84,140,79,55,15,148,153,171,153,255,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,30,250,90,182,13,232,254,222,84],"type":"Buffer"}'

**Door closed :**

[2024-12-31 23:02:57] debug: 	zh:zstack:unpi:parser: <-- [254,105,68,129,0,0,0,252,12,20,1,1,0,185,0,64,188,186,0,0,85,5,90,18,109,0,79,171,64,0,69,249,169,128,57,0,13,6,96,124,223,81,181,235,228,150,40,179,236,12,81,1,183,99,8,210,159,199,141,245,35,134,39,146]
[2024-12-31 23:02:57] debug: 	zh:zstack:unpi:parser: --- parseNext [254,105,68,129,0,0,0,252,12,20,1,1,0,185,0,64,188,186,0,0,85,5,90,18,109,0,79,171,64,0,69,249,169,128,57,0,13,6,96,124,223,81,181,235,228,150,40,179,236,12,81,1,183,99,8,210,159,199,141,245,35,134,39,146]
[2024-12-31 23:02:57] debug: 	zh:zstack:unpi:parser: <-- [65,84,140,76,55,15,148,153,171,153,60,159,100,108,186,197,232,105,218,50,251,15,9,209,116,155,242,117,56,248,64,254,248,221,250,90,182,13,232,254,222,87,12,20,0,134]
[... cut ...] 2,87],"type":"Buffer"}'

**Close manually:**

[2024-12-31 22:42:21] debug: 	z2m: No converter available for 'WLCKG1' with cluster 'manuSpecificAssaDoorLock' and type 'raw' and data '{"data":[5,90,18,86,0,79,171,64,0,69,56,2,128,34,0,13,6,96,124,223,81,181,235,228,137,40,179,236,12,81,10,131,99,8,210,120,193,141,245,35,134,39,146,65,84,140,92,55,15,148,153,171,146,0,159,100,108,64,197,232,105,37,205,251,15,9,209,116,155,242,117,56,248,64,254,243,225,250,90,182,14,232,254,222,87],"type":"Buffer"}'

segdy, 2025-01-02:
I also tried this but for me, many fields are **different** :-( Even more so, when I repeatedly lock/unlock via app, the "unlock" actions have up to 36 bytes different from run to run! Are really just two bytes changing for you all?

segdy, 2025-01-02:
@gmallapp I did the same as you but as I mentioned above, many more fields change for me. I repeated the actions and some of them seem to be counters (or counter-like), some random and I think can be ignored (red).

I do not get any message for door open/close (probably because I have disabled "Door Position" and autolock/unlock.

This is my result:

[image]

Now my issue is both offsets 47 and 81 seem to represent lock/unlocked state (for offset 47, unlock=56 and lock=63 and for offset 81, unlock=220 and lock=223).

Worse now, more fields seem to represent app vs manual: Offset 58 is 91 for app and 95 for manual and similarly offsets 62 and 63.

Any insights? Which ones do you think are the right offsets?

Elderwolf1700, 2025-01-02:
> **EDIT**: When I compare with @Elderwolf1700 logs it's similar however, it's not clear to me what exactly he means with "open/close manually" and "door open/close". The values are all very different but the offsets look similar. What surprises me though is that my messages are always 81 bytes but his 85 ...

I followed what as asked of me to provide and gave the values for manual unlock, open the door, reclose it and then manually locking it.

gopal87-home, 2025-01-03:
I like the way @duartefq structured the event_type and eventState:
@Elderwolf1700, looking at your debug messages. your event_type is 57th offset and event_state is 46th offset. You haven't provided the LOCK/UNLOCK via APP/Z2M messages [I missed asking for it in my previous message]. But you can look at the value at the 57th offset in your debug message when you LOCK/UNLOCK via the app/z2m and get the event code for it. You can plug your values in the external convertor code and restart z2m.
 
[image]

Here are your values: 
```js
// Offset Position of the event type. 
// See SourceMessage for values found.
const EVENT_TYPE = 57;

// Offset Position of the event state. 
// For doors, it's either "open" or "closed" (see DoorState), 
// For locks, it's either "locked" or "unlocked" (see LockState).
const EVENT_STATE = 46;

// Values found for event types.
// AUTO means auto locked, and APP is lock via the official app.
const SourceMessage = {
  DOOR: 186,
  MANUAL: 64,
  AUTO: '??',
  APP: '??',
};

// Values found for Door State.
const DoorState = {
  OPEN: 79,
  CLOSED: 76,
};

// Values found for Lock State.
const LockState = {
  LOCKED: 92,
  UNLOCKED: 91,
};```

segdy, 2025-01-03:
@gopal87-home these offsets are somewhat consistent with mine... but why are you choosing 57 instead of 61 or 62 and why are you choosing 46 instead of 80 ? See my posted table.

Posting mine again with @Elderwolf1700 's as a comparison 

[image]

gopal87-home, 2025-01-03:
@segdy ,
The debug msg data field is of type array/list, so when I say index value 46th, it's element at index 46 in the array. the first element  "5" is at index 0 not index 1 as in your spreadsheet image.
so Column AW in your spread sheet is index 46 not 47 which is the EVENT_STATE
column BH is index 57 not 58, which is the EVENT_TYPE

The wyze lock is also able to detect if your door was opened or closed and it sends messages for those. That is what I asked for. The code I provided earlier still doesn't expose the door_status yet. 

```js
// Offset Position of the event type. 
// See SourceMessage for values found.
const EVENT_TYPE = 57;

// Offset Position of the event state. 
// For doors, it's either "open" or "closed" (see DoorState), 
// For locks, it's either "locked" or "unlocked" (see LockState).
const EVENT_STATE = 46;

// Values found for event types.
// AUTO means auto locked, and APP is lock via the official app.
const SourceMessage = {
  DOOR: '??',
  MANUAL: 95,
  AUTO: '??',
  APP: '91',
};

// Values found for Door State.
const DoorState = {
  OPEN: ??,
  CLOSED: ??,
};

// Values found for Lock State.
const LockState = {
  LOCKED: 63,
  UNLOCKED: 56,
};
```

you should be able to plug these values in the external convertor code.

gopal87-home, 2025-01-03:
@segdy sorry, sent my response before I saw your update. I agree that in your case, values index 46 and 80 do look similar (Though it doesn't in Elderwolf's message), and 57,61,62 also look to have the same codes. we don't really know what any of these values really mean given that Wyze hasn't provided us with any documentation. We are just reverse engineering it and trying to make it work based on trial and error. if those other values work better for you, you can use them as well.

Elderwolf1700, 2025-01-03:
@gopal87-home ill try that when I get home. I'm just curious, is it normal for the lock to fail pairing but then still pair with Z2M?

segdy, 2025-01-03:
I picked offsets 46 and 57 now, same as @gmallapp . Works now. Thank you!!

Elderwolf1700, 2025-01-03:
@gopal87-home Added the value for the 57th offset and then added it all to the external Converters. Same as always, no status update in Z2M when done manually or in the Wyze app.

I'll add the numbers just to be sure but I think this may be something else with all the errors when pairing.

**Auto lock:** I got 244

5,90,18,28,0,79,171,64,0,69,34,179,128,133,0,13,6,96,124,223,81,181,235,228,233,40,179,236,12,82,108,193,99,8,210,233,154,141,245,35,134,39,146,65,84,140,92,55,15,148,153,168,244,71,159,100,108,77,197,232,105,37,205,251,15,9,209,116,155,242,117,56,248,64,253,149,166,250,90,182,14,232,254,222,87

**App lock:** I got 247

5,90,18,29,0,79,171,64,0,69,125,41,128,137,0,13,6,96,124,223,81,181,235,228,232,40,179,236,12,82,111,140,99,8,210,44,90,141,245,35,134,39,146,65,84,140,91,55,15,148,153,168,247,119,159,100,108,68,197,232,105,38,35,251,15,9,209,116,155,242,117,56,248,64,253,150,150,250,90,182,13,232,254,222,87

Elderwolf1700, 2025-01-04:
Z2M broke and then upon restarting it had updated somehow.... the lock still doesn't work properly and I also cant find where to add external converters now.

mariusmuja, 2025-02-07:
This stopped working on ZHA a while ago apparently, I just posted a working ZHA version [here](https://gist.github.com/mariusmuja/d1953a965ea3755e0ef4099cfaafd732), in case people looking for that end up reading this thread.

The ZHA version uses offsets 41 and 52 vs 46 and 57 on Z2M, because the ZCL header is not included in the arguments passed to the ZHA message handler.