Koenkk/zigbee2mqtt #3019: Innr bulbs - ZCL OTA: Query Next Image Request network chatter

DomiStyle, 2020-02-28:
# Bug Report
## What happened

Since the OTA feature was introduced the network gets spammed with Next Image Requests for every Innr bulb every few seconds. The mesages repeat for each bulb over and over again.

I have a lot of Innr bulbs, making the debug log pretty much unreadable because these messages are coming in multiple times a second. Wireshark also shows a lot of traffic on the network.

    Received Zigbee message from 'hallway03', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4454,"imageType":256,"fileVersion":285605891}' from endpoint 1 with groupID 0
    No converter available for 'RS 125' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4454,"imageType":256,"fileVersion":285605891}'

    Received Zigbee message from 'bathroom_mirror01', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4454,"imageType":276,"fileVersion":268768615}' from endpoint 1 with groupID 0
    No converter available for 'RS 228 T' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4454,"imageType":276,"fileVersion":268768615}'

    Received Zigbee message from 'staircase03', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4454,"imageType":257,"fileVersion":285605891}' from endpoint 1 with groupID 0
    No converter available for 'RB 165' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4454,"imageType":257,"fileVersion":285605891}'

Is there a way to stop these messages?

## What did you expect to happen
Nothing

## How to reproduce it (minimal and precise)
Get the latest zigbee2mqtt version, add an Innr bulb. Models where it happens for sure: RS 228 T, RB 165, RS 125.

## Debug Info
zigbee2mqtt version: 1.11.0
CC1352P-2

    {"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}

Kryzek, 2020-02-28:
Related : https://github.com/Koenkk/zigbee-herdsman-converters/issues/1015

Koenkk, 2020-02-28:
Should be fixed in the latest dev branch.

In case a device requests an OTA and we don't support it we respond with `NO_IMAGE_AVAILABLE`, hopefully the device stops requesting OTAs after that. @DomiStyle can you verify this?

DomiStyle, 2020-02-28:
@Koenkk I'm getting this exception with the latest dev branch from Docker Hub:

    Received Zigbee message from 'hallway04', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4454,"imageType":256,"fileVersion":285605891}' from endpoint 1 with groupID 0
    No converter available for 'RS 125' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4454,"imageType":256,"fileVersion":285605891}'
    Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: Parameter 'manufacturerCode' is missing
        at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:356:27)
        at Generator.next (<anonymous>)
        at /app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:8:71
        at new Promise (<anonymous>)
        at __awaiter (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:4:12)
        at Endpoint.commandResponse (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:16)
        at OTAUpdate.onZigbeeEvent (/app/lib/extension/otaUpdate.js:43:28)
        at Controller.callExtensionMethod (/app/lib/controller.js:330:44))

Koenkk, 2020-02-28:
@DomiStyle did you also update the dependencies? (with `npm ci`) because it requires the latest zigbee-herdsman.

DomiStyle, 2020-02-28:
@Koenkk I'm using the latest-dev Docker image from 10 minutes ago, should I mount the herdsman repository manually?

Koenkk, 2020-02-28:
No, found the issue will fix it.

Koenkk, 2020-02-29:
Fixed in latest dev.

DomiStyle, 2020-02-29:
@Koenkk Looks like that did the trick, the bulbs will now only ask once. Network is quiet in Wireshark and debug log is readable again.

Koenkk, 2020-02-29:
@DomiStyle cool! Thanks for finding and verifying the issue :smile: 

Assuming this can be closed now.

closed 2020-02-29 by Koenkk

DomiStyle, 2020-03-03:
@Koenkk From what I have seen so far the bulbs seem to ask again every 5-10 minutes. Which is already a huge improvement and I assume there is no way to prevent that and it's just programmed into the Innr bulbs?

Koenkk, 2020-03-03:
@DomiStyle I guess so, at least I don't know what we can do else from the zigbee2mqtt side.