Koenkk/zigbee2mqtt #3020: Legrand Power socket interview fails

swingerman, 2020-02-28:
# Bug Report
Failed to pair Legran power socket. I followed the instructions for both the device and zigbee2mqtt.
- Herdsman logging enabled
- Tried 50x+ times
- Tried with reflashed sniffer adapter
- Tried with another sniffer adapter
- Tried with USB extension cable
- Tried holding the adapter next to device
- Tried full reboot

## What happened
Tried to pair Legrand power socket as per the instructions but the interview fails

### Failures:
#### zigbee2mqtt addon v1.11.0
`Interview failed because can not get node descriptor ('0x00047400000adb80')`
#### zigbee2mqtt addon v1.10.x
`Interview failed because can not get active endpoints ('0x00047400000adb80')`

## What did you expect to happen
I expected that the interview or pairing will be successful.

## How to reproduce it (minimal and precise)
- Enable joining new ZigBee devices
- Put the Legrand power socket into pairing mode (hold reset for 8s)
- Check zigbee2mqtt addon log

## Debug Info
zigbee2mqtt version: 1.11.0, 1.10.x
CC253X firmware version: 20190608
https://pastebin.com/wnY6SZKE
https://pastebin.com/7FA1FB72

Koenkk, 2020-02-28:
@cemonneau  do you know the reason of this?

npfp, 2020-04-05:
Hi there,

I got a similar issue.

**zigbee2mqtt addon v1.12.0**

```zigbee2mqtt:info  2020-04-05 17:16:20: Starting interview of '0x00047400000*****'
zigbee2mqtt:info  2020-04-05 17:16:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00047400000*****"}}'
2020-04-05T15:16:20.092Z zigbee-herdsman:controller:device Interview - start device '0x00047400000*****'
2020-04-05T15:16:20.094Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-05T15:16:20.095Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-05T15:16:20.107Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-05T15:16:20.107Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-05T15:16:20.107Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-05T15:16:20.107Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-05T15:16:20.107Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-05T15:16:30.104Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x00047400000*****', attempt 1
```

Have you found a solution to pair it?

Koenkk, 2020-04-05:
@npfp Can you try with the latest edge addon in 4 hours from now?

npfp, 2020-04-06:
@Koenkk Many thanks for the quick answer.

I'm testing 

`zigbee2mqtt:info  2020-04-06 07:54:36: Starting zigbee2mqtt version 1.12.0-dev (commit #52d4193bc714a52a52e30801d2ab15d07695886b)`

but unfortunately it still doesn't work

```
zigbee2mqtt:info  2020-04-06 08:15:02: Starting interview of '0x00047400000*****'
zigbee2mqtt:info  2020-04-06 08:15:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00047400000*****"}}'
zigbee2mqtt:error 2020-04-06 08:16:02: Failed to interview '0x00047400000*****', device has not successfully been paired
zigbee2mqtt:info  2020-04-06 08:16:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00047400000*****"}}'
```

Koenkk, 2020-04-06:
Pushed another fix, can you try again with latest dev?

npfp, 2020-04-06:
Just tested it, got this log:

zigbee2mqtt:info  2020-04-06 21:10:48: Starting interview of '0x00047400000*****'
zigbee2mqtt:info  2020-04-06 21:10:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00047400000*****"}}'
2020-04-06T19:10:48.120Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-06T19:10:48.120Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-06T19:10:48.120Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-06T19:10:48.121Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-06T19:10:48.121Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:10:58.114Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000*****', attempt 1
2020-04-06T19:10:58.116Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-06T19:10:58.117Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-06T19:10:58.129Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-06T19:10:58.130Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-06T19:10:58.130Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-06T19:10:58.130Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-06T19:10:58.131Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:10:58.825Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-06T19:10:58.826Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-06T19:10:58.827Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [108,136,194,212,10,0,0,116,4,0,0,0] - 11
2020-04-06T19:10:58.828Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":34924,"extaddr":"0x00047400000*****","parentaddr":0}
2020-04-06T19:10:58.828Z zigbee-herdsman:controller:log Device '0x0004740000*****' joined
2020-04-06T19:10:58.829Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:10:58.829Z zigbee-herdsman:controller:log Device '0x00047400000*****' accepted by handler
2020-04-06T19:10:58.829Z zigbee-herdsman:controller:log Not interviewing '0x00047400000*****', completed 'false', in progress 'true'
2020-04-06T19:11:08.123Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000*****', attempt 2
2020-04-06T19:11:08.125Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-06T19:11:08.126Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-06T19:11:08.138Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-06T19:11:08.139Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-06T19:11:08.139Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-06T19:11:08.139Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-06T19:11:08.139Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:11:09.564Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-06T19:11:09.565Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-06T19:11:09.565Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [108,136,194,212,10,0,0,116,4,0,0,0] - 11
2020-04-06T19:11:09.566Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":34924,"extaddr":"0x00047400000*****","parentaddr":0}
2020-04-06T19:11:09.567Z zigbee-herdsman:controller:log Device '0x00047400000*****' joined
2020-04-06T19:11:09.568Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:11:09.568Z zigbee-herdsman:controller:log Device '0x00047400000*****' accepted by handler
2020-04-06T19:11:09.568Z zigbee-herdsman:controller:log Not interviewing '0x00047400000*****', completed 'false', in progress 'true'
2020-04-06T19:11:18.132Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000*****', attempt 3
2020-04-06T19:11:18.134Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-06T19:11:18.135Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-06T19:11:18.150Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-06T19:11:18.151Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-06T19:11:18.151Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-06T19:11:18.152Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-06T19:11:18.152Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:11:25.311Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
[... cut ...]
2020-04-06T19:11:48.157Z zigbee-herdsman:controller:error Interview failed for '0x00047400000***** with error 'Error: Interview failed because can not get node descriptor ('0x00047400000*****')'
zigbee2mqtt:error 2020-04-06 21:11:48: Failed to interview '0x00047400000*****', device has not successfully been paired
zigbee2mqtt:info  2020-04-06 21:11:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00047400000*****"}}'

Koenkk, 2020-04-06:
Pushed another possible fix, can you try again with latest dev?

npfp, 2020-04-06:
Got this:

`
zigbee2mqtt:info  2020-04-06 21:26:13: Starting zigbee2mqtt version 1.12.0-dev (commit #52d4193bc714a52a52e30801d2ab15d07695886b)

zigbee2mqtt:info  2020-04-06 21:27:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00047400000*****"}}'
2020-04-06T19:27:32.316Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-06T19:27:32.317Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-06T19:27:32.317Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [108,136,194,212,10,0,0,116,4,0,0,0] - 11
2020-04-06T19:27:32.318Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":34924,"extaddr":"0x00047400000*****","parentaddr":0}
2020-04-06T19:27:32.319Z zigbee-herdsman:controller:log Device '0x00047400000*****' joined
2020-04-06T19:27:32.319Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:27:32.320Z zigbee-herdsman:controller:log Device '0x00047400000*****' accepted by handler
2020-04-06T19:27:32.320Z zigbee-herdsman:controller:log Interview '0x00047400000*****' start
2020-04-06T19:27:32.329Z zigbee-herdsman:controller:device:log Interview - start device '0x00047400000*****'
2020-04-06T19:27:32.331Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-06T19:27:32.334Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
zigbee2mqtt:info  2020-04-06 21:27:32: Starting interview of '0x00047400000*****'
zigbee2mqtt:info  2020-04-06 21:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00047400000*****"}}'
2020-04-06T19:27:32.347Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-06T19:27:32.348Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-06T19:27:32.348Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-06T19:27:32.348Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-06T19:27:32.349Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-06T19:27:42.337Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000*****', attempt 1
2020-04-06T19:27:42.339Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-06T19:27:42.340Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-06T19:27:42.353Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-06T19:27:42.354Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-06T19:27:42.354Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-06T19:27:42.355Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-06T19:27:42.355Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
`

Koenkk, 2020-04-06:
I see you are using Hassio, this takes a little bit longer, try again in 4 hours from now.

npfp, 2020-04-06:
Ok, thx a lot!

npfp, 2020-04-07:
I'm afraid I still got an error despite the fact that I reinstalled the edge version several times. Is there a way to double check that the very last commit has been pulled by the hassio add-on?

Koenkk, 2020-04-08:
@npfp you can check with zigbee2mqtt 1.12.1, should be available soon for hassio.

npfp, 2020-04-09:
Just tested with 1.12.1

I'm afraid I still got this error:
```
2020-04-09T08:42:44.438Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-09T08:42:44.440Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-09T08:42:44.452Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-09T08:42:44.453Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-09T08:42:44.454Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-09T08:42:44.455Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-09T08:42:44.455Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-09T08:42:46.149Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-09T08:42:46.150Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-09T08:42:46.150Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [108,136,194,212,10,0,0,116,4,0,0,0] - 11
2020-04-09T08:42:46.151Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":34924,"extaddr":"0x00047400000*****","parentaddr":0}
2020-04-09T08:42:46.152Z zigbee-herdsman:controller:log Device '0x00047400000*****' joined
2020-04-09T08:42:46.152Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-09T08:42:46.153Z zigbee-herdsman:controller:log Device '0x00047400000*****' accepted by handler
2020-04-09T08:42:46.153Z zigbee-herdsman:controller:log Not interviewing '0x00047400000*****', completed 'false', in progress 'true'
2020-04-09T08:42:54.441Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '34924' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2020-04-09T08:42:54.442Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":34924,"options":0,"radius":30}
2020-04-09T08:42:54.444Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,108,136,0,30,158]
2020-04-09T08:42:54.458Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2020-04-09T08:42:54.458Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2020-04-09T08:42:54.458Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2020-04-09T08:42:54.459Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2020-04-09T08:42:54.459Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-09T08:42:56.885Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-09T08:42:56.886Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,108,136,194,212,10,0,0,116,4,0,0,0,11]
2020-04-09T08:42:56.887Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [108,136,194,212,10,0,0,116,4,0,0,0] - 11
2020-04-09T08:42:56.888Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":34924,"extaddr":"0x00047400000*****","parentaddr":0}
2020-04-09T08:42:56.888Z zigbee-herdsman:controller:log Device '0x00047400000*****' joined
2020-04-09T08:42:56.889Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-09T08:42:56.889Z zigbee-herdsman:controller:log Device '0x00047400000*****' accepted by handler
2020-04-09T08:42:56.889Z zigbee-herdsman:controller:log Not interviewing '0x00047400000*****', completed 'false', in progress 'true'
2020-04-09T08:42:57.462Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-09T08:42:57.463Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-09T08:42:57.476Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-09T08:42:57.476Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-09T08:42:57.477Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-09T08:42:57.477Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-09T08:42:57.477Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-09T08:43:07.468Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000*****', attempt 5
2020-04-09T08:45:52.291Z zigbee-herdsman:controller:device:log Interview - failed for device '0x00047400000*****' with error 'Error: Interview failed because can not get node descriptor ('0x00047400000*****')
    at Device.<anonymous> (/zigbee2mqtt-1.12.1/node_modules/zigbee-herdsman/dist/controller/model/device.js:330:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.1/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)'
2020-04-09T08:45:52.297Z zigbee-herdsman:controller:error Interview failed for '0x00047400000***** with error 'Error: Interview failed because can not get node descriptor ('0x00047400000*****')'
zigbee2mqtt:error 2020-04-09 10:45:52: Failed to interview '0x00047400000*****', device has not successfully been paired
zigbee2mqtt:info  2020-04-09 10:45:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00047400000*****"}}'```

Koenkk, 2020-04-09:
Can you try why holding it close to the coordinator?

npfp, 2020-04-09:
Trying very close and with 12.2 but still got issues: 

```
2020-04-09T20:14:47.397Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000*****', attempt 4
2020-04-09T20:14:47.399Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":34924,"nwkaddrofinterest":34924}
2020-04-09T20:14:47.400Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,108,136,108,136,35]
2020-04-09T20:14:47.412Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-09T20:14:47.413Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-09T20:14:47.414Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-09T20:14:47.414Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-09T20:14:47.415Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

npfp, 2020-04-09:
CC2531 firmware zStack12 version:  20190608

swingerman, 2020-04-10:
Still having the issue with V1.12.2:

Log
2020-04-10T15:31:57.929Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":3515,"nwkaddrofinterest":3515}
2020-04-10T15:31:57.930Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,187,13,187,13,35]
2020-04-10T15:31:57.941Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-10T15:31:57.941Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-10T15:31:57.941Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-10T15:31:57.942Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-10T15:31:57.942Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-10T15:32:07.935Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '3515' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2020-04-10T15:32:07.936Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":3515,"options":0,"radius":30}
2020-04-10T15:32:07.936Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,187,13,0,30,204]
2020-04-10T15:32:07.947Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2020-04-10T15:32:07.947Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2020-04-10T15:32:07.947Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2020-04-10T15:32:07.947Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2020-04-10T15:32:07.947Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-10T15:32:10.952Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":3515,"nwkaddrofinterest":3515}
2020-04-10T15:32:10.953Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,187,13,187,13,35]
2020-04-10T15:32:10.963Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-10T15:32:10.963Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-10T15:32:10.964Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-10T15:32:10.964Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-10T15:32:10.965Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-10T15:32:20.953Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000adb80', attempt 3
2020-04-10T15:32:20.958Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":3515,"nwkaddrofinterest":3515}
2020-04-10T15:32:20.959Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,187,13,187,13,35]
2020-04-10T15:32:20.969Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-10T15:32:20.970Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-10T15:32:20.970Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-10T15:32:20.970Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-10T15:32:20.970Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-10T15:32:30.967Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '3515' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2020-04-10T15:32:30.968Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":3515,"options":0,"radius":30}
2020-04-10T15:32:30.968Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,187,13,0,30,204]
2020-04-10T15:32:30.979Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2020-04-10T15:32:30.979Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2020-04-10T15:32:30.979Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2020-04-10T15:32:30.980Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2020-04-10T15:32:30.980Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-10T15:32:33.982Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":3515,"nwkaddrofinterest":3515}
2020-04-10T15:32:33.983Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,187,13,187,13,35]
2020-04-10T15:32:33.990Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-10T15:32:33.991Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-04-10T15:32:33.991Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-04-10T15:32:33.991Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-04-10T15:32:33.992Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-04-10T15:32:43.982Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00047400000adb80', attempt 4
2020-04-10T15:32:43.984Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":3515,"nwkaddrofinterest":3515}
2020-04-10T15:32:43.985Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,187,13,187,13,35]
2020-04-10T15:32:43.998Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-04-10T15:32:43.999Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
[... cut ...]
2020-04-10T15:33:30.045Z zigbee-herdsman:controller:error Interview failed for '0x00047400000adb80 with error 'Error: Interview failed because can not get node descriptor ('0x00047400000adb80')'
zigbee2mqtt:error 2020-04-10 18:33:30: Failed to interview '0x00047400000adb80', device has not successfully been paired
zigbee2mqtt:info  2020-04-10 18:33:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00047400000adb80"}}'
zigbee2mqtt:debug 2020-04-10 18:34:35: Saving state to file /share/zigbee2mqtt/state.json

Koenkk, 2020-04-11:
I'm not sure why the device doesn't respond to the requests, could you sniff the traffic? Maybe this will give us more information why it happens. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

swingerman, 2020-04-11:
I will try if as soon as I will have some time

npfp, 2020-04-11:
Same here. In the meantime, here are a some info I found while looking for a solution:

- Some cluster information

EndPoint: 0x01
Profile ID: 0x0104 (ZigBee HA)
Device ID: 0x010A (Unknown)
Input Cluster Count: 7
Cluster 0: Cluster ID: 0x0000 (General: Basic)
Cluster 1: Cluster ID: 0x0003 (General: Identify)
Cluster 2: Cluster ID: 0x0004 (General: Groups)
Cluster 3: Cluster ID: 0x0006 (General: On/Off)
Cluster 4: Cluster ID: 0x0005 (General: Scenes)
Cluster 5: Cluster ID: 0xFC01 (Unknown)
Cluster 6: Cluster ID: 0x0B04 (Unknown)
Output Cluster Count: 3
Cluster 0: Cluster ID: 0x0000 (General: Basic)
Cluster 1: Cluster ID: 0xFC01 (Unknown)
Cluster 2: Cluster ID: 0x0019 (General: OTA)

 0x0B04 corresponds to the power information of the power socket.
0xFC01 corresponds to Legrand's cluster.

https://faire-ca-soi-meme.fr/domotique/2018/09/24/test-legrand-celiane-by-netatmo-zigate/#Clusters_de_la_prise_pilotee

In this [thread](https://github.com/fairecasoimeme/ZiGate/issues/59), there is this image
https://user-images.githubusercontent.com/20152487/46192854-b7714100-c2fc-11e8-8bca-dc787b9d3308.png

and they also mention something that there is some kind of start time, and that the gateway and the socket should their power should be turn on at the same time, so they have the same start-time. https://github.com/fairecasoimeme/ZiGate/issues/59#issuecomment-430029496

In this [thread](https://github.com/KiwiHC16/Abeille/issues/243#issuecomment-509670637), they refer to 

dresden-elektronik/deconz-rest-plugin#883

> 	<cluster id="0xfc01" name="Legrand - Specific clusters" mfcode="1021">
> 		<description>Legrand Specific clusters.</description>
> 		<server>
> 			<attribute id="0x0000" name="Dimmer" type="dat16" default="0x0101" access="rw" required="m" showas="hex">
> 				<description>0100 = Dimmer Off, 0101 = Dimmer On</description>
> 			</attribute>
> 			<attribute id="0x0001" type="bool" name="LED" required="m" access="rw" default="0">
> 				<description>Enable LED in dark</description>
> 			</attribute>
> 		</server>
> 		<client>
> 		</client>
> 	</cluster>
> if you use it "out of the box" it works without problem but I don't think dimmer will work.
> if you update its firmware with the gateway, it will be harder to pair, you need to use another device as router to make it possible, I have used an ikea bulb.
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/pull/962/files
> You need to use the attribute in special cluster to enable dimmer

npfp, 2020-04-11:
In case, it helps, here are some sniffs on this thread:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-431605330

npfp, 2020-04-28:
> I'm not sure why the device doesn't respond to the requests, could you sniff the traffic? Maybe this will give us more information why it happens. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

I'm trying to sniff the traffic, but so far I can't make it works:  the CC2531 USB sniffer is not found by the system:

`ls: cannot access '/dev/ttyACM0': No such file or directory`

even though:
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 005: ID 0451:16ae Texas Instruments, Inc.
```

npfp, 2020-04-28:
I managed to make it work on macOS. But wireshark would only give me IEEE 802.15.4 packets

npfp, 2020-05-20:
I got this kind of frame. It doesn't really make sense to me so I don't know if they are relevant...

```
Frame 19: 10 bytes on wire (80 bits), 10 bytes captured (80 bits) on interface -, id 0
    Interface id: 0 (-)
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan  1, 1970 01:12:07.656776000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 727.656776000 seconds
    [Time delta from previous captured frame: 5.909225000 seconds]
    [Time delta from previous displayed frame: 5.909225000 seconds]
    [Time since reference or first frame: 718.831334000 seconds]
    Frame Number: 19
    Frame Length: 10 bytes (80 bits)
    Capture Length: 10 bytes (80 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan]

IEEE 802.15.4 Command, Dst: Broadcast
    Frame Control Field: 0x0803, Frame Type: Command, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: None
        .... .... .... .011 = Frame Type: Command (0x3)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .0.. .... = PAN ID Compression: False
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        00.. .... .... .... = Source Addressing Mode: None (0x0)
    Sequence Number: 179
    Destination PAN: 0xffff
    Destination: 0xffff
    Command Identifier: Beacon Request (0x07)
    FCS: 0xe737 (Correct)
```

or
```
Frame 23: 10 bytes on wire (80 bits), 10 bytes captured (80 bits) on interface -, id 0
    Interface id: 0 (-)
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan  1, 1970 01:12:19.473913000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 739.473913000 seconds
    [Time delta from previous captured frame: 0.298432000 seconds]
    [Time delta from previous displayed frame: 0.298432000 seconds]
    [Time since reference or first frame: 730.648471000 seconds]
    Frame Number: 23
    Frame Length: 10 bytes (80 bits)
    Capture Length: 10 bytes (80 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan]

IEEE 802.15.4 Command
    Frame Control Field: 0x0853, Frame Type: Command, Frame Pending, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: None
        .... .... .... .011 = Frame Type: Command (0x3)
        .... .... .... 0... = Security Enabled: False
        .... .... ...1 .... = Frame Pending: True
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        00.. .... .... .... = Source Addressing Mode: None (0x0)
    Sequence Number: 185
    [Expert Info (Error/Malformed): Invalid Setting for PAN ID Compression]
        [Invalid Setting for PAN ID Compression]
        [Severity level: Error]
        [Group: Malformed]```

```Frame 31: 11 bytes on wire (88 bits), 11 bytes captured (88 bits) on interface -, id 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan  1, 1970 01:12:48.718740000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 768.718740000 seconds
    [Time delta from previous captured frame: 5.611800000 seconds]
    [Time delta from previous displayed frame: 5.611800000 seconds]
    [Time since reference or first frame: 759.893298000 seconds]
    Frame Number: 31
    Frame Length: 11 bytes (88 bits)
    Capture Length: 11 bytes (88 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan]

IEEE 802.15.4 Command, Dst: Broadcast, Bad FCS
    Frame Control Field: 0x0803, Frame Type: Command, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: None
        .... .... .... .011 = Frame Type: Command (0x3)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .0.. .... = PAN ID Compression: False
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        00.. .... .... .... = Source Addressing Mode: None (0x0)
    Sequence Number: 199
    Destination PAN: 0xffff
    Destination: 0xffff
    Command Identifier: Beacon Request (0x07)
    FCS: 0x0000 (Incorrect, expected FCS=0x0032)
    [Expert Info (Warning/Checksum): Bad FCS]
        [Bad FCS]
        [Severity level: Warning]
        [Group: Checksum]
```

npfp, 2020-05-20:
```
Frame 44: 10 bytes on wire (80 bits), 10 bytes captured (80 bits) on interface -, id 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan  1, 1970 01:13:47.811364000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 827.811364000 seconds
    [Time delta from previous captured frame: 11.518811000 seconds]
    [Time delta from previous displayed frame: 11.518811000 seconds]
    [Time since reference or first frame: 818.985922000 seconds]
    Frame Number: 44
    Frame Length: 10 bytes (80 bits)
    Capture Length: 10 bytes (80 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:data]

IEEE 802.15.4 Ack, Sequence Number: 229, Dst: Broadcast, Bad FCS
    Frame Control Field: 0x0802, Frame Type: Ack, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: None
        .... .... .... .010 = Frame Type: Ack (0x2)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .0.. .... = PAN ID Compression: False
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        00.. .... .... .... = Source Addressing Mode: None (0x0)
    Sequence Number: 229
    Destination PAN: 0xffff
    Destination: 0xffff
    FCS: 0x0000 (Incorrect, expected FCS=0x3e13)
    [Expert Info (Warning/Checksum): Bad FCS]
        [Bad FCS]
        [Severity level: Warning]
        [Group: Checksum]
```

npfp, 2020-05-20:
or

```
Frame 47: 68 bytes on wire (544 bits), 68 bytes captured (544 bits) on interface -, id 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan  1, 1970 01:14:00.226410000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 840.226410000 seconds
    [Time delta from previous captured frame: 6.207570000 seconds]
    [Time delta from previous displayed frame: 6.207570000 seconds]
    [Time since reference or first frame: 831.400968000 seconds]
    Frame Number: 47
    Frame Length: 68 bytes (544 bits)
    Capture Length: 68 bytes (544 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan]

IEEE 802.15.4 Reserved
    Frame Control Field: 0x0844, Frame Type: Reserved, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: None
        .... .... .... .100 = Frame Type: Reserved (0x4)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        00.. .... .... .... = Source Addressing Mode: None (0x0)
    Sequence Number: 237
    [Expert Info (Error/Malformed): Invalid Setting for PAN ID Compression]
        [Invalid Setting for PAN ID Compression]
        [Severity level: Error]
        [Group: Malformed]

0000   44 08 ed ff ff bf ff 07 f4 9e e4 29 30 79 02 e2   D..........)0y..
0010   c9 a6 6b c5 fb 52 4f 7f 2f f1 03 b9 30 f0 76 14   ..k..RO./...0.v.
0020   35 a3 c9 7e 48 a3 83 ef 0d 54 6d 46 65 b0 ad 52   5..~H....TmFe..R
0030   6e 49 27 82 dd 20 79 4b 2c c8 ce f0 31 5f 7e 4a   nI'.. yK,...1_~J
0040   ca 49 9f 90                                       .I..
```

sergeolkhovik, 2020-05-25:
Gentlemen, got the same thing, trying to pair my z2m/CC2531 stick and Aqara magic cube. Error is completely the same:
```
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":18892,"options":0,"radius":30} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,204,73,0,30,255] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":18892,"nwkaddrofinterest":18892} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,204,73,204,73,35] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0002a6fe85', attempt 6 +23s
  zigbee-herdsman:controller:device:log Interview - failed for device '0x00158d0002a6fe85' with error 'Error: Interview failed because can not get node descriptor ('0x00158d0002a6fe85')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:330:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +2ms
  zigbee-herdsman:controller:error Interview failed for '0x00158d0002a6fe85 with error 'Error: Interview failed because can not get node descriptor ('0x00158d0002a6fe85')' +0ms
zigbee2mqtt:error 2020-05-25 15:17:33: Failed to interview '0x00158d0002a6fe85', device has not successfully been paired
```

My environment: CC2531 (finally got it visible in my Intel NUC box after flashing 1.2 DEFAULT), Intel NUC/Ubuntu 20.04 and HA 0.110.1. I connected stick via USB wire and directly to front panel, doesn't matter - error is still the same, node v13.14, z2m v1.13.1 (commit 3046680).

I'm new to zigbee protocol and want to get cube working, have no other device to test. Please let me know if you need any extra info from me.

sergeolkhovik, 2020-05-25:
I tried both 1.2 source routing and z2m dev, neither worked, the problem is still the same.

npfp, 2020-06-20:
Hi @Koenkk, I just wonder if the frames I provided are enough to find a fix or should I provide more?

Thx!

Koenkk, 2020-06-21:
@npfp could you provide me the whole pcapng file?

npfp, 2020-06-21:
Hi,

Attached. Please let me know if I can be of any help.

Thx

npfp, 2020-06-21:
[first_try.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/4809836/first_try.pcapng.zip)

Koenkk, 2020-06-24:
Somehow all commands are corrupted or show bad FCS, this could be when the sniffer is too far from the coordinator. Can you check if you can fix this?

npfp, 2020-06-27:
Is it better with this one:

npfp, 2020-06-27:
[second_try.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4841176/second_try.pcapng.gz)

npfp, 2020-06-27:
Another :

[third_try.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4841188/third_try.pcapng.gz)

Koenkk, 2020-06-29:
The bad FCS errors are gone but no commands are shown, are you sure you are sniffing on the correct channel? (the channel were zigbee2mqtt is running on)

npfp, 2020-06-30:
It should be on the right channel. Attached another try:

[fourth_try.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4854083/fourth_try.pcapng.gz)

Koenkk, 2020-07-04:
@npfp on what channel is zigbee2mqtt running? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-647183422 shows channel 11 is required for this device.

cemonneau, 2020-07-05:
Hi guys (not fully back, see main thread). 
From what I'm seeing (will look more closely) it's missing an answer for their custom frames. I did not had troubles my power plug.
Requirements for all legrand devices: 
- channel 11
- custom frame (that is implemented in zigbee2mqtt) to allow the devices to pair : sometimes it's asked before the pairing, sometimes after, if it does not respond: it either leaves or does not pair.

@npfp  Would you have a file from wireshard (log) when you try to pair the device so I could check wat's going on ?

Also was the product ever paired to the official gateway (if that's the case It could have been updated)

npfp, 2020-07-05:
> Also was the product ever paired to the official gateway (if that's the case It could have been updated)m

Hi,
 
Thx for the answers!

My zibee2mqtt is indeed on channel 11 and I tried to sniff on this channel. 

The products never paired with the official gateway. 

I suppose the custom pair is missing somehow (even thought I don't fully understand this part ;)

Also, I tried to reset and switch on all devices (I tried 2 different plugs) and z2m about the same time but it didn't have any effect. 

The logs were taken while switching on/resetting the plugs. 

I've got only one CC2531 so I'm not able to sniff while trying to pair not with a transmitter as advised in the other thread. 

I'm buying another stick and I'll try sniffing as soon as I'll get it. 

Thx

npfp, 2020-07-18:
Hi @cemonneau @Koenkk,

Here are more logs where the legrand devices appear and communicate with the controller.

[sniff.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4941846/sniff.pcapng.gz)

cemonneau, 2020-07-19:
Hi @npfp 
For the first look I took at the data it seems the socket is not answering anything to any request. It receives the transport key, then the coordinator is asking for the node descriptor, and nothing is sent back.
The device is also not asking for anything, that's the weird part. Every device I have asked something : this one no.
I will ask to other projects if they have any idea but currently I have none.

Will report if I find/get something.

By any chance, would you have the official Legrand gateway and would be able to try to pair it AND at the same time log the frames ?

npfp, 2020-07-20:
Thx for the look! Also, the led stays red and only blinks once when pushing the reset button.

Unfortunately I don’t have the official Legrand gateway, I’ll try to get one if it can help.

closed 2020-08-28 by github-actions[bot]

swingerman, 2021-08-16:
Just wanted to share that I solved the issue by changing my cc2531 Zigbee stick to more powerful CC2652RB stick.