Koenkk/zigbee2mqtt #3022: Status of Group with Tradfri Remote

R-Rosenow, 2020-02-28:
# Bug Report
## What happened
When I switch my Lights on with the Ikea E1810 Remote (via Group), then Zigbee2Mqtt don't send a State to my OpenHab Server. The remote has the update via OTA, but the solution for binding at groups don't work. It just work with the classic work with the original group ID of the remote.

Is there any Solution?

In the MQTT Messages via Zigbee2Mqtt.service nothing happen. Just the message of the toggle from the remote.

## What did you expect to happen
I want to see wenn my lights are on and off (Send a State when change).

## Debug Info
zigbee2mqtt version: latest release
CC1352P firmware version: latest firmware

Koenkk, 2020-02-29:
For this you need to enable the report feature: https://www.zigbee2mqtt.io/information/report.html

R-Rosenow, 2020-02-29:
Thank you! Exactly what I searching for.

closed 2020-02-29 by R-Rosenow