Koenkk/zigbee2mqtt #3023: Default to start npm

Torr79, 2020-02-28:
Hello,
I have an installation with zigbee2mqtt a raspberry and a sniffer cc2531 and several temperature probes and xiaomi aperture sensors.
Since an update, nothing works, I can't find a solution to solve this problem, could you help me?
thank you.
Here is the log when I launch npm start manually:

pi@raspberrypi:/opt/zigbee2mqtt $ sudo npm start

> zigbee2mqtt@1.10.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-02-28 16:17:54: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-02-28.16-17-54' filename: log.txt
zigbee2mqtt:info  2020-02-28 16:17:56: Starting zigbee2mqtt version 1.10.0 (commit #8df6be8)
zigbee2mqtt:info  2020-02-28 16:17:56: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-28 16:17:56: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-28 16:17:56: Failed to start zigbee
zigbee2mqtt:error 2020-02-28 16:17:56: Exiting...
zigbee2mqtt:error 2020-02-28 16:17:56: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.serialPort.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at binding.open.then.err (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-28T15_17_56_962Z-debug.log

mr2c12, 2020-02-28:
Given this line:
zigbee2mqtt:error 2020-02-28 16:17:56: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

You should check your serial port, the permission on that and your configuration file.

Torr79, 2020-02-29:
thank you for your reply
I also think that the problem stems from this, but I cannot find a solution.

the command: ls -l / dev / serial / by-id
gives me the following result: total 0
lrwxrwxrwx 1 root root Feb 13 29 12:04 usb-Texas_Instruments_TI_CC2531_USB_CDC ___ 0X00124B0018E20A02-if00 -> ../../ttyACM1

And my YAML file is as follows:
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt: // localhost: 1883'
  user: domoticz
  password: ####
serial:
  port: / dev / ttyACM1
devices:
  '0x00158d00022b6ba5':
    friendly_name: bathroom
  '0x00158d00022c9485':
    friendly_name: children_room
  '0x00158d0002d79488':
    friendly_name: living_room
  '0x00158d000204bdc7':
    friendly_name: porte_entree
  '0x00158d0001fa755a':
    friendly_name: '0x00158d0001fa755a'
  '0x00158d0001bbe8bb':
    friendly_name: water_sensor
  '0x00158d000272b53b':
    friendly_name: cellar
  '0x00158d0003121241':
    friendly_name: 'porte_placard'

Thanks for your help as I really can't find a solution to this problem.

Koenkk, 2020-02-29:
> Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

Means that another application is already occupying the serial port. You should be able to find the application using `ls -l /proc/[0-9]/fd/ |grep /dev/ACM1`

Torr79, 2020-02-29:
Thank you for your response, when I type this command, here is the return:

pi@raspberrypi:~ $ ls -l /proc/[0-9]/fd/ |grep /dev/ACM1
ls: impossible d'ouvrir le répertoire '/proc/1/fd/': Permission non accordée
ls: impossible d'ouvrir le répertoire '/proc/2/fd/': Permission non accordée
ls: impossible d'ouvrir le répertoire '/proc/3/fd/': Permission non accordée
ls: impossible d'ouvrir le répertoire '/proc/4/fd/': Permission non accordée
ls: impossible d'ouvrir le répertoire '/proc/8/fd/': Permission non accordée
ls: impossible d'ouvrir le répertoire '/proc/9/fd/': Permission non accordée

and this way here is the answer:
pi@raspberrypi:~ $ sudo ls -l /proc/[0-9]/fd/ |grep /dev/ACM1
pi@raspberrypi:~ $

And always the same problem when launching the npm start command.

Kryzek, 2020-02-29:
Seems like a permission issue (Permission non accordée). Check https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port for possible solution.

closed 2020-05-06 by stale[bot]