Koenkk/zigbee2mqtt #3025: Source routing coordinator with more than 5 direct children

mr2c12, 2020-02-29:
Coordinator: cc2530+cc2591
Fw: info  2020-02-26 19:42:08: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
Flashed fw with sourcerouting.

[image]

Given the network map above, it seems to me that my coordinator has more children than expected.
Is it correct?
From the fw readme, I read that my coordinator flashed with source routing should have only 5 direct connections. How can I verify this?
Is it possibile that something went wrong when flashing the firmware? I used to have the same version/release but without source routing. When z2m starts, the firmware version logged seems the same to me.

Koenkk, 2020-02-29:
I think the LQIs of 0 and 1 indicate there is not direct connection.

mr2c12, 2020-02-29:
Good to know!
Thanks!

closed 2020-02-29 by mr2c12