Koenkk/zigbee2mqtt #3026: IKEA LED1624G9 MAC channel access failure

nldroid, 2020-02-29:
# Bug Report
I have this one bulb IKEA LED1624G9 that doesn't want to play along. It often fails with a timeout. I thought it might be because it's the farthest away from the coordinator but the link quality is ok. The (battery powered) wall button from Xiaomi is in the same room and those clicks are registered. I also have a Xiaomi cube in the same room and those actions are registered too.

## What happened
Two days ago the lamp wouldn't respond to any action from zigbee2mqtt. It still worked with a ikea button that is binded to the bulb.

I got this error all the time:

```
Feb 28 06:55:37 nuc npm[10334]: zigbee2mqtt:error 2020-02-28 06:55:37: Publish 'set' 'state' to 'slaapkamer_light' failed: 'Error: Command 0xd0cf5efffe2268bb/1 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'
Feb 28 06:55:37 nuc npm[10334]: zigbee2mqtt:info  2020-02-28 06:55:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'slaapkamer_light' failed: 'Error: Command 0xd0cf5efffe2268bb/1 genOnOff.on({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"slaapkamer_light"}}'
```
I removed the bulb from the lamp so the power was off for a while. That didn't help.

Waited two days to see if it resolved itself because that happens sometimes with this bub. That didn't help.

I restarted zigbee2mqtt an that helped immediatly.

I blamed the lamp previously but because a restart of zigbee2mqtt fixed it immediatly made me think it might be a software/firmware bug after all.

## What did you expect to happen
The bulb not failing ;)

## How to reproduce it (minimal and precise)
Hard to reproduce. It works for a while and then it stops working.

## Debug Info
zigbee2mqtt version: 1.11-dev
CC26X2R1 firmware version: CC26X2R1_20191106

The lamp is upgraded to the latest version with the OTA feature.

Database entry:
`{"id":6,"type":"Router","ieeeAddr":"0xd0cf5efffe2268bb","nwkAddr":46875,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}}},"binds":[]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1582839546921}`

Link quality:
[image]

nldroid, 2020-02-29:
I've updated to 20200229. Will see if i still get dropouts.

Koenkk, 2020-02-29:
good, let me know.

nldroid, 2020-03-01:
No luck:

The first line below is from the battery powered wall switch. Range shouldn't be an issue.

```
Mar 01 13:49:59 nuc npm[12475]: zigbee2mqtt:info  2020-03-01 13:49:59: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":36,"battery":100,"voltage":3005,"click":""}'
Mar 01 13:50:10 nuc npm[12475]: zigbee2mqtt:error 2020-03-01 13:50:10: Publish 'set' 'state' to 'slaapkamer_light' failed: 'Error: Command 0xd0cf5efffe2268bb/1 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 46875 - 1 - 59 - 6 - 11 after 6000ms)'
Mar 01 13:50:10 nuc npm[12475]: zigbee2mqtt:info  2020-03-01 13:50:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'slaapkamer_light' failed: 'Error: Command 0xd0cf5efffe2268bb/1 genOnOff.on({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 46875 - 1 - 59 - 6 - 11 after 6000ms)'","meta":{"friendly_name":"slaapkamer_light"}}'
Mar 01 13:50:25 nuc npm[12475]: zigbee2mqtt:error 2020-03-01 13:50:25: Publish 'set' 'state' to 'slaapkamer_light' failed: 'Error: Command 0xd0cf5efffe2268bb/1 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 46875 - 1 - 60 - 6 - 11 after 6000ms)'
Mar 01 13:50:25 nuc npm[12475]: zigbee2mqtt:info  2020-03-01 13:50:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'slaapkamer_light' failed: 'Error: Command 0xd0cf5efffe2268bb/1 genOnOff.on({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 46875 - 1 - 60 - 6 - 11 after 6000ms)'","meta":{"friendly_name":"slaapkamer_light"}}'
```

When i use the 4 button Ikea swith binded to the lamp, it goes on.

Koenkk, 2020-03-02:
Now it seems to fail with a different error (timeout instead of `225`) error. 

Any chance you could sniff the traffic when the command fails? This should give us some insight in were things go wrong. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

nldroid, 2020-03-02:
I'll get the sniffer stick out of the drawer and will try to sniff the traffic when it fails again.

nldroid, 2020-03-14:
I'll close this issue for now. It worked 11 days without any problem. Yesterday morning the lamp wouldn't turn off so i thought to sniff the traffic in the evening. It workd again.

This morning it turned on from HA without any problem but it wouldn't turn off from HA. This time i had my sniffer ready but as soon as i ran wireshark, it worked again.

Nothing in the logs either. It looks like there's just some delay.

I'll try the latest dev with extra routing fixes. See if that helps.

closed 2020-03-14 by nldroid

nldroid, 2020-03-15:
Some more info about delays. I'm beginning to wonder if Home Assistant is to blame.

```
info  2020-03-15 08:15:52: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch/click', payload 'single'
info  2020-03-15 08:15:52: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":24,"battery":100,"voltage":3005,"click":"single"}'
info  2020-03-15 08:15:52: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":24,"battery":100,"voltage":3005,"click":""}'
info  2020-03-15 08:16:14: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch/click', payload 'single'
info  2020-03-15 08:16:14: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":21,"battery":100,"voltage":3005,"click":"single"}'
info  2020-03-15 08:16:14: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":21,"battery":100,"voltage":3005,"click":""}'
info  2020-03-15 08:16:51: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch/click', payload 'single'
info  2020-03-15 08:16:51: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":18,"battery":100,"voltage":3005,"click":"single"}'
info  2020-03-15 08:16:51: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch', payload '{"linkquality":18,"battery":100,"voltage":3005,"click":""}'

info  2020-03-15 08:17:24: MQTT publish: topic 'zigbee2mqtt/slaapkamer_lights', payload '{"state":"OFF","brightness":33,"color":{"x":0.467,"y":0.402}}'
info  2020-03-15 08:17:24: MQTT publish: topic 'zigbee2mqtt/slaapkamer_light', payload '{"state":"OFF","brightness":33,"color":{"x":0.467,"y":0.402},"update_available":false}'
info  2020-03-15 08:17:25: MQTT publish: topic 'zigbee2mqtt/slaapkamer_lights', payload '{"state":"ON","brightness":33,"color":{"x":0.467,"y":0.402}}'
info  2020-03-15 08:17:25: MQTT publish: topic 'zigbee2mqtt/slaapkamer_light', payload '{"state":"ON","brightness":33,"color":{"x":0.467,"y":0.402},"update_available":false}'
info  2020-03-15 08:17:30: MQTT publish: topic 'zigbee2mqtt/slaapkamer_lights', payload '{"state":"OFF","brightness":33,"color":{"x":0.467,"y":0.402}}'
info  2020-03-15 08:17:30: MQTT publish: topic 'zigbee2mqtt/slaapkamer_light', payload '{"state":"OFF","brightness":33,"color":{"x":0.467,"y":0.402},"update_available":false}'
```
First click on 08:15:52, nothing happens
Second click 22 seconds later on  08:16:14, nothing happens
Third click 37 seconds later on 08:16:51 and still a wait before something happens

at 08:17:24 an OFF command (from a toggle action) is sent from Home Assistant. And that's strange because the bulb already was off. Home Assistant probably didn't know that because i had to switch it off last night with a directly binded switch.

One second later at 08:17:25 an ON command is sent from Home Assistant

And then 5 seonds later at 08:17:30 another OFF command is sent from Home Assistant

In Wireshark with this filter:
`
!(zbee_nwk.cmd.id == 0x01) && !(zbee_nwk.cmd.id == 0x08) && !(zbee_nwk.cmd.id == 0x05) && !(zbee_nwk.cmd.id == 0x02) && !(wpan.frame_type == 0x2)  && !(zbee_zcl.cmd.id == 0x0b)`

shows:

[image]

Timelines are mostly in line with the zigbee2mqtt log.

What's a bit strange is that the first OFF in the log is at 08:17:**24** while wireshark shows it at 08:17:**14**. It probably logs it on the Ack.

nldroid, 2020-03-15:
Reload of automations in Home Assistant recovered responsiveness for click of the button..

Koenkk, 2020-03-15:
To verify this, you should enable the debug logging (as then the incoming MQTT messages are logged). I would expect that immediately after Zigbee2mqtt publishes `info  2020-03-15 08:15:52: MQTT publish: topic 'zigbee2mqtt/slaapkamer_wall_switch/click', payload 'single'` you should see a received MQTT message.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

nldroid, 2020-03-15:
Done that now. 

Testing all day now and it seems that https://www.home-assistant.io/integrations/device_trigger.mqtt/ is not stable. It looks like a Home Assistant bug and not a Zigbee2mqtt bug. 

Reverted to old script:

```
- id: slaapkamer_wall_switch_click
  alias: Slaapkamer lamp toggle
  trigger:
    platform: mqtt
    topic: zigbee2mqtt/slaapkamer_wall_switch
  condition:
    condition: template
    value_template: '{{ "single" == trigger.payload_json.click }}'
  action:
    service: light.toggle
    entity_id: light.slaapkamer_light
```
See what happens.