Koenkk/zigbee2mqtt #3027: Busch Jaeger Light Link buttons not supported

Simanias, 2020-02-29:
The Busch Jaeger Zigbee Light Link power supply/relay/dimmer (6735/6736/6737) is supported and recognized. However I have the 6737, which has 8 buttons of which the top two are only supported as a switch. The other buttons just keep repeating the last sent command if pressed. Is there a way to make those buttons function independent? 

[image]

Regards,
Simania

Koenkk, 2020-02-29:
Please provide more context of the issue with debug logging of the good an bad behaviour.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Simanias, 2020-02-29:
@Koenkk 
```
zigbee2mqtt:debug 2020-02-29 20:22:07: Received Zigbee message from '0xd85def11a100162e', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 18 with groupID 0
zigbee2mqtt:info  2020-02-29 20:22:07: MQTT publish: topic 'zigbee2mqtt/0xd85def11a100162e', payload '{"state":"OFF","linkquality":89}'
```
Thiss mesaage keeps repeating, I was thinking it had something to do with the button I pressed, but it seems its just a message about the state and the linkquality.

Simanias, 2020-03-01:
With the new version I get a different message:
`zigbee2mqtt:info  2020-03-01 22:28:13: MQTT publish: topic 'zigbee2mqtt/0xd85def11a100162e', payload '{"state":"OFF","linkquality":81,"action":"row_1_off"}'`
The "row_1_off action is new, however this is repeated at every press of another button, eg. row 2 right button.

Koenkk, 2020-03-02:
Can you provide the debug log when pressing all buttons and indicate what each buttons generated?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

jeppper, 2020-03-10:
I have just paired one of these 6737 buttons successfully but get no mqtt messages at all when pressing any of the buttons.

Koenkk, 2020-03-10:
@jeppper can you share your `database.db`?

jeppper, 2020-03-11:
Yes of course :-)
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/4318693/database.txt)

Koenkk, 2020-03-11:
@jeppper your device is not supported at the moment.

To add support for it, could you try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L5525 to `zigbeeModel: ['RM01', 'RB01'],`. After this repair the device.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `devices.js`.

jeppper, 2020-03-15:
I guess I have to have a little more help doing this.
I have tried to edit my node_modules/zigbee-herdsman-converters/devices.js file in Filezilla, but it returns a syntax error.

Koenkk, 2020-03-15:
If you send me your `devices.js` file (post it on pastebin.com) I can make the modifications and send you a new one.

jeppper, 2020-03-17:
It should be on pastebin now.
User: Jeppper and file devices.js

Koenkk, 2020-03-17:
I'm not sure how to search on pastebin, can you provide the link?

jeppper, 2020-03-17:
I have used pastebin.com for the first time for this.
When I copy the URL and paste it in a browser without being logged in nothing comes up??

Koenkk, 2020-03-17:
Could you otherwise send it to me on telegram (@koenkk)?

jeppper, 2020-03-17:
Could I just attach it here?

Koenkk, 2020-03-17:
Yes

jeppper, 2020-03-17:
Fine ☺

jeppper, 2020-03-17:
I think it was blocked. I try to zip it.

jeppper, 2020-03-21:
Hi

Did you receive the compressed file?

Best regards
Jesper

I think it was blocked. I try to zip it.

Koenkk, 2020-03-21:
No, but I've added it in the latest dev branch now. Can you check if it works there? https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

jeppper, 2020-03-25:
I normally use when updating to the latest branch:
"git pull
rm -rf node_modules
npm install"

You say:
"git pull
npm ci"

Which is more correct?

Koenkk, 2020-03-26:
``` 
git pull
npm ci
```

(although there is no big difference)

jeppper, 2020-03-28:
I have now updated the dev branch but still getting the same result. No mqtt messages when pressing any of the buttons.
Maybe I am not doing the pairing correct. There is no problem in adding the device to zigbee2mqtt, but I am not quite sure if I have to do more with the buttons afterwards.

Koenkk, 2020-03-28:
Can you provide the debug log of Zigbee2mqtt startup and when pressing buttons?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

jeppper, 2020-03-29:
Yes of course.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4398482/log.txt)

I can see no change in the log when pressing any of the buttons, but as I said there might maybe be more to pairing than I did.

Koenkk, 2020-03-29:
Thanks, can you also share your `database.db`?

jeppper, 2020-03-30:
Yes :-)

[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/4404784/database.zip)

Koenkk, 2020-03-31:
@jeppper thanks, I did another attempt, please checkout with the latest dev branch again in +- 4 hours from now. Repairing might be necessary, if it doesn't work provide your `database.db` again.

jeppper, 2020-04-02:
I updated but no change unfortuneately. database is attached.
There are no messages from the device.
I know when using it together with a Philips Hue bridge, one has to perform a Network scan after pairing it with the bridge. With this scan it finds available lamps on the Network. Finally you can make scenes including these founds lamps.
I don´t know if this could have an influence here.
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/4419039/database.zip)

Koenkk, 2020-04-02:
It looks that the binds are successfully setup now. That makes it strange that no messages can be seen yet. Could you provide the log when running with herdsman debug logging and pressing the buttons? 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

jeppper, 2020-04-04:
Do I just put "DEBUG=zigbee-herdsman* npm start" in putty?
When doing this I just get some npm errors.

Simanias, 2020-04-17:
Sorry to have no joined the issue for so long.
Since I'm running zigbee2MQTT via home assistant as a docker, I think I'm not able to use the dev branch to do some debugging, is that right?
Still I appreciate all your hard work.

Koenkk, 2020-04-17:
@Simanias you can use the edge version of the addon and set `zigbee_herdsman_debug: true`.

jeppper, 2020-06-22:
Was this ever solved?

Simanias, 2020-06-28:
> Was this ever solved?

No not yet. I have not used the edge version to debug, because when I look at the date of the files, they have not changed for 4 to 9 months...... I wonder if they are up-to-date

jeppper, 2020-07-29:
Any news regarding this?

Simanias, 2020-08-31:
Just to pick up on the issue, I made the switch to the debug version and I was looking in the devices.js and found the following passage:

`        // In order to manually capture scenes as described in the devices manual, the endpoint
        // corresponding to the row needs to be unbound (https://www.zigbee2mqtt.io/information/binding.html)
        // If that operation was successful, the switch will respond to button presses on that
        // by blinking multiple times (vs. just blinking once if it's bound).`

The rows are unbound because when I press one of the buttons, the light blinks. But I don't get any message in MQTT, is this normal,eg. how do I know how the 8-buttons are named?
This passage didn't gave me the required information:

`            // Depending on the actual devices - 6735, 6736, or 6737 - there are 1, 2, or 4 endpoints.`

  `                  // The total number of bindings seems to be severely limited with these devices.
                    // In order to be able to toggle groups, we need to remove the scenes cluster
  `

Anything I could do to help this problem to sort out @Koenkk ?

Simanias, 2020-09-19:
@Koenkk I'm still hopefully to make this switch work. I've switched to the 'edge' build, but still nothing to be seen in the log. Do you have any tips?

Koenkk, 2020-09-19:
@docbobo could you help with this?

Simanias, 2020-09-29:
@docbobo any chance you could help us to make this switch work?

wwijn, 2020-10-10:
@Koenkk Any update on this?
I'm currently planning my villa renovation and I'm looking for a Zigbee Light Link switch to implement throughout the house.
Gira and Jung switches are discontinued as I understand. The Busch-Jaeger one is still an option, provided it's fully supported by zigbee2mqtt.
If not; any other suggestions for modern - integrated in commercial switchgear - zigbee enabled switches with multiple buttons?
Preferably a/c powered, not battery powered...

Thanks for all your efforts! 👍

Koenkk, 2020-10-10:
@wwijn I don't plan to work on this, I hope someone who owns the device wants to help.

wwijn, 2020-10-10:
@Simanias @jeppper 
Did you get this working?
I’m considering these Busch jaeger switches, but only when we can get them
to work... thanks!

jeppper, 2020-10-10:
Hi
No I never got it working, unfortuneately.
It seems that it pairs all right but nothing happens when pressing the buttons.

Med Venlig Hilsen / Best Regards / Mit Freundlichen Grüssen

Jesper Marschner

wwijn, 2020-10-10:
@Koenkk 
Koen, my new home is in Helmond (Stiphout) haha very near :) 
Can I convince you to spend some time
on this? I can sponsor you a device.. I can also sponsor you some time invested.
I’m running against the wall deciding how to gear my house with zigbee enabled switches combined with traditional switch gear of an a brand like Gira, Jung, Busch jaeger, etc
I’m planning on using your zigbee2mqtt and home assistant to build my smart new home.
Thanks for your work and hope to have a road to a solution here.

Koenkk, 2020-10-11:
@wwijn fun to hear! For your use case I would recommend a different kind of switch which can be put behind a traditional switch. In my home I have a lot of Shelly 1's installed (WiFi, because at that time such Zigbee switches were too expensive). However nowadays there seem to be a lot of options:
- https://www.zigbee2mqtt.io/devices/ZG102-BOX-UNIDIM.html (dimmer)
- https://www.zigbee2mqtt.io/devices/QS-Zigbee-S04-2C-LN.html
- https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-2C-LN.html (dimmer)
- https://www.zigbee2mqtt.io/devices/12126.html

If you look at https://www.zigbee2mqtt.io/information/supported_devices.html you will find many similar looking devices. There are also options which don't require a neutral wire: https://nl.aliexpress.com/item/4001178298316.html?spm=a2g0o.productlist.0.0.4cf6472aZ1aaTy&algo_pvid=605b6866-467d-47a1-a3e6-ca0afd09558b&algo_expid=605b6866-467d-47a1-a3e6-ca0afd09558b-0&btsid=0b0a119a16024194548637488ec292&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_ (also supported)

closed 2020-11-18 by github-actions[bot]