Koenkk/zigbee2mqtt #3028: Set sensitivity of the Aqara Vibration sensor

vanlooverenkoen, 2020-02-29:
In the documentation we can see that we can set the sensitivity of the Aqara vibration/shock sensor
https://www.zigbee2mqtt.io/devices/DJT11LM.html

My topic
`zigbee2mqtt/0x00158d00039da3a9/set`
My Payload
`{"sensitivity": "high"}`

This is the error I recieve:
```
zigbee2mqtt:error 2020-02-29 12:27:41: Publish 'set' 'sensitivity' to '0x00158d00039da3a9' failed: 'Error: Write 0x00158d00039da3a9/1 genBasic({"65293":{"value":1,"type":32}}, {"timeout":6000,"manufacturerCode":4447,"disableDefaultResponse":true}) failed (Error: AREQ - AF - dataConfirm after 3000ms)'
zigbee2mqtt:info  2020-02-29 12:27:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'sensitivity' to '0x00158d00039da3a9' failed: 'Error: Write 0x00158d00039da3a9/1 genBasic({\"65293\":{\"value\":1,\"type\":32}}, {\"timeout\":6000,\"manufacturerCode\":4447,\"disableDefaultResponse\":true}) failed (Error: AREQ - AF - dataConfirm after 3000ms)'","meta":{"friendly_name":"0x00158d00039da3a9"}}'```

Koenkk, 2020-02-29:
You need to wakeup the sensor right before executing this command (to wake it up press the button the sensor)

vanlooverenkoen, 2020-03-02:
I am trying to use the vibration sensor to detect when a door is opened/  moved. I first wanted to use a door sensor but what if you do not completely close the door. I want to detect motion on the door. But the sensor most of the time not triggered. I thought that the sensitivity would work. No success at this point

Koenkk, 2020-03-03:
Fixed in the latest dev branch.

Note that after setting the sensitivity you immediately have to start pressing the reset button with an interval of 1 second until you see Zigbee2mqtt publishing the new sensitivity to MQTT. (same as what the official Xiaomi gateway shows):

[image]

closed 2020-03-03 by Koenkk

vanlooverenkoen, 2020-03-07:
This is what the sonsor returns. Is this correct?

When moving the door or opening the door. No events are triggered. The sensor itself does not detect motion when opening a door.
```
zigbee2mqtt:info  2020-03-07 09:55:49: MQTT publish: topic 'zigbee2mqtt/0x00158d00039da3a9', payload '{"battery":100,"voltage":3025,"linkquality":90,"angle":7,"angle_x":2,"angle_y":-73,"angle_z":17,"angle_x_absolute":88,"angle_y_absolute":163,"strength":9,"sensitivity":"high"}'
```

Koenkk, 2020-03-07:
Yes, sensitivity is set to high now so thats expected.

vanlooverenkoen, 2020-03-09:
I found that the sensor only triggers every minute. Not handy during development :D I figured it out. Thanks :D

ThreeSixty123, 2022-04-21:
its more like an of topic... but for me as a beginner its not quite clear with HA how to send the command ...

metawops, 2023-01-05:
In a [YT video (German)](https://youtu.be/47Zst9xzFq0?t=257) I saw that the sensitivity of this sensor can bet set to numerical values between 1 and 21 (this is done in the video via the ZHA integration).
I guess I can do this on the dev console tab of that sensor in the zigbee2mqtt integration, too.
But I was wondering: what values are matched to the "low", "medium" and "high" settings we can choose on the Exposes tab? And what would be the most sensitiv value? 1 or 21?
Because I'm trying to use my Aqara vibration sensor to recognize new mail in my mailbox and even the current "high" setting is not sensible enough ...
Thanks for any hints! 🙏

metawops, 2023-01-06:
Oh ... From other discussions on the HA community forum I learned that "high" seems to correspond to the value 1, "medium" to 11 and "low" to 21. So there seems to be no "even more sensitive" setting than "high", unfortunately. 😞

daloki83, 2024-01-23:
I just switched from ZHA to Z2M and sensitivity 5 was a good value in ZHA. Is there a way to set this in Z2M too? I'm using this for an old doorbell. High is far too sensitive and medium doesn't detect the bell

thokaro, 2024-09-12:
I have exactly the same problem. It would be great if the sensitivity could be adjusted more precisely.

Koenkk, 2025-01-25:
Implemented the more precise control!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

ptr727, 2025-04-06:
@Koenkk any way you can maintain backwards compatibility with the old values being reported as enum/text from existing sensors?  
With this change all existing sensors generate parsing errors, see #26919

Koenkk, 2025-04-07:
@ptr727 I don't see how can easily do it, setting this value again should fix the issue.

ptr727, 2025-04-07:
> [@ptr727](https://github.com/ptr727) I don't see how can easily do it, setting this value again should fix the issue.

Ok thx, seems to work, I changed the sensitivity value to 11, pressed button couple times, and now value is reported as 11, and HA / MQTT seems happy. I'll do the same with remaining sensors when I feel like getting on a ladder to reach them.

stphnduvall, 2026-11-15:
> Fixed in the latest dev branch.
> 
> Note that after setting the sensitivity you immediately have to start pressing the reset button with an interval of 1 second until you see Zigbee2mqtt publishing the new sensitivity to MQTT. (same as what the official Xiaomi gateway shows):
> 
> [image]

Can this instruction be put in the device documents? I had a hard time finding this until now.