Koenkk/zigbee2mqtt #3030: IKEA E1743 kicking out Aqara motion sensor

mr2c12, 2020-02-29:
- coordinator cc2530+ccc2591, fw 1.2 source routing
- one cc2530 router
- mixed osram, ikea and philips bulbs
- one innr smart plug
- one xiaomi smart plug
- a bunch of battery powered end devices
- z2m 1.9 running on a raspberry

This problem is driving me crazy.
Sometimes (like 1-2 times a day) the Ikea E1743 on/off remote takes a long time to get back (i.e., you press the button, nothing happens, no log from z2m, no mqtt message, so on). After several button presses, than it comes "online" and works as usual.
When *this* happens, the Aqara motion sensor 0x00158d00019fdd44 disappears. Nothing is received by z2m, even if the motion is triggered for sure.

When the E1743 is already "online" (i.e., immediately react to the button press) this is not happening.
The two devices are quite close (like 1 meter apart).
In the same area (1 square meter..) there are a bunch of devices:
- a cc2531 router (0x00124b001943099c)
- a Philips hue light (0x0017880106e353c9)
- another Aqara motion sensor (0x00158d0004609d31) that ALWAYS work but seems connected directly to the coordinator (given the network map, but seems reasonable because it reports a linkquality of zero and that area is a bit far from the coordinator)

This is the map. I added a red arrow pointing from the offending device to the "victim".
[image]

This has been happening for two days now. Anyway, in the latest few days, I:
- added a lot of new sensors
- switched from "not source routing" to "source routing"

Adding another router in the same area didn't change.
Any other end device seems unaffected.

Can you please help me?

Koenkk, 2020-02-29:
Could you sniff the network when this happens? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

mr2c12, 2020-02-29:
Sure. I just installed and configured the necessary tools.
Already tried a first capture but.. the remote was "online" and the problem didn't occured. Anyway, keeping the capture just in case.
Will try later or maybe tomorrow and let you know.

I suspect something is going on with the ikea remote sleep.

Hilpas, 2020-03-14:
I also discovered major stabillity problems with my z2m network and i actually added two E1743 lately. Some devices simpliy dropping off the network which can only be recovered by either restarting or pairing it again. I will remove my two remotes and try whether that fixes it for me.

mr2c12, 2020-05-13:
I hae not been able to replicate the problem.

closed 2020-05-13 by mr2c12