Koenkk/zigbee2mqtt #3031: tests failing for dev branch

sjorge, 2020-02-29:
```
[hyperon :: sjorge][~/Documents/repos/zigbee2mqtt][dev ✔]
[■]$ npm test

> zigbee2mqtt@1.11.0-dev test /Users/sjorge/Documents/repos/zigbee2mqtt
> jest test

 PASS  test/bridgeConfig.test.js
 PASS  test/otaUpdate.test.js
 PASS  test/deviceBind.test.js
 PASS  test/deviceReport.test.js
 PASS  test/deviceConfigure.test.js
 PASS  test/entityPublish.test.js
 PASS  test/utils.test.js
 PASS  test/data.test.js
 PASS  test/networkMap.test.js
 PASS  test/settings.test.js
 PASS  test/deviceEvent.test.js
 PASS  test/logger.test.js
 PASS  test/homeassistant.test.js (5.13s)
 FAIL  test/group.test.js (5.233s)
  ● Groups › Should not publish state change off if any lights within are still on when changed via device

    expect(jest.fn()).toHaveBeenCalledWith(...expected)

    - Expected
    + Received

      "zigbee2mqtt/bulb_color",
    - "{\"state\":\"OFF\"}",
    + "{\"state\":\"OFF\",\"brightness\":50,\"color_temp\":370}",
      {"qos": 0, "retain": false},
      [Function anonymous],

    Number of calls: 1

      359 |         await flushPromises();
      360 |         expect(MQTT.publish).toHaveBeenCalledTimes(1);
    > 361 |         expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/bulb_color", '{"state":"OFF"}', {"retain": false, qos: 0}, expect.any(Function));
          |                              ^
      362 |     });
      363 |
      364 |     it('Should not publish state change off if any lights within are still on when changed via shared group', async () => {

      at Object.<anonymous> (test/group.test.js:361:30)

  ● Groups › Should not publish state change off if any lights within are still on when changed via shared group

    expect(jest.fn()).toHaveBeenCalledWith(...expected)

    Expected: "zigbee2mqtt/group_2", "{\"state\":\"OFF\"}", {"qos": 0, "retain": false}, Any<Function>
    Received
           1
              "zigbee2mqtt/bulb_color",
              "{\"state\":\"OFF\",\"brightness\":50,\"color_temp\":370}",
              {"qos": 0, "retain": false},
              [Function anonymous],
           2
              "zigbee2mqtt/group_2",
              "{\"state\":\"OFF\",\"brightness\":50,\"color_temp\":370}",
              {"qos": 0, "retain": false},
              [Function anonymous],

    Number of calls: 2

      384 |         await flushPromises();
      385 |         expect(MQTT.publish).toHaveBeenCalledTimes(2);
    > 386 |         expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/group_2", '{"state":"OFF"}', {"retain": false, qos: 0}, expect.any(Function));
          |                              ^
      387 |         expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/bulb_color", '{"state":"OFF"}', {"retain": false, qos: 0}, expect.any(Function));
      388 |     });
      389 |

      at Object.<anonymous> (test/group.test.js:386:30)

  ● Groups › Should publish state change off if all lights within turn off

    expect(jest.fn()).toHaveBeenCalledWith(...expected)

    Expected: "zigbee2mqtt/bulb_color", "{\"state\":\"OFF\"}", {"qos": 0, "retain": false}, Any<Function>
    Received
           1
              "zigbee2mqtt/bulb_color",
              "{\"state\":\"OFF\",\"brightness\":50,\"color_temp\":370}",
              {"qos": 0, "retain": false},
              [Function anonymous],
           2
              "zigbee2mqtt/group_1",
              "{\"state\":\"OFF\",\"brightness\":50,\"color_temp\":370}",
              {"qos": 0, "retain": false},
              [Function anonymous],
           3
              "zigbee2mqtt/bulb",
              "{\"state\":\"OFF\",\"brightness\":50,\"color_temp\":370,\"linkquality\":99}",
              Object {
                "qos": 0,
            -   "retain": false,
            +   "retain": true,
              },
              [Function anonymous],

    Number of calls: 3

      410 |         await flushPromises();
      411 |         expect(MQTT.publish).toHaveBeenCalledTimes(3);
    > 412 |         expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/bulb_color", '{"state":"OFF"}', {"retain": false, qos: 0}, expect.any(Function));
          |                              ^
      413 |         expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/bulb", '{"state":"OFF","brightness":50,"color_temp":370,"linkquality":99}', {"retain": true, qos: 0}, expect.any(Function));
      414 |         expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/group_1", '{"state":"OFF","brightness":50,"color_temp":370}', {"retain": false, qos: 0}, expect.any(Function));
      415 |     });

      at Object.<anonymous> (test/group.test.js:412:30)

 PASS  test/deviceAvailability.test.js (5.216s)
 PASS  test/deviceReceive.test.js (5.935s)
 PASS  test/controller.test.js (6.716s)

Test Suites: 1 failed, 16 passed, 17 total
Tests:       3 failed, 349 passed, 352 total
Snapshots:   0 total
Time:        8.221s
Ran all test suites matching /test/i.
npm ERR! Test failed.  See above for more details.
```

I did a fresh checkout of the dev branch, ran:
- `npm install`
- `npm test`

And I get these failures? Did I mess something? I thought I might have skipped a commit so I did a fresh checkout with the same result. (So no git fetch git rebase)

sjorge, 2020-03-02:
Test succeed on azure, probably because I am using node 13 on my dev machine. Will close.

closed 2020-03-02 by sjorge

Koenkk, 2020-03-02:
Tests are also working on node 12 now. thanks!