Koenkk/zigbee2mqtt #3032: Unable to use Tradfri Motion Sensor

papaiatis, 2020-02-29:
# Bug Report
## What happened
Just bought a brand new Tradfri Motion Sensor in IKEA. I wanted to connect this device to my network and from the logs it seemed to be successful. But the sensor does not send any data and since it's joined all my three Tradfri lights are not working either. I tried resetting the motion sensor but I can't see anything in the logs while its supposed to be rejoining again.
## What did you expect to happen
I expected the motion sensor to just work fine as its supposed to be a supported device.
## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: latest from master
CC253X firmware version: Can't remember. I flashed the firmware about a year ago.

Logs when the device first joined: https://hastebin.com/xebenevayi.apache

Koenkk, 2020-02-29:
The firmware version on the coordinator is very important here, you need a recent one to make this device work. Probably this is the cause of your issue.

papaiatis, 2020-02-29:
Okay, I'm going to flash newest firmware to see if it resolves the problem.

dfsx1, 2020-02-29:
Did you tried both reset and pairing commands?
Reset = 4x pair button in 5sec
Pairing = hold pairing button 10sec

Also try power-off those bulbs before re-pairing the motion sensor to the coordinator.
And note that the sensor needs 3 minutes of non-motion before it re-detects any new motion.

Mine works fine btw, (with the latest firmware) so I would also update that.

papaiatis, 2020-02-29:
Current firmware:
`Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20181024}}'
`

@dfsx1 Only the Reset seemed to be doing something. Holding the button for 10 seconds didn't have any effect (no red LEDs turned on).
Good idea though to power off the lights!

papaiatis, 2020-02-29:
@Koenkk 
Updated firmware to latest. 
Now I'm still trying to fix the Tradfri lights first.
I see this exception in the logs when I try to switch on the light from HASS:

```
febr 29 18:39:33 atis-brix npm[28191]: zigbee2mqtt:debug 2020-02-29 18:39:33: Error: Command 0x90fd9ffffe2975d2/1 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (TypeError: entity_1.default.adapter.sendZclFrameNetworkAddress is not a function)
febr 29 18:39:33 atis-brix npm[28191]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:347:23)
febr 29 18:39:33 atis-brix npm[28191]:     at Generator.next (<anonymous>)
febr 29 18:39:33 atis-brix npm[28191]:     at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:8:71
febr 29 18:39:33 atis-brix npm[28191]:     at new Promise (<anonymous>)
febr 29 18:39:33 atis-brix npm[28191]:     at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:4:12)
febr 29 18:39:33 atis-brix npm[28191]:     at Endpoint.command (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:321:16)
febr 29 18:39:33 atis-brix npm[28191]:     at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:103:26)
febr 29 18:39:33 atis-brix npm[28191]:     at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:325:60)
febr 29 18:39:33 atis-brix npm[28191]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:173:52)
febr 29 18:39:33 atis-brix npm[28191]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:329:44)
```
Does it mean anything to you?

papaiatis, 2020-02-29:
Another one while trying to pair the Tradfri Motion sensor:

```
ebr 29 19:36:33 atis-brix npm[3510]: zigbee2mqtt:info  2020-02-29 19:36:33: Starting interview of '0xec1bbdfffeb7e648'
febr 29 19:36:33 atis-brix npm[3510]: zigbee2mqtt:info  2020-02-29 19:36:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffeb7e648"}}'
febr 29 19:36:42 atis-brix npm[3510]: (node:3539) UnhandledPromiseRejectionWarning: TypeError: this.adapter.supportsDiscoverRoute is not a function
febr 29 19:36:42 atis-brix npm[3510]:     at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:499:49)
febr 29 19:36:42 atis-brix npm[3510]:     at Generator.throw (<anonymous>)
febr 29 19:36:42 atis-brix npm[3510]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:6:65)
febr 29 19:36:42 atis-brix npm[3510]: (node:3539) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 6)
```

papaiatis, 2020-02-29:
It was probably the old firmware on CC253X that prevented me using this sensor. 
Although I had other issues as well after I tried to pair the sensor (with the old FW) and again more issues after I upgraded the FW finally I decided to start from the beginning, wiped out my entire zigbee2mqtt installation (made a backup though), pulled the latest software and paired all my 20 devices all over again. Some of the devices got disconnected in the process and had to re-pair them again but after all it all works now, including the Tradfri Motion Sensor.

Thanks for the help!

closed 2020-02-29 by papaiatis