Koenkk/zigbee2mqtt #3033: Home Assistant MQTT triggers

Koenkk, 2020-02-29:
Home Assistant 0.106 adds support for MQTT device triggers ([link](https://www.home-assistant.io/integrations/device_trigger.mqtt/)). 

This allows to trigger on a device `action` or `click` without using the state change workaround on `sensor_action`/`sensor_click` ([link](https://www.zigbee2mqtt.io/integration/home_assistant.html#via-home-assistant-entity)).

This has now been implemented in the latest Zigbee2mqtt dev branch.

To use it first trigger the event once on the device (e.g. press a button on your remote), Zigbee2mqtt will now discover the MQTT device trigger into Home Assistant.

After this you can e.g. use the Home Assistant automation editor to respond to this event.

[image]

Big thanks to @emontnemery for implementing this on the Home Assistant side.

Kryzek, 2020-02-29:
Interesting, would it ne possible to create `button-hold-for`-style triggers i.e. hold remote button for >5 seconds to turn off all lights or would this require per-device support for held buttons?

FutureCow, 2020-03-01:
Will it also be possible to remove the "hack" to send two messages when receiving a click when HA is set to true?

`info  2020-03-01 10:56:05: MQTT publish: topic 'zigbee2mqtt/Schakel_Knop_Woonkamer', payload '{"battery":100,"voltage":3025,"linkquality":36,"last_seen":1583056565457,"click":"single","elapsed":16002}'
info  2020-03-01 10:56:05: MQTT publish: topic 'zigbee2mqtt/Schakel_Knop_Woonkamer', payload '{"battery":100,"voltage":3025,"linkquality":36,"last_seen":1583056565457,"click":""}'
`

sjorge, 2020-03-01:
+1 for eventually removing the hack... or making it an option

Koenkk, 2020-03-01:
@Kryzek no, it will provide the same functionality as the `sensor_action` does.

@FutureCow @sjorge good idea, you can now do this in the latest dev branch by setting in `configuration.yaml` 

```yaml
advanced:
  homeassistant_legacy_triggers: false # Default = true as we don't want a breaking change
```

This will:
- Prevent the clear payload (e.g. `click: 'single'` -> `click: ''`)
- Prevent the `sensor_action` and `sensor_click` from being discovered

closed 2020-05-07 by stale[bot]