Koenkk/zigbee2mqtt #3035: zigbee2mqtt & TP-Link Deco M9 Plus

bartplessers, 2020-02-29:
Hi

Just a question:
Can zigbee2mqtt and TP-Link Deco M9 Plus form a single mesh?
I want zigbee2mqtt to be my master, I have a custom config with networkkey etc....,
Does anybody know if you can add such devices as simple zigbee routers?

kind regards,
Bart

VincentSC, 2020-03-25:
What I found is that TP-link does not allow the Deco's to work as Zigbee-repeaters - for me this was the reason I wanted to buy this specific version. Can't find the link right now - was somewhere on their forums. I hope TP-link does start doing this, once they found out there is far too much competition for yet another bridge with support for a limited set of devices

stefanodevrieso, 2020-05-24:
Yes, I have the same question. I am using a CC2531 USB stick (my zigbee router), but my zigbee mesh network is not good enough. The same with my wifi network, I do not have a signal on the third floor. I want to buy a TP-link Deco M9 to improve my wifi network and I was wondering if I could also use the TP-link stations as repeaters in my zigbee network

closed 2020-07-26 by stale[bot]

andreapx, 2023-02-12:
Have anybody had the chance to use the Deco M9 as router and not coordinator?

VincentSC, 2023-02-12:
It needs to be rooted to do that, as it's a walled garden. So it is really just an attempt of TP-Link to make their own Z2M with very, very limited device-support.

I bought them and returned them when I found out it was all a farce. 

Best place to discuss this subject is actually the general one: https://community.home-assistant.io/t/extend-zigbee-via-ethernet/417436/3