Koenkk/zigbee2mqtt #3036: Cant set state of Weiser SmartCode 10 Touch after upgrade

luckypants123, 2020-03-01:
Hi,

Just upgraded to 1.11 via Hassio. Was previously on 1.09. 

Not sure what happened but I can no longer control my Weiser SmartCode 10 Touch door lock.

Tried to re-pair, and re-install z2m but no luck.

On publishing 'lock' to zigbee2mqtt/backdoorlock/set for this device:
```
zigbee2mqtt:info  2020-03-01 00:38:23: backdoorlock (0x002446fffd01e4f3): 9GED21500-005 - Weiser SmartCode 10 Touch (EndDevice)
```

I get in the log:
```
zigbee2mqtt:error 2020-03-01 00:38:56: Publish 'set' 'state' to 'backdoorlock' failed: 'Error: Command 0x002446fffd01e4f3/2 closuresDoorLock.lockDoor({"pincodevalue":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: AREQ - AF - dataConfirm after 3000ms)'
zigbee2mqtt:info  2020-03-01 00:38:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'backdoorlock' failed: 'Error: Command 0x002446fffd01e4f3/2 closuresDoorLock.lockDoor({\"pincodevalue\":\"\"}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":true}) failed (Error: AREQ - AF - dataConfirm after 3000ms)'","meta":{"friendly_name":"backdoorlock"}}'
```

Any idea what it could be?

Thanks much,
Ian.

GuGu927, 2020-03-01:
Similar error here.
some switch, light shows thie error.
This error randomly happens and delaying when turning on/off.

Here's my error logs.
zigbee2mqtt:error 2020-03-01 17:51:50: Publish 'set' 'state' to 'gate_lightstrip' failed: 'Error: Command 0x0017880102af43f4/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'

luckypants123, 2020-03-02:
I resolved this somehow through a series of repairings of the device. Not sure why but works great now.

@GuGu927 I'd recommend trying the same. Good luck!

closed 2020-03-02 by luckypants123