Koenkk/zigbee2mqtt #3038: Bitron Video thermostat (AV2010/32) Home Assistant integration

lociii, 2020-03-01:
Hey folks,

first I wanted to thank all people involved and Koen in special for all the hard and great work on these tools! I cannot imagine my smart home without them anymore.

I recently bought a Bitron Video wall thermostat since it's supported by z2m. However the Home Assistant integration is not yet defined. As I'm pretty new to the internals of z2m/herdsman, I would like to clarify first if I understood the relations right.

Current Home Assistant definition for the device:
https://github.com/Koenkk/zigbee2mqtt/blob/ed75c85b69051692b86eaeb3739936e847a2d777/lib/extension/homeassistant.js#L759

Features provided by herdsman:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/2c599294a51dc735784ca027810e69024398ab4d/devices.js#L4503

If I understand the converters correctly, the AV2010/32 should be able to support the following features:

* sensor_local_temperature via bitron_thermostat_att_report
* thermostat with occupied_heating_setpoint  via bitron_thermostat_att_report
* sensor_battery via bitron_battery_att_report
* sensor_voltage via bitron_battery_att_report

So the following Home Assistant setup should be working:
```
'AV2010/32': [
    cfg.thermostat(7, 30, 'occupied_heating_setpoint', 0.5), cfg.sensor_local_temperature, cfg.sensor_battery, cfg.sensor_voltage,
],
```
Min/max temperature tested on device. Turns off below 7 and caps at 30.

Any hints/feedback would be appreciated. Not sure how to test properly without impacting my "production" installation.

Thanks, Jens

Koenkk, 2020-03-01:
I've added the following config (as this is how also other thermostats are integrated).

```js
    'AV2010/32': [cfg.thermostat(7, 30, 'occupied_heating_setpoint', 0.5), cfg.sensor_battery],
```

You should be able to test this on your production setup easily now by switching to the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

lociii, 2020-03-01:
Works perfect. Awesome. Thank you so much.

[image]

closed 2020-03-02 by Koenkk