Koenkk/zigbee2mqtt #3039: [Question] How do I pair the DIYRuZ 20 button remote?

Calimerorulez, 2020-03-01:
First of all sorry for opening an issue (it is more a question). I can't seen to find the answer anywhere :(

I see that the DIYRuZ 20 button remote (https://modkam.ru/?p=1114) is supported by zigbee2mqtt, and I built two of them, but I can't seem to pair it with my network.

Does anyone knows the button combination to pair the device?

Thanks in advance.

closed 2020-05-07 by stale[bot]

mo6zes, 2020-05-28:
Did you get it to work? I am facing the same issue

Calimerorulez, 2020-06-05:
@mo6zes Well, I managed to get it to work by using my Zigbee 1.2 USB stick. With my 3.0 coordinator it does not work.

You will have to press and hold key number 17 (that is what the file name of the firmware says), until the led starts to flash. Then release the button and press it again shortly...