Koenkk/zigbee2mqtt #3041: Heiman HS1CG-E reconized as Piri HSIO18008

maxparrela, 2020-03-01:
# Bug Report
## What happened
Smart combustible gas sensor  Heiman HS1CG-E was reconized as Piri HSIO18008 by Zigbee2mqtt Hass.io Add-on
At the first time it was reconized as Piri HSIO18008, but didn't work.
At the second, it worked, including on testing with a lighter (https://pastebin.com/as4JASsS)
[image]
Despite showing the battery status, the device doesn't have one.

On edge version, it failed to interview (https://pastebin.com/F9N6tkei)

I think that HS1CG-E is very similar to HS1CG-M.

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: CC2531_DEFAULT_20190608.zip

PS: English is not my first language.

Thank you in advance for any help.

Koenkk, 2020-03-02:
Can you provide the herdsman debug logging of the interview fail?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

maxparrela, 2020-03-03:
Koenkk, folow:

https://pastebin.com/gVeN5Hcq

CodeFinder2, 2020-03-09:
I can confirm this issue, I have bought the same sensor.  I could pair the sensor successfully but have 3 problems (v1.11.0):

- No battery information is visible in Home Assistant ("unknown")
- The device is incorrectly identified as TODO
- When I trigger a gas alarm (using real LPG gas 😆 ) I do NOT get a notification via Z2M (nothing is shown in the logs actually). I assume this isn't due to a faulty unit, isn't it? 🙈 

Commented log (log_level=debug):
```
# Pairing:
info  2020-03-09 19:00:30: Device '0x000d6f0014f70b11' joined
info  2020-03-09 19:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f0014f70b11"}}'
info  2020-03-09 19:00:30: Starting interview of '0x000d6f0014f70b11'
info  2020-03-09 19:00:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f0014f70b11"}}'
debug 2020-03-09 19:00:30: Device '0x000d6f0014f70b11' announced itself
debug 2020-03-09 19:00:40: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'genBasic', data '{"modelId":"GASSensor-EM","manufacturerName":"HEIMAN"}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:40: No converter available for 'HSIO18008' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"GASSensor-EM","manufacturerName":"HEIMAN"}'
info  2020-03-09 19:00:40: MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f0014f70b11/gas/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.gas }}","device_class":"gas","state_topic":"zigbee2mqtt/0x000d6f0014f70b11","json_attributes_topic":"zigbee2mqtt/0x000d6f0014f70b11","name":"0x000d6f0014f70b11_gas","unique_id":"0x000d6f0014f70b11_gas_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f0014f70b11"],"name":"0x000d6f0014f70b11","sw_version":"Zigbee2mqtt 1.11.0","model":"Combustible gas sensor (HSIO18008)","manufacturer":"Piri"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-03-09 19:00:40: MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f0014f70b11/battery_low/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.battery_low}}","device_class":"battery","state_topic":"zigbee2mqtt/0x000d6f0014f70b11","json_attributes_topic":"zigbee2mqtt/0x000d6f0014f70b11","name":"0x000d6f0014f70b11_battery_low","unique_id":"0x000d6f0014f70b11_battery_low_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f0014f70b11"],"name":"0x000d6f0014f70b11","sw_version":"Zigbee2mqtt 1.11.0","model":"Combustible gas sensor (HSIO18008)","manufacturer":"Piri"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-03-09 19:00:40: MQTT publish: topic 'homeassistant/sensor/0x000d6f0014f70b11/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f0014f70b11","json_attributes_topic":"zigbee2mqtt/0x000d6f0014f70b11","name":"0x000d6f0014f70b11_linkquality","unique_id":"0x000d6f0014f70b11_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f0014f70b11"],"name":"0x000d6f0014f70b11","sw_version":"Zigbee2mqtt 1.11.0","model":"Combustible gas sensor (HSIO18008)","manufacturer":"Piri"},"availability_topic":"zigbee2mqtt/bridge/state"}'
debug 2020-03-09 19:00:41: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":1}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:41: No converter available for 'HSIO18008' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1,"zclVersion":1}'
debug 2020-03-09 19:00:41: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'genBasic', data '{"appVersion":18,"stackVersion":2}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:41: No converter available for 'HSIO18008' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":18,"stackVersion":2}'
debug 2020-03-09 19:00:41: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16,"dateCode":"2018.3.26"}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:41: No converter available for 'HSIO18008' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":16,"dateCode":"2018.3.26"}'
debug 2020-03-09 19:00:41: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:41: No converter available for 'HSIO18008' with cluster 'genBasic' and type 'readResponse' and data '{}'
debug 2020-03-09 19:00:41: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":1}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:41: No converter available for 'HSIO18008' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":1}'
debug 2020-03-09 19:00:42: Received Zigbee message from '0x000d6f0014f70b11', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b001ca5fdaf","zoneState":1}' from endpoint 1 with groupID 0
debug 2020-03-09 19:00:42: No converter available for 'HSIO18008' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b001ca5fdaf","zoneState":1}'
[... cut ...]
info  2020-03-09 19:25:03: Successfully renamed - 0x000d6f0014f70b11 to basement/combustible_gas_leakage 
info  2020-03-09 19:25:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_renamed","message":{"from":"0x000d6f0014f70b11","to":"basement/combustible_gas_leakage"}}'
```
These are **all messages** related to the device. And I made to sensor trigger its alarm (beep, blinking) twice during the above log recoding.

I can also support debugging the issue. Would be great if this can be fixed! :-)

Koenkk, 2020-03-09:
Could you provide the same log when you trigger the gas alarm?

Also, where do you see the TODO? (I don't see it in your logs).

CodeFinder2, 2020-03-09:
Sorry that was a typo, I meant HSIO18008, not TODO. :see_no_evil:

The log above is the log when I triggered the alarm twice.

maxparrela, 2020-03-10:
@CodeFinder2 actually I've tested using a Lighter. Using that I can release small and safe quantity of LPG.

CodeFinder2, 2020-03-10:
Yeah I used a camping cooker but did it actually trigger a change in zigbee2mqtt and a change in the HA entities you screenshooted above when you triggered the gas alarm with a lighter? I mean did the sensor change from 'clear' to whatever? 🙈

This is how mine currently looks like in HA:
[image]
Despite the fact that it does not have a battery, 'off' and 'unknown' seems wrong. 🤔 To me it seems that my device cannot communicate with my coordinator?! 😐 But note that after a network scan, the device seems to appear correctly in my network map (LQI's >> 0).

maxparrela, 2020-03-10:
@CodeFinder2 

"...you screenshooted above when you triggered the gas alarm with a lighter?" No, but i took it when the integration was working with HA.

"I mean did the sensor change from 'clear' to whatever?" When I triggered the sensor, it status at Lovalace change to "Detected"

"To me it seems that my device cannot communicate with my coordinator?" I don't think so. In my opinion, the problem is that the sensor is not recognized correctly by the zigbee2mqtt.

Link quality was "unknown" until the first LPG detection. After that, it started show values.

CodeFinder2, 2020-03-10:
@maxparrela thanks! Unfortunately, I never received a 'detected' info.

@Koenkk here's my db entry for that device, surprisingly, it contains "HEIMAN" as manufacturer:
```
{"id":35,"type":"Router","ieeeAddr":"0x000d6f0014f70b11","nwkAddr":17476,"manufId":4619,"manufName":"HEIMAN","powerSource":"Mains (single phase)","modelId":"GASSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,9,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"GASSensor-EM","manufacturerName":"HEIMAN","powerSource":1,"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.26"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001ca5fdaf","zoneState":1}}},"binds":[]}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.26","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1583776842465}
```

Please let me know if you need more information!

Koenkk, 2020-03-10:
Could you sniff the traffic when triggering the alarm? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

CodeFinder2, 2020-03-12:
Will do so this evening (needed to wait for a sniffer board) and report back. I already tried repairing it (remove + repair) but it still shows being "off" and "unknown" (see screenshot above) in HA and no logs from the device in Z2M.

CodeFinder2, 2020-03-12:
Realllly strange: I was about to do the sniffing and did all the preparations while I realized that the LQI wasn't "unknown" (but 138) anymore. Once I triggered a gas alarm, HA and Z2M noticed this correctly:
```
zigbee2mqtt:debug 2020-03-12 23:50:56: Received Zigbee message from 'basement/combustible_gas_leakage', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":33,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-12 23:50:56: MQTT publish: topic 'zigbee2mqtt/basement/combustible_gas_leakage', payload '{"gas":true,"tamper":false,"battery_low":false,"linkquality":99,"last_seen":"2020-03-12T22:50:56.340Z"}'
[...]
zigbee2mqtt:debug 2020-03-12 23:51:09: Received Zigbee message from 'basement/combustible_gas_leakage', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":32,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-12 23:51:09: MQTT publish: topic 'zigbee2mqtt/basement/combustible_gas_leakage', payload '{"gas":false,"tamper":false,"battery_low":false,"linkquality":144,"last_seen":"2020-03-12T22:51:09.285Z"}'
```
As already written above, I repaired it but it didn't work afterwards right away. May reconnecting it to mains did the trick? I don't know. @Koenkk do you have an idea? Really spooky and somewhat annoying because I would like to *understand* what's happening to avoid such issues/requests for you in the future... 🤔 

Nonetheless, the device is still incorrectly identified. Can you fix this?

Koenkk, 2020-03-14:
- Regarding incorrectly identified, I want to see what the `manufacturer` of the Piri sensor is, asked it here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/604 . In that way I can check how to distinguish between them.
- Strange that it started working out of the blue indeed. For such device I expect it to work when the zoneState of the device is set to `1`, this means that the device indicated that it was setup succesfully. `{"iasCieAddr":"0x00124b001ca5fdaf","zoneState":1}` (from `database.db`). Can you reproduce the problem by repairing/factory resetting the device? If yes, please sniff the traffic while doing so and triggering the alarm.

maxparrela, 2020-03-15:
Dear @Koenkk , I tried that, but something went wrong.

1) Flashed CC2531 with zboss_sniffer.hex from Zboss folder;
2) Plugged CC2531 dongle to USB port. I'm using W10 64 bits. I had to disable w10 driver signature enforcement to use Zboss CC2531 driver.
3) Installed Wireshark version 3.2.2 (v3.2.2-0-ga3efece3d640);
4) Zboss Sniffer runned  as admin:
[image]

5) Configured keys:
[image]

6) Tried to pair HS1CG-E, but it didin't work:
[image]

Log from wireshark: https://pastebin.com/h8dcuyzr

If there's something that i can do, I'm here to collaborate.

CodeFinder2, 2020-03-15:
Thanks from recording a log from my side, too. Just to be sure: did you a) reset the device (by pushing a pin into the reset hole) and b) deleted the device in Z2M via `zigbee2mqtt/bridge/config/remove`  (possibly forcefully if necessary) *before* doing the above recordings?

Just to prevent any effects stemming from your previous trials. 🙈

maxparrela, 2020-03-15:
@CodeFinder2 
a) reset the device (by pushing a pin into the reset hole)
A: yep

and b) deleted the device in Z2M via zigbee2mqtt/bridge/config/remove (possibly forcefully if necessary) before doing the above recordings?
A: No, but I removed MQTT integration (retained topics), deleted any device entry on \share\zigbee2mqtt\devices.yaml and stopped add-on on Home Assistant.

I did the process in a laptop with W10 following https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html.

CodeFinder2, 2020-03-15:
To b): not sure if devices.yaml is really important. AFAIK, database.db is important and the question would be if there's still an residual entry from the device of interest.

maxparrela, 2020-03-20:
Okay, where is located the database.db? You know?

CodeFinder2, 2020-03-20:
It should be right next to` devices.yaml`, that is, in `share/zigbee2mqtt/`.

maxparrela, 2020-03-20:
@CodeFinder2  Found it with Samba. Before I was using Visual Studio Code.

Koenkk, 2020-03-20:
It should be correctly identified in the latest dev branch now.

maxparrela, 2020-03-21:
@Koenkk, at the first time It was appeared as HS1CG-E on log (herdsman_debug = false) and on HA integration, but didn't work (test with lighter, status)
https://pastebin.com/WCwLh4an

After that, I've tried many times (herdsman_debug = true), but interview failed.
https://pastebin.com/xAVshRwF

Koenkk, 2020-03-21:
@CodeFinder2 does it pair fine in your case?

maxparrela, 2020-03-23:
@Koenkk I tried again and now It was recognized correctly. 
[image]

Triggered with lighter
[image]

Log: https://pastebin.com/mtF8f9mY

maxparrela, 2020-03-26:
@Koenkk after days on-line and with many lighter test, It's still working like a charm. Thank you.

Koenkk, 2020-03-27:
Good, can this be closed?

closed 2020-03-27 by maxparrela

CodeFinder2, 2020-03-28:
@Koenkk  Now tested with latest dev (sorry for the delay here) and everything is working fine. I just realized that there's still a `battery_low` indicator (see also @maxparrela screenshot above). Is this really necessary or can we get rid of it? At least *this* kind of sensor is powered by mains. ;-)

Koenkk, 2020-03-29:
@CodeFinder2 done, you might have to clear the retained MQTT discovery manually in order to prevent it from coming back.