Koenkk/zigbee2mqtt #3043: Interview fail Osram Smart+ Plug AB32570

jkrenzer, 2020-03-01:
# Bug Report
## What happened

- Tried to pair a Osram Smart+ Plug AB32570 with Zigbee2Mqtt
- Device does join but Interview fails after some time
  - Herdsman complains about not getting node descriptor
  - Problem persists after trying....
     - Reflashing CC2531 multiple times
     - Trying source-rounting and default zigbee ha 1.2 firmware
     - Cleaning up device database (suggested in other issue)
     - Trying TouchLink resetting (fails)
     - Putting plug very near to the adapter and also tryping in other rooms (to eliminate interference as cause)
- Bulbs still pair fine

## What did you expect to happen

- Sucessfull pairing

## How to reproduce it (minimal and precise)

- Allow device join
- Reset plug by long-pressing button until it clicks (~10s)
- Zigbee2Mqtt reports device joining
- After some time (several attempts to get node descriptor it seems) -> Failure of interview

## Debug Info
zigbee2mqtt version: 1.11.0 (tried also 1.10.0)
CC253X firmware version: CC2531_SOURCE_ROUTING_20190619 (Default 20190608 also tried)

### Seemingly relevant hersman/z2m log excerpt

[See here](https://hastebin.com/hucomiqoqi.pl)

Cheers,

Jörn

Koenkk, 2020-03-02:
As you've already tried everything, could you sniff the traffic when pairing the device? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

jkrenzer, 2020-03-02:
Ok, I will try to dig up my old Ember adapter and have a look.

jkrenzer, 2020-03-02:
Ok, I got the packages to and from the plug's hardware-address extracted into a file:

[zigbee_traffic_pairing_osram_plug.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4277976/zigbee_traffic_pairing_osram_plug.pcapng.gz)

Hope it sheds some light on what is happening.

Koenkk, 2020-03-05:
The device joins the network via a router (not the coordinator) and it seems that that routes does not send the network key to the device. Can you try pairing it close to the coordinator and provide the sniff of that?

jkrenzer, 2020-03-05:
I did not have time to capture new traffic yet, but I had the device like 3cm next to the CC-Stick if I remember correctly. I read in another issue about plugs having bad antenna, so I wanted to be sure. 

Today I bought a new tint-bulb to pair. Just to be sure it is not some hiccup with new devices and the others (old Tradfri ones) did just re-pair because they or the stick had cached something. The new bulb took less then 3 seconds to interview and join. I also tried to pair the plug near this new router in case something is wrong with the tradfri bulbs handing over the join.

Everything still not successful. So it seems this plug is doing something differently. I will sniff again when I come home from my pending journey.

Thanks for the help!

-Jörn

jkrenzer, 2020-03-10:
Very funny: While wildly experimenting with the setup I somehow solved this particular problem.

What I did:

- Force-remove the device from Z2M
- Stopping Z2M and deleting all excess entries of the device from the database
- Unplug the stick for 10s, then replug and start Z2M, allowing pairing in the config
- Disconnected all routers near the plug and coordinator from power
- Connect a newly paired router (Innr bulb) to power and put the plug very near to it
- Factory-reset the plug (10+ sec. button pressing)

Then it did connect immediately! Mind you: I am using the source-routing firmware on my CC-Stick!

What did **not** help before:

- Removing the device from the db
- Pairing near to the coordinator
- Factory-resetting manually or trying by touchlink
- Pairing near the coordinator while all near routers are disconnected

So this lets me wonder:

- Is the handling of the connection request the cause?
  - Maybe my coordinator was already saturated with direct children, so the plug turned to a router for connection (source-routing FW)
  - But somehow the plug acts differently in this setup than for example the Innr bulb, so the Tradfri bulbs transfer the key to the latter but not to the plug. Different protocol implementations? Plug seems to be Zigbee 3.0, Tradfri is ZHA 1.2.
  - The Innr bulb again seems to handle the connection request differently and provides the plug with the need information. (Did not find information on Innr Zigbee version yet)
- Or was it the db-cleanup + force_remove that helped?

I propose we keep an eye on other people having pairing issues with the source-routing firmware. Maybe we can let them try the procedure "pairing it close to another (newly paired/newly acquired different type) router" and get some confidence if this helps in such setups or not. Maybe add this as something to try in the troubleshooting section then?

Meanwhile I will see if I can reproduce the behavoir while sniffing the traffic and pinpointing the step which made the difference. 

- Jörn

Koenkk, 2020-03-10:
I don't think a db cleanup or force remove could solve these issues, the CC2531 doesn't know about it and nor it is transferred to it.

I guess the replug helped, this is also mentioned in the docs (recently added). https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all (see CC2531/CC2530 coordinator only)

jkrenzer, 2020-03-10:
I am not sure if this is the case as I replugged  the CC multiple times already. At least for reflashing the firmware I had to remove it at least three times from the PI and carry it to my PC in the living room.

closed 2020-05-16 by stale[bot]