Koenkk/zigbee2mqtt #3044: Pi4 + Zigbee2MQTT error

MDLefevere, 2020-03-02:
Hi there,

Recently i've been moving my stuff around from a Pi3 to a Pi4.
Clean install and running from an SSD connected to the Pi4

Everything is working great untill i was at the point were i needed to add my MQTT devices.
I've installed everything from scratch but now i'm getting an error and i don't know what it's about.

`zigbee2mqtt:info  2020-03-02 08:23:55: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-02.08-23-55' filename: log.txt
zigbee2mqtt:info  2020-03-02 08:23:55: Starting zigbee2mqtt version 1.11.0-dev (commit #ce48bee4cd0eac6cb36ed36b20a39024e1ecf0de)
zigbee2mqtt:info  2020-03-02 08:23:55: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-02 08:25:00: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-02 08:25:00: Failed to start zigbee
zigbee2mqtt:error 2020-03-02 08:25:00: Exiting...
zigbee2mqtt:error 2020-03-02 08:25:00: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0-dev start: node index.js
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.11.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-02T07_25_00_580Z-debug.log
2020-03-02T08:25:01: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-03-02T08:25:01: PM2 log: App [npm:0] starting in -fork mode-
2020-03-02T08:25:01: PM2 log: App [npm:0] online
> zigbee2mqtt@1.11.0-dev start /app
> node index.js`

All is fine. Username. Password. Tried the dev version but that gave me the same error.

MDLefevere, 2020-03-02:
Okey. So it looks like i got it working. But nothing is joining even when it's set to True

`zigbee2mqtt:info  2020-03-02 08:48:48: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-02.08-48-48' filename: log.txt
zigbee2mqtt:info  2020-03-02 08:48:49: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
zigbee2mqtt:info  2020-03-02 08:48:49: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-03-02 08:49:15: zigbee-herdsman started
zigbee2mqtt:info  2020-03-02 08:49:15: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-03-02 08:49:15: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-03-02 08:49:15: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-03-02 08:49:15: Allowing new devices to join.
zigbee2mqtt:warn  2020-03-02 08:49:15: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-03-02 08:49:15: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-02 08:49:15: Connecting to MQTT server at mqtt://core-mosquitto
zigbee2mqtt:info  2020-03-02 08:49:16: Connected to MQTT server
zigbee2mqtt:info  2020-03-02 08:49:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-02 08:49:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'`

Miguelin96, 2020-03-02:
zigbee2mqtt 1.11.0: 

zigbee2mqtt:info  2020-03-03 00:22:00: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-03.00-22-00' filename: log.txt
zigbee2mqtt:info  2020-03-03 00:22:01: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
zigbee2mqtt:info  2020-03-03 00:22:01: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-03 00:22:09: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-03 00:22:09: Failed to start zigbee
zigbee2mqtt:error 2020-03-03 00:22:09: Exiting...
zigbee2mqtt:error 2020-03-03 00:22:09: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

micromad, 2020-04-14:
Same thing for me

arcstar7, 2020-04-25:
Me as well

SimonMerrett, 2020-05-04:
I have a Pi4 4GB + boot from SD card. sudo apt upgraded today before trying to do any of this and it seems to work for me. Maybe try upgrading Raspian now and reinstalling.

micromad, 2020-05-05:
I just found the mistake and solve it.
I found two different versions of node.js installed on my rpi.
Just type
`whereis npm`
and found one in `/usr/bin` and another one in `/usr/local/bin`

Clean all and download the last version of node.js for my rpizero (armv6l) version 10 in my case.
Backup your data directory of your zigbee2mqtt installation.
Reinstall zigbee2mqtt and replace the file in the `data` directory with your backup files and it works for me.

closed 2020-07-11 by stale[bot]