Koenkk/zigbee2mqtt #3048: Issues re-pairing some hue motion sensors

mf-social, 2020-03-02:
## Bug Report

I had a load of hue motion sensors on a hue hub, but was bored of the delay in homeassistant so am using the zigbee2mqtt addon and a CC26X2R1.  Did a 'test run' whilst I read all the documentation and everything was fine, everything connected first time, no problems but signal was low  (wifi interference).

So armed with the knowledge of my test run I cleared all the old databases etc, purged the old entries from homeasistant/mqtt, uninstalled and re-installed the homeassistant addon, changed the channel and security key and off I went.  All the bulbs connected without a hitch, first motion sensor no problem.

Then....

## What happened

Second motion sensor I went to add kept 'failing' the interview.  Interestingly when it fails the interview, I don't seem to be able to get it to interview again.  I have to delete the most recent line from database.db (the one that tell me the interview failed) for it to interview again next time.  Eventually after 6 goes it pairs.

Tried the third one, same sort of issues but regardless, several attempts, just fails to interview every time.  Gave up with this one, tried the next

Went to pair the fourth motion sensor - same problem.  Getting grumpy now 😛 

Of note, these motion sensors haven't moved an inch from when they were on Hue Hub, and through the 'test run' time on zigbee2mqtt, but I just cannot get the damn things to pair and it's going to get dark soon and I'll be in trouble with the wife!

## What did you expect to happen

I expected to be able to pair the motion sensors.  And if they fail to pair the first time, I should be able to try again without having to manipulate the database.

## How to reproduce it (minimal and precise)

Come to my house and try and pair a motion sensor.

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

I've posted below a full log (including herdsman, debug and info) from restarting the addon at 14.07 hours, through to failing to interview the fourth sensor.  I hit the button at 14:08:59 and it starts the interview...

```
info  2020-03-02 14:07:43: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-02.14-07-43' filename: log.txt
debug 2020-03-02 14:07:43: Removing old log directory '/share/zigbee2mqtt/log/2020-03-02.10-22-22'
debug 2020-03-02 14:07:44: Loaded state from file /share/zigbee2mqtt/state.json
info  2020-03-02 14:07:44: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
info  2020-03-02 14:07:44: Starting zigbee-herdsman...
debug 2020-03-02 14:07:44: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-03-02 14:07:45: zigbee-herdsman started
info  2020-03-02 14:07:45: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
debug 2020-03-02 14:07:45: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":25}
info  2020-03-02 14:07:45: Currently 14 devices are joined:
info  2020-03-02 14:07:45: 0x00158d0001c377eb (0x00158d0001c377eb): RS 125 - Innr GU10 spot (Router)
info  2020-03-02 14:07:45: 0x00158d000211caec (0x00158d000211caec): RS 125 - Innr GU10 spot (Router)
info  2020-03-02 14:07:45: 0x00158d0001c3f4b9 (0x00158d0001c3f4b9): RS 125 - Innr GU10 spot (Router)
info  2020-03-02 14:07:45: 0x001788010328fe0e (0x001788010328fe0e): 9290012607 - Philips Hue motion sensor (EndDevice)
info  2020-03-02 14:07:45: 0x0017880102feb6dc (0x0017880102feb6dc): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x0017880102fe5088 (0x0017880102fe5088): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x00178801041d4438 (0x00178801041d4438): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x00178801023d4438 (0x00178801023d4438): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
info  2020-03-02 14:07:45: 0x00178801023d41b8 (0x00178801023d41b8): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
info  2020-03-02 14:07:45: 0x0017880102fc8a91 (0x0017880102fc8a91): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x00178801030a39b6 (0x00178801030a39b6): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x0017880102febb87 (0x0017880102febb87): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x0017880102fd3867 (0x0017880102fd3867): 8718696449691 - Philips Hue White Single bulb B22 (Router)
info  2020-03-02 14:07:45: 0x0017880103280b98 (0x0017880103280b98): 9290012607 - Philips Hue motion sensor (EndDevice)
info  2020-03-02 14:07:45: Zigbee: disabling joining new devices.
info  2020-03-02 14:07:45: Connecting to MQTT server at mqtt://10.0.0.68
info  2020-03-02 14:07:45: Connected to MQTT server
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0001c377eb', payload '{"state":"OFF","device":{"friendlyName":"0x00158d0001c377eb","model":"RS 125","ieeeAddr":"0x00158d0001c377eb","networkAddress":13813,"type":"Router","manufacturerID":4454,"manufacturerName":"innr","powerSource":"Mains (single phase)","applicationVersion":1,"stackVersion":2,"zclVersion":3,"hardwareVersion":1,"dateCode":"20161202-140","softwareBuildID":"2.1"}}'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x00158d000211caec', payload '{"state":"OFF","device":{"friendlyName":"0x00158d000211caec","model":"RS 125","ieeeAddr":"0x00158d000211caec","networkAddress":63952,"type":"Router","manufacturerID":4454,"manufacturerName":"innr","powerSource":"Mains (single phase)","applicationVersion":1,"stackVersion":2,"zclVersion":3,"hardwareVersion":1,"dateCode":"20161202-140","softwareBuildID":"2.1"}}'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0001c3f4b9', payload '{"state":"OFF","device":{"friendlyName":"0x00158d0001c3f4b9","model":"RS 125","ieeeAddr":"0x00158d0001c3f4b9","networkAddress":61966,"type":"Router","manufacturerID":4454,"manufacturerName":"innr","powerSource":"Mains (single phase)","applicationVersion":1,"stackVersion":2,"zclVersion":3,"hardwareVersion":1,"dateCode":"20161202-140","softwareBuildID":"2.1"}}'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x001788010328fe0e', payload '{"battery":58.5,"linkquality":78,"occupancy":false,"temperature":16.95,"illuminance":21430,"illuminance_lux":138,"device":{"friendlyName":"0x001788010328fe0e","model":"9290012607","ieeeAddr":"0x001788010328fe0e","networkAddress":20544,"type":"EndDevice","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20190219","softwareBuildID":"6.1.1.27575"}}'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x0017880102feb6dc', payload '{"state":"OFF","brightness":255,"device":{"friendlyName":"0x0017880102feb6dc","model":"8718696449691","ieeeAddr":"0x0017880102feb6dc","networkAddress":279,"type":"Router","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20181115","softwareBuildID":"1.46.13_r26312"}}'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x0017880102fe5088', payload '{"state":"OFF","brightness":255,"device":{"friendlyName":"0x0017880102fe5088","model":"8718696449691","ieeeAddr":"0x0017880102fe5088","networkAddress":27177,"type":"Router","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20181115","softwareBuildID":"1.46.13_r26312"}}'
info  2020-03-02 14:07:45: MQTT publish: topic 'zigbee2mqtt/0x00178801041d4438', payload '{"state":"OFF","brightness":1,"device":{"friendlyName":"0x00178801041d4438","model":"8718696449691","ieeeAddr":"0x00178801041d4438","networkAddress":19432,"type":"Router","manufacturerID":4107,"manufacturerName":"Ph [... cut ...] rID":4107,"manufacturerName":"Philips","powerSource":"Mains (single phase)","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20181115","softwareBuildID":"1.46.13_r26312"}}'
error 2020-03-02 14:09:49: Failed to interview '0x0017880102130541', device has not successfully been paired
info  2020-03-02 14:09:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x0017880102130541"}}'
```

mf-social, 2020-03-02:
Oh arse, I've just realised now I've pasted it in that the herdsman stuff isn't in that log 🙄 

How do I get that?  The log in the addon screen shows it intertwined with the above log (which is what I thought it would output in to the file, but apparently not!) but only shows like 100 lines at a time so I can't capture all of it 😢

ccorderod, 2020-03-02:
@mf-social Try: DEBUG=zigbee-herdsman* npm start > log.txt 2>&1. Note you must stop zigbee2mqtt before (with sudo systemctl stop zigbee2mqtt)

mf-social, 2020-03-02:
As I said, I'm using the homeassistant addon so that isn't an option.  Adding "zigbee_herdsman_debug": true to the configuration shows the zigbee herdsman logs in the logging pane for the addon, along with the info and debug logs as above, but it doesn''t save the output to the logfile, annoyingly.

Koenkk, 2020-03-02:
Make sure to turn the hue network off (hub and devices in it).

mf-social, 2020-03-02:
Yeah, hue device has been switched off and in a box since this experiment started.

I've kept hammering away at it 4 and 5 are now connected, I'm just about to try 6 and then I'll be back to 3!!  Maybe it's just a persistence issue on my part 🤷‍♂ - but 7 or 8 goes to pair something is a bit frustrating, and it would be nice if it gave a bit more human readable info on why it failed the interview if such a thing is possible 🙂 

Also, whilst I have you scratching your head at this, how come the 'illuminance' sensor initially passes the 'illuminance' value from the sensors, but after a while passes the lux value instead??  Obviously it's the lux that I wanted, but it's weird it passes the raw figure for the first few updates (possibly until a restart of the addon/homeassistant/both).

mf-social, 2020-03-02:
OK - 6 went on first time 🎉 - back to 3, which is still refusing to pair...

mf-social, 2020-03-02:
OK

NOBODY BREATHE!!!

They're all back on - I lost count of how many times I tried, but after A LOT of goes, we're all paired up!

Would still be interested in any ideas why it took so many goes, and also my query re the light level readings that come through?

mf-social, 2020-03-02:
Ugh - hold the phone...

So, two of them have decided to get stuck and haven't provided an update for a while - interestingly one of them is number 3 (the one that I struggled to pair the most), and the other is one that paired straight away.  I can see that they are still detecting motion, but one of them last updated to 'off' about 10 minutes ago, and the other got stuck in 'on' about 4 minutes ago.

Any thoughts?  Anything I can do to get them back??

mf-social, 2020-03-02:
A further excerpt from the diary of my zigbee2mqtt journey today. Sensor 3 has magically come back online.

I'm not sure about the other one yet, and I'm not going to tempt fate by walking upstairs to try it. 

Curious how to debug what's occurring here if anyone can guide me please? 

Is it likely a manifestation of this... 

https://github.com/Koenkk/zigbee2mqtt/issues/2693

?

Koenkk, 2020-03-02:
You will indeed need the firmware from: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin, otherwise Zigbee2mqtt won't be reachable anymore when it the motion sensor switches parent.

mf-social, 2020-03-02:
OK, will flash the new one in the morning - fingers crossed 😁

mf-social, 2020-03-03:
Well - everything mostly worked for the last 24 hours - just flashed the new firmware (which according to the website should not require re-pairing of anything), and now my log is just filled with

```
zigbee2mqtt:info  2020-03-03 19:24:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x0017880102fe5088' failed: 'Error: Command 0x0017880102fe5088/11 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x0017880102fe5088"}}'
zigbee2mqtt:error 2020-03-03 19:24:56: Failed to execute LQI for '0x00178801030a39b6'
```

And I cannot control anything at all.  And it's dark.

Going to be honest, this is not the best experience I've ever had with any software.

mf-social, 2020-03-03:
OK - if the Github clock is to be believed I've managed to re-pair everything in 31 minutes, which wasn't as bad as I was expecting in the dark!  Everything went on first time (except one of the dimmer switches that took a few goes, even though it's actually the closest one to the coordinator 😕 ) which is the exact opposite of the experience that made me generate this thread in the first place.

Of course once re-paired they were already set up in homeassistant, so I didn't have to do anything else - which is nice.

I'll close this now, but I'll be apprehensive about flashing updated firmware again and definitely make sure it's daylight!!

Thanks for reading 🙂

closed 2020-03-03 by mf-social

Koenkk, 2020-03-03:
A no network route can indeed happen, I believe a simple repower if the bulb is enough for the coordinator to retrieve a new route. (then it should work again).