Koenkk/zigbee2mqtt #3049: Pairing message on zigbee2mqtt/bridge/config not always published

claudio-petrini, 2020-03-02:
Hello,
I'd like to subscribe to `zigbee2mqtt/bridge/config` topic in order to retrieve the `device_connected` event type.

The first time the device is paired the library pushes to this topic, here are the logs:
```
zigbee2mqtt:info  2020-03-02 16:42:08: Starting touchlink factory reset...
zigbee2mqtt:info  2020-03-02 16:42:12: Successfully factory reset device through Touchlink
zigbee2mqtt:info  2020-03-02 16:42:13: Device '0x0017880103fa5754' joined
zigbee2mqtt:info  2020-03-02 16:42:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880103fa5754"}}'
zigbee2mqtt:info  2020-03-02 16:42:13: Starting interview of '0x0017880103fa5754'
zigbee2mqtt:info  2020-03-02 16:42:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880103fa5754"}}'
zigbee2mqtt:info  2020-03-02 16:42:14: Successfully interviewed '0x0017880103fa5754', device has successfully been paired
zigbee2mqtt:info  2020-03-02 16:42:14: Device '0x0017880103fa5754' is supported, identified as: Philips Hue white and color ambiance E26/E27/E14 (9290012573A)
zigbee2mqtt:info  2020-03-02 16:42:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0017880103fa5754","model":"9290012573A","vendor":"Philips","description":"Hue white and color ambiance E26/E27/E14","supported":true}}'
```

After removing the device and executing the touchlink procedure again no new messages are published to this topic. It seems like the device is still in the network.

Thank you

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: Z-Stack_Home_1.2 (default)
--

Koenkk, 2020-03-02:
How did you remove the device?

claudio-petrini, 2020-03-02:
I've tried both:
- by publishing a message on `zigbee2mqtt/bridge/config/remove` topic
- by removing the entry from `database.db` and `configuration.yaml` and restarting the bridge

I'm using a Docker container if it can help.
I'll be happy to provide additional details if needed, thank you again for your assistance.

Koenkk, 2020-03-03:
Unfortunately I could not reproduce it:

```
zigbee2mqtt:info  2020-03-03 20:50:06: Removing '0x00158d0001c378a9'
zigbee2mqtt:warn  2020-03-03 20:50:06: Device '0x00158d0001c378a9' left the network
zigbee2mqtt:info  2020-03-03 20:50:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00158d0001c378a9"}}'
zigbee2mqtt:info  2020-03-03 20:50:25: Starting touchlink factory reset...
zigbee2mqtt:info  2020-03-03 20:50:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_started","meta":{"status":"started"}}'
zigbee2mqtt:info  2020-03-03 20:50:28: Successfully factory reset device through Touchlink
zigbee2mqtt:info  2020-03-03 20:50:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"touchlink","message":"reset_success","meta":{"status":"success"}}'
zigbee2mqtt:info  2020-03-03 20:50:38: Device '0x00158d0001c378a9' joined
zigbee2mqtt:info  2020-03-03 20:50:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0001c378a9"}}'
zigbee2mqtt:info  2020-03-03 20:50:38: Starting interview of '0x00158d0001c378a9'
zigbee2mqtt:info  2020-03-03 20:50:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0001c378a9"}}'
zigbee2mqtt:info  2020-03-03 20:50:39: Successfully interviewed '0x00158d0001c378a9', device has successfully been paired
zigbee2mqtt:info  2020-03-03 20:50:39: Device '0x00158d0001c378a9' is supported, identified as: Innr E27 bulb RGBW (RB 185 C)
zigbee2mqtt:info  2020-03-03 20:50:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00158d0001c378a9","model":"RB 185 C","vendor":"Innr","description":"E27 bulb RGBW","supported":true}}'
```

Note that for the bulb I needed to re-power it in order to have it join the network again.

claudio-petrini, 2020-03-05:
Hello @Koenkk, thank you again for your support.
I'm unable to reproduce this systematically but by re-powering the bulb always works properly.

I'm closing the issue for now and will reopen it if I can provide additional feedback, thank you again.

closed 2020-03-05 by claudio-petrini