Koenkk/zigbee2mqtt #3050: Cannot remove devices from the network

chiarabiancacci, 2020-03-02:
Hello!
After having successfully paired a device, I'm unable to remove it from the network since it seems that it does not exist.

These are the output log I get after pairing and trying to remove it:

``` 
zigbee2mqtt:info  2020-03-02 16:59:31: Currently 1 devices are joined:
zigbee2mqtt:info  2020-03-02 16:59:31: 0x0017880103fa5754 (0x0017880103fa5754): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
zigbee2mqtt:warn  2020-03-02 16:59:31: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-03-02 16:59:31: Allowing new devices to join.
zigbee2mqtt:warn  2020-03-02 16:59:31: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-03-02 16:59:31: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-03-02 16:59:31: Connecting to MQTT server at mqtt://mosquitto
zigbee2mqtt:info  2020-03-02 16:59:31: Connected to MQTT server
zigbee2mqtt:info  2020-03-02 16:59:31: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-03-02 16:59:31: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0","commit":"31e5678","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
zigbee2mqtt:error 2020-03-02 16:59:58: Cannot force remove, device '{"friendly_name":"0x0017880103fa5754"}' does not exist
```
In order to remove the device I publish a message on `zigbee2mqtt/bridge/config/force_remove` with the following payload: `{ “friendly_name“: “0x0017880103fa5754“ }`.

How is this possible? Can you help me?
Thank you 

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: Z-Stack_Home_1.2 (default)

hwijers, 2020-03-02:
mosquitto_pub -h 192.168.1.x -t zigbee2mqtt/bridge/config/force_remove -m '0x0017880103fa5754'

Koenkk, 2020-03-02:
As @hwijers mentioned, the payload is indeed not correct instead of `{"friendly_name":"0x0017880103fa5754"}` it should be `0x0017880103fa5754`

closed 2020-03-02 by Koenkk