Koenkk/zigbee2mqtt #3052: mgmtPermitJoinReq when removing trådfri bulb that was in a group.

jsassner, 2020-03-02:
I had an bulb that was the wrong size, sticking out of the lamp holder, so i wanted to swap it out for a less high bulb. I swapped a trådfri for another model of trådfri.

mosquitto_pub -t zigbee2mqtt/bridge/config/remove -m Lamp1_Matrum_TF_Home
When i executed that command, i got the following error:
```
zigbee2mqtt:info  2020-03-02 19:35:15: Removing 'Lamp1_Matrum_TF_Home'
zigbee2mqtt:warn  2020-03-02 19:35:15: Device '0xccccccfffe620ef8' left the network
zigbee2mqtt:info  2020-03-02 19:35:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xccccccfffe620ef8"}}'
zigbee2mqtt:warn  2020-03-02 19:35:16: Device '0xccccccfffe620ef8' left the network
zigbee2mqtt:info  2020-03-02 19:35:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xccccccfffe620ef8"}}'
zigbee2mqtt:error 2020-03-02 19:35:25: Failed to remove Lamp1_Matrum_TF_Home (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
zigbee2mqtt:error 2020-03-02 19:35:25: See https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove for more info
zigbee2mqtt:info  2020-03-02 19:35:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed_failed","message":"Lamp1_Matrum_TF_Home"}'
(node:15024) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (d:\Services\zigbee2mqtt\node_modules\zigbee-herdsman\dist\utils\waitress.js:44:24)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:15024) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:15024) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
And now zigbee2mqtt just locked down, no more messages in the output window appeared, and i had to restart it to work.

What might be the problem here was that the lamp i tried to remove was part of a group, and i hadn't removed it from the group before i tried to remove the lamp with the command shown above.

zigbee2mqtt version: 1.11.0 (commit #31e5678)
CC253X firmware version: 20190608

Koenkk, 2020-03-02:
This indicated that the CC2531 firmware crashed, https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing should increase stability.

jsassner, 2020-03-02:
Ok. Thanks, will try it out.

closed 2020-03-02 by jsassner