Koenkk/zigbee2mqtt #3054: Xiaomi ZNCZ02LM - Power On State Error

johnjoemorgan, 2020-03-03:
# Bug Report
## What happened
Want to retain power setting after power outage.  Plug is working as it should with power toggle working and power level and signal being reported.  Sent MQTT topic:

zigbee2mqtt/Xiaomi Plug/set

with payload

{"power_outage_memory": true}

## What did you expect to happen
Power state retain after power outage
## How to reproduce it (minimal and precise)
Log gave error:
zigbee2mqtt:error 2020-03-03T04:20:21: No converter available for 'power_outage_memory' (true)

## Debug Info
zigbee2mqtt version: 
1.7.1
CC253X firmware version:

tricoos, 2020-03-03:
That's a pretty old version of Zigbee2MQTT - the current version is 1.11.0 - maybe you want to try again with a more recent version?  It might not have been supported in 1.7.0 (just a guess).

ryanbeaton, 2020-03-03:
Yeah it was a v1.9.0 feature
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.9.0

johnjoemorgan, 2020-03-03:
OK Running in Hass.io.  Will try to force an update.

johnjoemorgan, 2020-03-03:
Well, won't ya look at that?!  Somehow I must have deleted the repo from the Supervisor Store.  Placed it back updated and works like a charm.  Apologies for not doing due diligence and thank you for leading me to a fix :-)

closed 2020-05-09 by stale[bot]

dreem2001, 2021-05-19:
Hello. I have the same issue on HASSIO zigbee2mqtt 1.18.1 add-on and Xiaomi ZNLDP12LM
Any ideas how to check\debug why?
Thanks.

timurlutfullin, 2024-11-30:
Тоже столкнулся с такой проблемой, под рукой не было шлюза для обновления. Обновил через zigbee2mqtt подсунув нужную прошивку.
Имеем zncz02lm прошивка 22 от 02-28-2017 штатно постоянно выдает ошибку по OTA с 22 на 92, через MiHome обновлений не видит. 
Надо обновить на промежуточную 30 прошивку. Обсуждение тут https://github.com/Koenkk/zigbee2mqtt/issues/6906
Синтаксис локального обновления тут https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
Все делаем через FileEditor
1. В настройках /homeassistant/zigbee2mqtt/configuration.yaml добавляем строки
ota:
    zigbee_ota_override_index_location: ota.json
2. Создаем /homeassistant/zigbee2mqtt/ota.json и пишем там
[
    {
        "url": "LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota",
        "force": true
    }
]
3. В  /homeassistant/zigbee2mqtt заливаем прошивку распакованную прошивку  https://github.com/Koenkk/zigbee2mqtt/files/10374315/LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota.zip
4. Полностью перегружаем систему, а не только HA. 
5. Видим что zigbee2mqtt теперь предлагает обновиться с 22 на 30. Обновляемся.
6. Откатываем пункты 1, 2, 3 и полностью перезагружаемся.
7. Теперь уже zigbee2mqtt предлагает нам обновиться с 30 на 92, штатно обновляемся.