Koenkk/zigbee2mqtt #3055: Newly added Innr RB 248 T - errors when publishing a command

thehijacker, 2020-03-03:
# Bug Report

I added two Innr RB 238 T lightbulbs in my bedroom (for nightlamps). In bedroom I already have an Osram Smart+ bulb that was working fine. Now I see some issues.

## What happened

I am trying to turn on/off the bulbs from Home Assistant using Mosquito MQTT broker.

## What did you expect to happen

I expect that light will turn on/off, change brightness as soon as I make action in Home Assistant.

## How to reproduce it (minimal and precise)

I can reproduce this always when trying to do multiple actions on the bulbs.

## Debug Info
zigbee2mqtt version: 1.10.0-dev
CC253X firmware version: CC2530_CC2591_SOURCE_ROUTING_20190619

The mesh network should be fine. I have many Osram Smart+ bulbs all over the house on both floors acting as router. I also have two additional CC2530+CC2591 routers around the house. Link quality should be great for all devices yet it shows under 20 for some devices on top floor.

This are the error messages I got from today. In morning when I woke up I tried to turn on/off the Innr nightlights but they did not turn on or only one of them did.

`error 2020-03-03 00:00:18: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-03 01:00:21: Failed to execute LQI for 'Bulb Nightlamp Andrej'
error 2020-03-03 02:00:28: Failed to ping 'Bulb Bedroom'
error 2020-03-03 02:00:40: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-03 02:05:41: Failed to ping 'Bulb Nightlamp Andrej'
error 2020-03-03 03:03:23: Failed to ping 'Bulb Bedroom'
error 2020-03-03 03:09:12: Failed to ping 'Bulb Nightlamp Simona'
error 2020-03-03 04:00:16: Failed to execute LQI for 'Bulb Bedroom'
error 2020-03-03 04:00:27: Failed to execute LQI for 'Bulb Hallway'
error 2020-03-03 05:00:16: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-03 05:00:26: Failed to execute LQI for 'Bulb Nightlamp Andrej'
error 2020-03-03 05:45:53: Publish 'set' 'state' to 'Bulb Nightlamp Simona' failed: 'Error: Command 0x00158d0004ae9d02/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 58185 - 1 - 149 - 6 - 11 after 15000ms)'
info  2020-03-03 05:45:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Bulb Nightlamp Simona' failed: 'Error: Command 0x00158d0004ae9d02/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 58185 - 1 - 149 - 6 - 11 after 15000ms)'","meta":{"friendly_name":"Bulb Nightlamp Simona"}}'
error 2020-03-03 05:45:53: Publish 'set' 'state' to 'Bulb Nightlamp Andrej' failed: 'Error: Command 0x00158d0004ae3c7b/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 30813 - 1 - 150 - 6 - 11 after 15000ms)'
info  2020-03-03 05:45:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Bulb Nightlamp Andrej' failed: 'Error: Command 0x00158d0004ae3c7b/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 30813 - 1 - 150 - 6 - 11 after 15000ms)'","meta":{"friendly_name":"Bulb Nightlamp Andrej"}}'
error 2020-03-03 06:06:01: Publish 'set' 'state' to 'Bulb Hallway' failed: 'Error: Command 0x7cb03eaa00b101cb/3 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 42179 - 3 - 6 - 6 - 11 after 15000ms)'
info  2020-03-03 06:06:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Bulb Hallway' failed: 'Error: Command 0x7cb03eaa00b101cb/3 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 42179 - 3 - 6 - 6 - 11 after 15000ms)'","meta":{"friendly_name":"Bulb Hallway"}}'
error 2020-03-03 06:06:04: Publish 'set' 'state' to 'Bulb Hallway' failed: 'Error: Command 0x7cb03eaa00b101cb/3 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
info  2020-03-03 06:06:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Bulb Hallway' failed: 'Error: Command 0x7cb03eaa00b101cb/3 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"Bulb Hallway"}}'
error 2020-03-03 07:00:15: Failed to execute LQI for 'Bulb Nightlamp Andrej'`

I had issues before with Osram bulbs. Got lot of "MAC no ack". With latest dev firmware they reduced significantly but I can still see them (Bulb Hallway).

Compared to all my Tasmota devices and my bad WiFi mesh network, zigbee2mqtt proved to be much more reliable and stable. Just this little issues I am having lately.

Thank you.

[image]
[image]

thehijacker, 2020-03-03:
I think this is same issue mentioned in issue #2802 

This are all the error messages from this morning:

error 2020-03-03 10:41:01: Failed to execute LQI for 'Bulb Nightlamp Andrej'
error 2020-03-03 11:27:37: Failed to ping 'Bulb Nightlamp Simona'
error 2020-03-03 12:41:46: Failed to ping 'Bulb Nightlamp Simona'
error 2020-03-03 12:42:47: Failed to ping 'Bulb Nightlamp Andrej'
error 2020-03-03 13:00:14: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-03 13:20:36: Failed to ping 'Bulb Nightlamp Simona'
error 2020-03-03 13:23:45: Failed to ping 'Bulb Nightlamp Simona'
error 2020-03-03 14:00:15: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-03 14:38:53: Failed to ping 'Bulb Nightlamp Simona'

Koenkk, 2020-03-04:
Can you try upgrading to Zigbee2mqtt 1.11.0, it contains some improvements related to these errors.

thehijacker, 2020-03-05:
Hello Koenkk.

I am n 1.11.0-dev from yesterday. I can still see the errors. Most of them at full hour when my map is generated via automation taks.

```
error 2020-03-05 02:00:17: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-05 03:00:14: Failed to execute LQI for 'Bulb Bedroom'
error 2020-03-05 03:00:26: Failed to execute LQI for 'Bulb Nightlamp Simona'
error 2020-03-05 03:00:38: Failed to execute LQI for 'Bulb Nightlamp Andrej'
error 2020-03-05 05:00:14: Failed to execute LQI for 'Bulb Bedroom'
error 2020-03-05 05:00:27: Failed to execute LQI for 'Bulb Nightlamp Andrej'
error 2020-03-05 06:00:11: Failed to execute LQI for 'CC2530 Router Downstairs'
```

Both nightlamp bulbs are Innr from the topic. Bedroom light is Osram Smart+ bulb also acting as router.

Koenkk, 2020-03-05:
LQI errors are a different issue (#2802), besides that they are working fine?

thehijacker, 2020-03-05:
I have a button (WXKG11LM) that will toggle both nightlamps on double click. Sometimes they turn on at same time, sometimes only one and second one with up to 3-5 second delay.

Network looks OK. I have many routers. On coordinator I have source routing firmware since coordinator doesn't need to have that many devices connected to it when there are over 8 router devices.

In 90% case everything is stable but in 10% things are not turning on/off on command. I disabled automation generate network map. Last error message was at:

`error 2020-03-05 07:47:54: Failed to execute LQI for 'Bulb Upstairs'`

Currently no one is home but over evening lots of zigbee devices fire (many RF PIR sensors), from HA automations. I suspect when there is increase of traffice things are starting to fail.

Issues started with adding this two Innr bulbs. Did not notice any problems until then. Could be just my observation. Having many routers is good thing or overkill for single floor house?

LQI value for devices in HA is interesting. For some device they show under 20 LQI when I know they should have better signal. Not sure what value they use.

Koenkk, 2020-03-05:
- Do you use the availability feature? (can you post your `configuration.yaml`)
- Can you post the log at the moment things fail?

thehijacker, 2020-03-05:
My configuration.yaml content.

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: '!secret user'
  password: '!secret password'
serial:
  port: /dev/ttyUSB.CC2530-01
devices: devices.yaml
advanced:
  availability_timeout: 0
  report: false
  rtscts: false 
```

Log with 10 lines before/after this afternoon errors is here:

https://pastebin.com/ZP1vMZu7

Three so far:

error 2020-03-05 16:00:15: Failed to execute LQI for 'Bulb Bedroom'
error 2020-03-05 16:00:26: Failed to execute LQI for 'Bulb Nightlamp Andrej'
error 2020-03-05 18:18:06: Publish 'set' 'state' to 'Bulb Hallway' failed: 'Error: Command 0x7cb03eaa00b101cb/3 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed
with error: 'MAC no ack' (233))'

Koenkk, 2020-03-05:
Looking at your OP now, the linkquality of this bulb is only 10? This is very low, does the bulb work correctly when having it close to the coordinator?

thehijacker, 2020-03-06:
It on top floor, while coordinator is on bottom floor of the house. There are routers (bulbs) in between coordinator and this "problematic" bulb. For what connection is the LQI info? Connection to coordinator or furthest connection to first routers?

Koenkk, 2020-03-08:
My assumption is the direct parent, but the network map should show you this.

thehijacker, 2020-03-08:
Realy odd. In map I can see that there are zero quality for the problematic bulbs. But they work.

[image]

I am thinking on pairing them all once again. To delete database and start from zero. Is there a suggested way how to do it? Do I need to bring them all close to coordinator (source router firmware) or I can do router by router from one closes to coordinator and then next one closest to last paired router?

When all routers are up I would join all end devices and hope they will join to closes router.

To make sure all is really fresh I would change pan_id and network_key.

Can you confirm this is a good plan or can you suggest a better way? Currently I have 32 devices (12x routers - Osram bulbs, two extra CC2530 routers, Innr bulbs and Xiaomi bubls).

Thank you.

Koenkk, 2020-03-09:
I don't think that repairing the complete network makes sense (and is unlikely to fix the problem), maybe you can try only repairing the bulb itself, but still if the weak linkquality is the issue this won't fix the problem. In that case you need to improve your network range (https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html)

thehijacker, 2020-03-12:
@Koenkk I have CC2530+CC2591 soldered to CP2102 as coordinator running source routing firmware (32 devices in total). I tried to flash it to default firmware using SerialBootTool. It opens COM port fine and I can see LED blinking but it always fails with handshake error. So I can not flash it like that?

Another question. When I turn on bulb using HA, the command repeats three times:

```
info  2020-03-12 20:45:58: MQTT publish: topic 'zigbee2mqtt/Bulb Hallway', payload '{"state":"ON","linkquality":23,"brightness":216,"color_temp":260,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:45:58: MQTT publish: topic 'zigbee2mqtt/Bulb Hallway', payload '{"state":"ON","linkquality":23,"brightness":216,"color_temp":260,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:45:58: MQTT publish: topic 'zigbee2mqtt/Bulb Hallway', payload '{"state":"ON","linkquality":23,"brightness":216,"color_temp":260,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:46:01: MQTT publish: topic 'zigbee2mqtt/Bulb Hallway', payload '{"state":"OFF","linkquality":23,"brightness":216,"color_temp":260,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:46:01: MQTT publish: topic 'zigbee2mqtt/Bulb Hallway', payload '{"state":"OFF","linkquality":23,"brightness":216,"color_temp":260,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:46:01: MQTT publish: topic 'zigbee2mqtt/Bulb Hallway', payload '{"state":"OFF","linkquality":23,"brightness":216,"color_temp":260,"color":{"x":0.3805,"y":0.3769},"update_available":false}' 
``` 

But not for some bulbs:

```
info  2020-03-12 20:50:43: MQTT publish: topic 'zigbee2mqtt/Bulb Middle', payload '{"state":"ON","linkquality":15,"brightness":254,"color_temp":286,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:50:43: MQTT publish: topic 'zigbee2mqtt/Bulb Middle', payload '{"state":"OFF","linkquality":15,"brightness":254,"color_temp":286,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:50:44: MQTT publish: topic 'zigbee2mqtt/Bulb Middle', payload '{"state":"ON","linkquality":15,"brightness":254,"color_temp":286,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:50:45: MQTT publish: topic 'zigbee2mqtt/Bulb Middle', payload '{"state":"OFF","linkquality":15,"brightness":254,"color_temp":286,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:50:45: MQTT publish: topic 'zigbee2mqtt/Bulb Middle', payload '{"state":"ON","linkquality":15,"brightness":254,"color_temp":286,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
info  2020-03-12 20:50:46: MQTT publish: topic 'zigbee2mqtt/Bulb Middle', payload '{"state":"OFF","linkquality":15,"brightness":254,"color_temp":286,"color":{"x":0.3805,"y":0.3769},"update_available":false}'
```

Same bulb model. Osram Smart+.

With Aqara bulb this is repeated twice:

```
info  2020-03-12 20:52:23: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"ON","brightness":190,"color_temp":243,"linkquality":44}'
info  2020-03-12 20:52:23: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"ON","brightness":190,"color_temp":243,"linkquality":46}'
info  2020-03-12 20:52:25: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"ON","brightness":190,"color_temp":243,"linkquality":46}'
info  2020-03-12 20:52:25: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"OFF","brightness":190,"color_temp":243,"linkquality":46}'
info  2020-03-12 20:52:25: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"OFF","brightness":190,"color_temp":243,"linkquality":44}'
info  2020-03-12 20:52:26: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"OFF","brightness":190,"color_temp":243,"linkquality":44}'
info  2020-03-12 20:52:26: MQTT publish: topic 'zigbee2mqtt/Stairways Left', payload '{"state":"OFF","brightness":190,"color_temp":243,"linkquality":44}'
info  2020-03-12 20:52:26: MQTT publish: topic 'zigbee2mqtt/Stairways Right', payload '{"state":"ON","brightness":158,"color_temp":225,"linkquality":44}'
info  2020-03-12 20:52:26: MQTT publish: topic 'zigbee2mqtt/Stairways Right', payload '{"state":"ON","brightness":158,"color_temp":225,"linkquality":36}'
info  2020-03-12 20:52:27: MQTT publish: topic 'zigbee2mqtt/Stairways Right', payload '{"state":"OFF","brightness":158,"color_temp":225,"linkquality":36}'
info  2020-03-12 20:52:27: MQTT publish: topic 'zigbee2mqtt/Stairways Right', payload '{"state":"OFF","brightness":158,"color_temp":225,"linkquality":39}'
info  2020-03-12 20:52:28: MQTT publish: topic 'zigbee2mqtt/Stairways Right', payload '{"state":"OFF","brightness":158,"color_temp":225,"linkquality":39}'
info  2020-03-12 20:52:28: MQTT publish: topic 'zigbee2mqtt/Stairways Right', payload '{"state":"OFF","brightness":158,"color_temp":225,"linkquality":39}'
```

My current configuration is:

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: '!secret user'
  password: '!secret password'
serial:
  port: /dev/ttyUSB.CC2530-01
devices: devices.yaml
advanced:
  availability_timeout: 60
  report: true
  rtscts: false
```

Is this normal?

Koenkk, 2020-03-14:
- I've never flashed through SerialBootTool, so unfortunately no experience with that
- Regarding the messages, there are indeed a lot of duplicated, wondering where they come from could you enable debug logging and provide the log again?
- Regarding your configuration, please remove `availability_timeout` as it puts a lot of extra stress on the network

thehijacker, 2020-03-14:
I enabled debug log:

This is log.txt content:

http://freeweb.t-2.net/hijacker/log.txt

And here is SSH log messages that I got:

http://freeweb.t-2.net/hijacker/ssh.log

I turned on/off each of the bulbs at least once.

Koenkk, 2020-03-15:
These messages come from reports of the device because of `report: true` (so nothing strange).

thehijacker, 2020-03-15:
Today I took an hour and deleted database.db with new pan_id, network_key and changed channel to 26. They I repaired everything one by one.

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: '!secret user'
  password: '!secret password'
serial:
  port: /dev/ttyUSB.CC2530-01
devices: devices.yaml
advanced:
  pan_id: 0xXXXX
  channel: 26
  network_key: '!secret network_key'
  rtscts: false
```

Here are logs:

http://freeweb.t-2.net/hijacker/zigbee/log.txt
http://freeweb.t-2.net/hijacker/zigbee/ssh.txt

You can still see some errors. Can you tell where they came from?

For upstairs router I am using CC2530 with antenna. In ground floor I have CC2530+CC2591 antenna. Today I have ordered material for another CC2530+CC2591 to replace CC2530 only.

I hope this will increase some signal at top floor. I was keep thinking that having many Zigbee bulbs will make great reception over whole house. But it seems this is not true.

Koenkk, 2020-03-16:
- Regarding LQI, there is already an issue about this (#2802), however this is not really a big issue as its only used for the network map
- The bind errors: make sure to wake the device up when the configure happens, this only has to succeed one time and you can also trigger it manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

thehijacker, 2020-03-16:
I tried with configure of devices for which I got binding error but I always get:

warn  2020-03-16 19:36:15: Skipping configure of 'Stairways Left', device does not require this.
warn  2020-03-16 19:36:31: Skipping configure of 'Stairways Right', device does not require this.

I will wait for parts to assemble one more AC powered router using CC2530+CC2591 to replace the only CC2530 with antenna on the top floor. Should have better reception than only CC2530. The existing CC2530+CC2591 is working hard looking at the map. Most of the end device are connected to it. And it is in a nice looking box hidden on the ceiling.

[image]

closed 2020-05-22 by stale[bot]