Koenkk/zigbee2mqtt #3058: New setup: Devices are paired but don't show up in HA outside of the logs

dknia, 2020-03-03:
Summary: My zigbee devices doesn't show up on my overview page - help me diagnosing the issue (it's probably simple).

I have an almost virgin installation of Home Assistant in Docker on unRAID. Everything is updated to latest stable versions.

I have installed firmware updated  TI CC1352P-2. It has successfully found three different devices, that all should be known up front by zigbee2mqtt, and are according to the log registered correctly. 
Keypresses on the Ikea tradfri Remote Control are registered in the log.
The dimmer Nordtronic Box Dimmer 2.0 also registers on/off and dim events in the log (operating it manually though a physical switch)
The Ikea bulb is connected visibly in the log

The problem: None of these appear in the Overview page as neither sensors nor lights, or anywhere else I can think of looking. 
I have tried manually configuring them, but that didn't help (maybe I did something wrong). 

The only trace of the devices except the log entries is these lines automatically inserted into devices.yaml:
```
'0x680ae2fffe3a4533':
  friendly_name: '0x680ae2fffe3a4533'
'0x680ae2fffe36caec':
  friendly_name: '0x680ae2fffe36caec'
'0xec1bbdfffee5d291':
  friendly_name: '0xec1bbdfffee5d291'
```

zigbee2mqtt Config https://hastebin.com/cakasimage.makefilezigbee2mqtt 
LOG https://hastebin.com/asifozuyat.sql
configuration.yaml https://hastebin.com/lozofawaya.bash (I have tried moving :1883 to 'port: 1883' to no avail)

How can I get these devicesactive and into the Overview page so I can start using them through HA?

TIA for any input

Koenkk, 2020-03-04:
Your zigbee2mqtt config looks good, if also Home Assistant is configured correctly (https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery) you should be able to find the sensors under: Configuration -> Integrations -> MQTT

dknia, 2020-03-04:
Thanks for the response :-)

On that page, it says "This integration has no devices." 

Is there anything i should check with regards to the MQTT setting? I have reset the installation of the Mosquitto broker to default but to no avail...

Koenkk, 2020-03-05:
@dknia make sure to check to home assistant configuration.yaml: https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery

dknia, 2020-03-05:
I think it's as it should be - they are available in the links in the OP to check.

The log from the Mosquitti broker is here:
https://hastebin.com/ajirajazav.sql

Koenkk, 2020-03-05:
Does anything appear in the Home Assistant log when starting Zigbee2mqtt?

dknia, 2020-03-06:
No - I restarted the zigbee2mwtt at 11:59, nothing in the log since 09:56.
Logfile as it looks right now: https://hastebin.com/qetorewebi.sql

closed 2020-05-13 by stale[bot]

dknia, 2020-06-09:
I ended up migrating to a HassOS VM, where everything works great :-)