Koenkk/zigbee2mqtt #3059: Philips Hue GU10 Color Ambiance 929001953101: Interview failed

tobimori, 2020-03-03:
# Bug Report
## What happened

The interview of the Philips Hue GU10 Color Ambiance 929001953101 failed.

```
zigbee2mqtt:info  2020-03-03 15:58:46: Starting interview of '0x0017880106b83ab7'
zigbee2mqtt:error 2020-03-03 15:59:46: Failed to interview '0x0017880106b83ab7', device has not successfully been paired
```

## How to reproduce it (minimal and precise)

Come to my flat, try to reset a lamp with the Hue Dimmer and wait.

## Debug Info
zigbee2mqtt version: 1.11.0
CC253X firmware version: `'{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`

Koenkk, 2020-03-04:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair ?

tobimori, 2020-03-04:
Sure, this happens with all three lamps.

Providing debug log asap. Thanks :)

tobimori, 2020-03-05:
Closing. Fixed by reinserting CC2531 USB and restarting zigbee2mqtt. Sorry for disturbing and thanks for your help :)

closed 2020-03-05 by tobimori