Koenkk/zigbee2mqtt #3060: Error during zigbee2mqtt install

blackskull182, 2020-03-03:
# Bug Report
## What happened

Hello, I'm trying to install zigbee2mqtt on Armbian Buster in a Python Virtual Enviroment (but I've tried also to install directly on the machine without the VE) following the documentation. This is the error that I get:

`(zigbee2mqtt) mqttuser@orangepizero:/opt/zigbee2mqtt$ npm ci

> @serialport/bindings@8.0.7 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.15.1 runtime=node arch=arm libc= platform=linux)
gyp ERR! configure error 
gyp ERR! stack Error: Command failed: /opt/zigbee2mqtt/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                                ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack 
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:294:12)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:970:16)
gyp ERR! stack     at Socket.stream.socket.on (internal/child_process.js:389:11)
gyp ERR! stack     at Socket.emit (events.js:189:13)
gyp ERR! stack     at Pipe._handle.close (net.js:600:12)
gyp ERR! System Linux 5.4.20-sunxi
gyp ERR! command "/opt/zigbee2mqtt/bin/node" "/opt/zigbee2mqtt/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.1
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/mqttuser/.npm/_logs/2020-03-03T16_31_51_564Z-debug.log`

I also attached the log file.

[2020-03-03T16_31_51_564Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/4282544/2020-03-03T16_31_51_564Z-debug.log)

Thanks a lot

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

robertcsakany, 2020-03-04:
I have the very same error on orangePI with armbian/

robertcsakany, 2020-03-04:
For me the install of python 2.7 resolved the problem.

`sudo apt-get install python2.7`

closed 2020-05-10 by stale[bot]

CyberLine, 2020-05-11:
I have the same issue.

pydog, 2020-06-29:
try to run on Macos, get the similar failure.
prebuild-install WARN install No prebuilt binaries found (target=14.4.0 runtime=node arch=x64 libc= platform=darwin)
No receipt for 'com.apple.pkg.CLTools_Executables' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLILeo' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLI' found at '/'.

gyp: No Xcode or CLT version detected!
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/cnpm/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Darwin 19.5.0
gyp ERR! command "/usr/local/Cellar/node/14.4.0/bin/node" "/usr/local/lib/node_modules/cnpm/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v14.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@8.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

pydog, 2020-06-29:
> try to run on Macos, get the similar failure.
> prebuild-install WARN install No prebuilt binaries found (target=14.4.0 runtime=node arch=x64 libc= platform=darwin)
> No receipt for [...]

follow the following url to reinstall the xcode and it works.
https://medium.com/flawless-app-stories/gyp-no-xcode-or-clt-version-detected-macos-catalina-anansewaa-38b536389e8d

nickwild-999, 2022-02-15:
I have this issue too on RPI. I have Python 2.7 installed Any thoughts anyone?