Koenkk/zigbee2mqtt #3061: HUGE stability issue since a few days

way2many, 2020-03-03:
Hi all,

I'm running z2m latest docker dev version on raspberry pi 3 with a 1 meter extension cable and Zstack 3 on CC2531. Since a few days the network took a huge stability hit as devices even 2 meters from the coordinator have constant "No network route (205)".

I re-flashed the CC2531 to reset the routing table but still same result or even worse now. I'm on channel 11 and some wifi devices from the neighbors are on channel 6 so possible overlap but this was never an issue before. I checked with a wifi analyzer and nothing changed since the last few months.

The only thing that works FLAWLESSLY is anything in a group. Any command to a group and it works instantly. I'm even thinking of putting everything in its own group.

I can program a CC2531 to sniff traffic but then I don't know what to look for.

Any idea? Do I push the CC2531 too much? Should I go back to Z1.2 source firmware and place everything on channel 26?

derekoharrow, 2020-03-03:
Not sure if this helps, but I had similar issues and it was down to overloading the CC2531 with too many devices (about 25-30). I moved my light bulbs back to the Hue hub and everything is now working perfectly for the remaining switches, etc. I guess the CC2531 just wasn't up to the job.

way2many, 2020-03-04:
I have to be honest that I push the CC2531 to the limit. I'm about at 40 devices or even more.
Strange enough, end-devices are communicating fine but anything acting as a router has problems on the network. I already suspected some newly acquired Ikea GU10 bulbs (4 new) to destabilize the network.

Yesterday night I physically removed electricity to the 4 bulbs and this morning I have a stable network except for 1 Ikea GU10 and 1 Ikea E27 bulb.

It just occured to me that the bulbs having problems with communication and are possibly disrupting the network are bulbs I have updated with the new OTA feature.

sjorge, 2020-03-04:
I'm below 15 devices so I doubt I am pushing my CC2531 too hard. Both my ikea bulbs had no firmware updates available, the other one was is hue so also no updates.

Koenkk, 2020-03-04:
40+ devices with a non-source routing firmware (= zStack 3 or zStack12 default firmware) is definitely a no go. These filter do not store routes and therefore they query it again every time, this puts a lot of load on the network and can then cause `205` errors.  I would recommend to either:
- Switch to zStack 12 source routing firmware:  https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- Switch to a more powerful adapter (CC26X2R1 or CC1352P-2) https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1

way2many, 2020-03-04:
Oh I thought that even non-source routing firmware was storing routes but limited. I suppose the fact that I added 4 more bulbs put too much load on the network.

Am I correct to assume that commands to a group is send to the full network and the bulb reacts if it is in the correct group, hence why I have 0 errors with using groups? In that case, if I assign each bulb to a group, what could happen with the network?

mgrom, 2020-03-05:
@Koenkk is there a need for repairing after switching to source routing?

Koenkk, 2020-03-05:
@mgrom depends, https://www.zigbee2mqtt.io/information/FAQ.html#requires-repairing

mgrom, 2020-03-05:
BTW. I can confirm that update of: https://www.zigbee2mqtt.io/devices/LED1623G12.html has caused stability issue. Then I've tried to remove bulb from network without luck, I've tried to hard reset the bulb, but unfortunatelly it didn't reconnect to the network... 
One lesson: don't update when no need

way2many, 2020-03-27:
I'll close the issue as I migrated everything to source firmware Zstack 1.2 and it has been rock solid for more than a week. Seems that 40 devices is the upmost limit to not reach.

closed 2020-03-27 by way2many