Koenkk/zigbee2mqtt #3063: some bulbs constantly go offline<->online

sjorge, 2020-03-04:
# Bug Report
## What happened
Somewhere after 1.9 I have a few bulbs that keep flapping offline -> online with availability reporting enabled.

```
zigbee2mqtt:info  2020-03-04 07:40:28: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:error 2020-03-04 07:40:28: Failed to ping 'bedroom/night_light/bulb'
zigbee2mqtt:info  2020-03-04 07:41:10: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-04 07:43:15: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:error 2020-03-04 07:43:15: Failed to ping 'bedroom/night_light/bulb'
zigbee2mqtt:info  2020-03-04 07:43:23: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-04 07:45:28: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:error 2020-03-04 07:45:28: Failed to ping 'bedroom/night_light/bulb'
```

The bulbs responds fine via group or remote during the time it is offline.
If I send a manual mqtt message to the bulb it also responds.

It seems to be every other ping it is offline, then back online, then offline, ... I think this started happening after some of the changes to handle the no route, no mac, ... errors.

## What did you expect to happen
Not sure, as the ping itself does seem to fail at the time it was executed... I think the ping should already be tried a few times with the new no mac/no route handling so I am not sure if we need to do another retry.

Perhaps have a option to set the number of failed attempts? Like only mark it offline after X (1 by default, but could be set to 3 for example) failed pings?

Also odd that this bulb is currently a router for other devices (TRV + 2 remotes + 1 other bulb), and they work fine.

But sometiems the other bulb is the one that keeps flapping, and when I looked yesterday that one was the router and this one was conencted through that bulb. So perhaps there is still an edge case where multiple pings get confused if a router forwards it and gets one imediatly after for itself?

## How to reproduce it (minimal and precise)
Not sure, sometimes it's fine... other times all my bulbs have it until I restart the coordinator. Most of the time its the bulb acting a the router for my other bulbs that is the problem.

## Debug Info
zigbee2mqtt version: `1.11.0-dev (commit #2d10c15)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}`

way2many, 2020-03-04:
Same as me, I opened an issue here https://github.com/Koenkk/zigbee2mqtt/issues/3061#issue-574831405 about network instability. I suspect new Ikea bulbs/firmwares. Did you recently update some Ikea devices?

sjorge, 2020-03-04:
The bulbs are hue or tradfri’s that were already up to date

Koenkk, 2020-03-05:
I'm OK with adding some kind of minimum ping failure count, but it shouldn't be enabled by default as it will increase the time a bulb will be marked as offline.

sjorge, 2020-03-05:
I think you mean it will increase the time a bulb is marked online while it actually might be down...

I'm open for better suggetions though, the ping does go through the whole retrie on mac/route errors right?

sjorge, 2020-03-05:
It might also be the bulb, the most troublesome one always has this:
```
zigbee2mqtt:error 2020-03-05 08:25:35: Failed to read state of 'bedroom/night_light/bulb' after reconnect
zigbee2mqtt:info  2020-03-05 08:25:37: MQTT publish: topic 'zigbee2mqtt/bedroom/motion', payload '{"illuminance":18,"linkquality":55,"occupancy":true,"battery":100,"voltage":3015,"illuminance_lux":18}'
zigbee2mqtt:info  2020-03-05 08:25:59: Successfully setup reporting for '0x90fd9ffffee77fcf' - 1 - genOnOff
zigbee2mqtt:info  2020-03-05 08:25:59: Successfully setup reporting for '0x90fd9ffffee77fcf' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-03-05 08:26:00: Successfully setup reporting for '0x90fd9ffffee77fcf' - 1 - lightingColorCtrl
```

sjorge, 2020-03-05:
I the most troublesome one for a spare I had. It's also on an older firmware (zigbee 1.2 not the 3.0 firmware yet)

I will monitor it for a bit and if it stays stable for a until saturday, I will update. To see if the firmware might be an issue.

sjorge, 2020-03-05:
Scratch that, I was wrong... the bulb is already up to date according to z2m.
Oddly enough this one is also reportin color_temp in addition to color_xy, but I had that at some point in the past with my other ikea bulbs too... i think they support both color modes.

Koenkk, 2020-03-07:
TRADFRI bulbs have to be reconfigured every time they are powered off/on so this is expected.

sjorge, 2020-03-07:
Oh, that is interesting!

Also the new bulb also has the ping issue... I also moved it around and swapped it for a none CWS bulb... all trådfri bulbs seem to miss about 1/3 of the pings requests going to them

Koenkk, 2020-03-07:
When controlling it, do you see the same failure rate?

sjorge, 2020-03-07:
No, it always responds when controlling it (via node-red (homekit) -> mqtt -> z2m)

sjorge, 2020-03-07:
Wait, I am using that bulb in a group... so that is broadcast vs unicast right?
The trouble only started after the whole nomac/noroute fixes though, I wonder if something is still going wrong for these bulbs.

sjorge, 2020-03-07:
Interesting, it seems 100% of the pings to my ikea bulbs fail!

```
zigbee2mqtt:debug 2020-03-07 23:49:55: Received Zigbee message from 'bedroom/night_light/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":30015}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-03-07 23:49:55: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-07 23:49:56: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light', payload '{"color":{"x":0.458,"y":0.41},"brightness":254,"state":"ON","color_temp":120}'
zigbee2mqtt:info  2020-03-07 23:49:56: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb', payload '{"state":"ON","linkquality":34,"brightness":254,"color":{"x":0.458,"y":0.41},"update_available":false,"color_temp":120,"color_mode":1}'
zigbee2mqtt:debug 2020-03-07 23:50:01: Received Zigbee message from 'bedroom/desk_lamp/bulb', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt:error 2020-03-07 23:50:04: Failed to read state of 'bedroom/night_light/bulb' after reconnect
```

They only come back online because they send an attReport

sjorge, 2020-03-07:
Interesting, all my current joined ikea routers are failing LQI requests!

```
zigbee2mqtt:error 2020-03-07 23:57:27: Failed to execute LQI for 'masterbedroom/repeater'
zigbee2mqtt:error 2020-03-07 23:57:37: Failed to execute LQI for 'bedroom/night_light/bulb'
zigbee2mqtt:info  2020-03-07 23:57:37: Network scan finished
zigbee2mqtt:info  2020-03-07 23:57:37: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b001938a7e5" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001938a7e5 (0)|2020-03-07T23:57:37+00:00}"];
  "0x00124b001938a7e5" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="7"]
  "0x000d6ffffe8e8d4f" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{masterbedroom/repeater|0x000d6ffffe8e8d4f (47816)failed: lqi|IKEA TRADFRI signal repeater (E1746)|2020-03-07T23:56:49+00:00}"];
  "0x000d6ffffe8e8d4f" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="41"]
  "0x000d6ffffe8e8d4f" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="145"]
  "0x000d6ffffe197fe9" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/night_light/bulb|0x000d6ffffe197fe9 (36186)failed: lqi|IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)|2020-03-07T23:56:58+00:00}"];
  "0x000d6ffffe197fe9" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x000d6ffffe197fe9" -> "0x0017880104259333" [penwidth=0.5, weight=0, color="#994444", label="39"]
  "0x0017880104259333" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/desk_lamp/bulb|0x0017880104259333 (27321)|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|2020-03-07T23:57:02+00:00}"];
  "0x0017880104259333" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x14b457fffecc1315" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/night_light/remote|0x14b457fffecc1315 (44663)|IKEA TRADFRI ON/OFF switch (E1743)|2020-03-07T21:31:46+00:00}"];
  "0x14b457fffecc1315" -> "0x0017880104259333" [penwidth=1, weight=0, color="#994444", label="218"]
  "0x14b457fffeca351b" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/desk_lamp/remote|0x14b457fffeca351b (24931)|IKEA TRADFRI ON/OFF switch (E1743)|2020-03-07T19:12:59+00:00}"];
  "0x00158d0004148f89" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/motion|0x00158d0004148f89 (4263)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-03-07T23:29:42+00:00}"];
  "0x00158d0004148f89" -> "0x0017880104259333" [penwidth=1, weight=0, color="#994444", label="220"]
  "0x00158d00033ddfaa" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/sensor|0x00158d00033ddfaa (28407)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-03-07T23:50:04+00:00}"];
  "0x00158d0003f115c5" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{serverroom/sensor|0x00158d0003f115c5 (23324)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-03-07T23:47:32+00:00}"];
  "0x00158d0003f115c5" -> "0x00124b001938a7e5" [penwidth=1, weight=0, color="#994444", label="147"]
  "0x00158d0001ffaffc" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/radiator|0x00158d0001ffaffc (49490)|Eurotronic Spirit Zigbee wireless heater thermostat (SPZB0001)|2020-03-07T23:55:47+00:00}"];
  "0x04cf8cdf3c771820" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/light_sensor|0x04cf8cdf3c771820 (24216)|Xiaomi MiJia light intensity sensor (GZCGQ01LM)|2020-03-07T23:57:04+00:00}"];
}'
```

sjorge, 2020-03-09:
Oddly enough, it looks like it is the ikea repeater that drops from the netwerk and the ikea bulbs don't seem to pick a new route when this happens :/

Koenkk, 2020-03-09:
Can you provide the sniff log when this errors happens (`no network route`)? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

sjorge, 2020-03-09:
I can only sniff with an nRF52 for now (only have one TI usb)

I reset and rejoined both the repeater and the bulb again and it's behaving again... but it did it last time to after a rejoin.

sjorge, 2020-03-09:
Bah didn't mean to close it

sjorge, 2020-03-09:
Both my test trådfri and hue bulb dropped out again!

```
digraph G {
node[shape=record];
  "0x00124b001938a7e5" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b001938a7e5 (0)|2020-03-09T21:47:36+00:00}"];
  "0x00124b001938a7e5" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="114"]
  "0x000d6ffffe8e8d4f" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{masterbedroom/repeater|0x000d6ffffe8e8d4f (47816)|IKEA TRADFRI signal repeater (E1746)|2020-03-09T17:24:35+00:00}"];
  "0x000d6ffffe8e8d4f" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="62"]
  "0x000d6ffffe197fe9" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/night_light/bulb|0x000d6ffffe197fe9 (36186)failed: lqi|IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)|2020-03-09T17:24:35+00:00}"];
  "0x000d6ffffe197fe9" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="4"]
  "0x000d6ffffe197fe9" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="50"]
  "0x0017880104259333" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{bedroom/desk_lamp/bulb|0x0017880104259333 (27321)failed: lqi|Philips Hue white and color ambiance E26/E27/E14 (9290012573A)|2020-03-09T21:47:06+00:00}"];
  "0x0017880104259333" -> "0x00124b001938a7e5" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x0017880104259333" -> "0x000d6ffffe8e8d4f" [penwidth=0.5, weight=0, color="#994444", label="94"]
  "0x14b457fffecc1315" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/night_light/remote|0x14b457fffecc1315 (44663)|IKEA TRADFRI ON/OFF switch (E1743)|2020-03-09T04:50:06+00:00}"];
  "0x14b457fffeca351b" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/desk_lamp/remote|0x14b457fffeca351b (24931)|IKEA TRADFRI ON/OFF switch (E1743)|2020-03-09T21:47:02+00:00}"];
  "0x14b457fffeca351b" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="99"]
  "0x00158d0004148f89" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/motion|0x00158d0004148f89 (4263)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2020-03-09T21:47:00+00:00}"];
  "0x00158d00033ddfaa" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/sensor|0x00158d00033ddfaa (28407)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-03-09T06:23:41+00:00}"];
  "0x00158d0003f115c5" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{serverroom/sensor|0x00158d0003f115c5 (23324)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|2020-03-09T17:19:21+00:00}"];
  "0x00158d0003f115c5" -> "0x00124b001938a7e5" [penwidth=1, weight=0, color="#994444", label="147"]
  "0x00158d0001ffaffc" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/radiator|0x00158d0001ffaffc (49490)|Eurotronic Spirit Zigbee wireless heater thermostat (SPZB0001)|2020-03-09T21:46:08+00:00}"];
  "0x00158d0001ffaffc" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="105"]
  "0x04cf8cdf3c771820" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{bedroom/light_sensor|0x04cf8cdf3c771820 (24216)|Xiaomi MiJia light intensity sensor (GZCGQ01LM)|2020-03-09T21:47:12+00:00}"];
  "0x04cf8cdf3c771820" -> "0x000d6ffffe8e8d4f" [penwidth=1, weight=0, color="#994444", label="169"]
}
```

I suspect it might actually be the ikea repeater that is causing the issue. As it is late now, I will try to take a capture tomorowing using the nRF52, hopefully that will be usable for you.

sjorge, 2020-03-09:
```
zigbee2mqtt:info  2020-03-09 21:48:08: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-09 21:49:11: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-09 21:51:31: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-09 21:53:36: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb/availability', payload 'offline'
```

Yeah the hue bulb imediatly drops out again, I send a command to both... the hue responded and came online... then dropped again.

sjorge, 2020-03-10:
I made 2 caps, one to just the ikea bulb, one to the ikea bulb and the hue bulb... the later did respond. I also tried to read a value from my TRV... but it also failed.

I checked the netmap and all of them have shifted to being behind the ikea repeater. I think it might be to blame.

How do I get the caps + netkey to you? @Koenkk

sjorge, 2020-03-10:
pcaps are here: https://pkg.blackdot.be/extras/zigbee/

sjorge, 2020-03-10:
So to get the network stable again I had to do the following:
- power off/on repeater
- power off/on trådfri bulb
- remove/insert batteries for TRV

Now everything is working again, it's all going over the repeater to the coordinator.

sjorge, 2020-03-10:
And we're already back to one bulb not responding to pings :s

sjorge, 2020-03-10:
A watched kettle never boils I guess, @Koenkk was there anything useful in the other caps?

sjorge, 2020-03-10:
https://pkg.blackdot.be/extras/zigbee/ping_fail_attempt1.pcapng
https://pkg.blackdot.be/extras/zigbee/ping_fail_attempt2.pcapng
and
https://pkg.blackdot.be/extras/zigbee/sure_on_this_one.pcapng

Network key is in the same location but called `.zbk` (lets hope the scrapers don't grab it)

Koenkk, 2020-03-10:
I cannot open the files somehow, tried with wget and "Save link as", when opening in Wireshark I see: 

[image]

Also I don't see the key here? (https://pkg.blackdot.be/extras/zigbee/)

sjorge, 2020-03-10:
> Also I don't see the key here? (https://pkg.blackdot.be/extras/zigbee/)

Just ass `.zbl` at the end of the URL

Koenkk, 2020-03-10:
Ok you can remove the key now, saved it locally, still have the issue of https://github.com/Koenkk/zigbee2mqtt/issues/3063#issuecomment-597311635

sjorge, 2020-03-10:
See my screenshot, not sure what would be different aside from the key?

sjorge, 2020-03-10:
@Koenkk gah! It's because I use the nRF52 to capture (only one Ti stick)

```~/.config/wireshark/plugins/nrf802154_sniffer.lua``` You need the wireshark plugin

```
-- Copyright (c) 2019, Nordic Semiconductor ASA
-- All rights reserved.
--
-- Redistribution and use in source and binary forms, with or without modification,
-- are permitted provided that the following conditions are met:
--
-- 1. Redistributions of source code must retain the above copyright notice, this
--    list of conditions and the following disclaimer.
--
-- 2. Redistributions in binary form, except as embedded into a Nordic
--    Semiconductor ASA integrated circuit in a product or a software update for
--    such product, must reproduce the above copyright notice, this list of
--    conditions and the following disclaimer in the documentation and/or other
--    materials provided with the distribution.
--
-- 3. Neither the name of Nordic Semiconductor ASA nor the names of its
--    contributors may be used to endorse or promote products derived from this
--    software without specific prior written permission.
--
-- 4. This software, with or without modification, must only be used with a
--    Nordic Semiconductor ASA integrated circuit.
--
-- 5. Any software provided in binary form under this license must not be reverse
--    engineered, decompiled, modified and/or disassembled.
--
-- THIS SOFTWARE IS PROVIDED BY NORDIC SEMICONDUCTOR ASA "AS IS" AND ANY EXPRESS
-- OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES
-- OF MERCHANTABILITY, NONINFRINGEMENT, AND FITNESS FOR A PARTICULAR PURPOSE ARE
-- DISCLAIMED. IN NO EVENT SHALL NORDIC SEMICONDUCTOR ASA OR CONTRIBUTORS BE
-- LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
-- CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE
-- GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
-- HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
-- LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT
-- OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

nrf802154_sniffer_proto = Proto("nrf802154_sniffer","nRF Sniffer for 802.15.4 Data")
nrf802154_sniffer_proto.fields.channel = ProtoField.uint16("nrf802154_sniffer.channel", "Channel")
nrf802154_sniffer_proto.fields.rssi = ProtoField.int16("nrf802154_sniffer.rssi", "RSSI")
nrf802154_sniffer_proto.fields.lqi = ProtoField.uint16("nrf802154_sniffer.lqi", "LQI")

-- create a function to dissect it
function nrf802154_sniffer_proto.dissector(buffer, pinfo, tree)
    HEADER_LEN = 6
    if buffer:len() > HEADER_LEN then
        local t = tree:add(nrf802154_sniffer_proto, buffer(0,HEADER_LEN), "nRF Sniffer for 802.15.4 Data")
        t:add_le(nrf802154_sniffer_proto.fields.channel, buffer(0,2))
        t:add_le(nrf802154_sniffer_proto.fields.rssi, buffer(2,2))
        t:add_le(nrf802154_sniffer_proto.fields.lqi, buffer(4,2))
        Dissector.get("wpan_nofcs"):call(buffer(HEADER_LEN):tvb(), pinfo, tree)
    end
end

local wtap_encap_table = DissectorTable.get("wtap_encap")
wtap_encap_table:add(wtap.USER0, nrf802154_sniffer_proto)
```

My 2nd ti usb has been stuck in china since december so I doubt I'll get it anytime soon.

sjorge, 2020-03-10:
Nice! You can get the data without the lua script too by just setting the proper decoder and skipping the first 6 bytes (they hold nRF rssi info), sorry for the inconvenience 

[image]

sjorge, 2020-03-10:
@Koenkk better news, I am making a new cap without the nRF headers and I just cought one, give me a few minutes to upload... again sorry for the inconvenience

sjorge, 2020-03-10:
@Koenkk https://pkg.blackdot.be/extras/zigbee/3rd_no_nrf_just_tap.pcapng does this one open without the extra dissector? It does on my other laptop that doesn't have it.

sjorge, 2020-03-10:
Some extra info, the failing build db entry
```
{"id":3,"type":"Router","ieeeAddr":"0x000d6ffffe197fe9","nwkAddr":36186,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentX":30015,"currentY":26869,"colorMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1583859008338}
{"id":7,"type":"Group","groupID":20861,"members":[{"deviceIeeeAddr":"0x000d6ffffe197fe9","endpointID":1}],"meta":{}}
```

I noticed there are some packets marked malformed between coordinator and bulb around the time of the failure, followed by a rejoin from the bulb... that gets an invalid response.

sjorge, 2020-03-10:
Got another larger cap with multiple failures:
https://pkg.blackdot.be/extras/zigbee/larger_multi_ping_fail.pcapng, lots of odd stuff from the bulb too... unknown network updates, leave and join requests... very weird... I'm bad at reading those as the lower level zigbee knowledge is bad but I think:
- bulb does a network update
- coordinator doesn't respond / response bulb doesn't like
- bulb tries to leave the network (does it succeed?)
- bulb comes back into the network (odd, joining is disabled?) and it fails I think, but it sort of does now briefly respond again. :/

Hopefully uou can make more sense of it, I'm off to bed for now but if you need more captures... I can try again tomorrow. I think I got the hang of how to do it properly now. (assuming the last 2 ones are usable by you)

Koenkk, 2020-03-11:
I tried to open https://github.com/Koenkk/zigbee2mqtt/issues/3063#issuecomment-597325401 but still the same:

[image]

BTW regarding connectivity issues after rejoining, please check with latest dev (see second point of https://github.com/Koenkk/zigbee2mqtt/issues/2693#issuecomment-597271715)

sjorge, 2020-03-11:
I wonder if our wireshark versions are too different?
[image]

Koenkk, 2020-03-11:
I indeed had an old wireshark version, next problem is that I cannot decrypt the network traffic with the provided key, the one I used starts with 30: and ends with :f4

sjorge, 2020-03-11:
So you can open it (but not decrypt it) with a newer version, that's a good start :)

I've uploaded the entire `.config/wireshark/zigbee_pc_keys` to the same location as last time.

Koenkk, 2020-03-11:
I've added all 3 just to make sure but I still cannot decrypt it. Could you perhaps pair a device while sniffing so I can get the key from there?

Also I think it makes sense to first update to the latest dev branch and try to reproduce the issue from there again (to make sure we are not going to investigate an issue which has already may have been fixed, https://github.com/Koenkk/zigbee2mqtt/issues/3063#issuecomment-597769047)

sjorge, 2020-03-11:
Sure, I'm already up to date (did one this morning)... I'll try to do another fresh capture somewhere later this week.

sjorge, 2020-03-11:
I just unplugged the repeater and both bulbs and plugged them back in, the ikea bulb error after device announce is indeed gone.

Lets see if/when they start erroring again on ping.

```
zigbee2mqtt:info  2020-03-11 19:49:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"bedroom/night_light/bulb"}}'
zigbee2mqtt:info  2020-03-11 19:49:58: Successfully setup reporting for '0x000d6ffffe197fe9' - 1 - genOnOff
zigbee2mqtt:info  2020-03-11 19:49:58: Successfully setup reporting for '0x000d6ffffe197fe9' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-03-11 19:49:59: Successfully setup reporting for '0x000d6ffffe197fe9' - 1 - lightingColorCtrl
zigbee2mqtt:info  2020-03-11 19:49:59: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light', payload '{"color":{"x":0.458,"y":0.41},"brightness":254,"state":"ON","color_temp":120}'
zigbee2mqtt:info  2020-03-11 19:49:59: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb', payload '{"state":"ON","linkquality":34,"brightness":254,"color":{"x":0.458,"y":0.41},"update_available":false,"color_temp":120,"color_mode":1}'
```

PS: these are the capture settings I used for the last 2 caps:
[image]

sjorge, 2020-03-11:
I'm seeing a lot of unknown command in the capture... is that maybe due to a decrypt issue? If so I might have it locally too then?

sjorge, 2020-03-11:
Yeah I might have a wrong decrypt key ( I blanked mine locally and wireshark showed the result with blank.), I copied the section from `configuration.yaml` to the same location as before.

sjorge, 2020-03-12:
So far about a day without issues, after the last round of poweroff/on of the devices in question

Koenkk, 2020-03-12:
Good I can decrypt it now. I've checked https://github.com/Koenkk/zigbee2mqtt/issues/3063#issuecomment-597329882 and filtered on `zbee_nwk.src == 36186 || zbee_nwk.dst == 36186` (is `0x8d5a` the problematic device)?? I don't see the ping requests in there. 

- Could you indicate at which ID it fails and preferably also provide the zigbee2mqtt debug loggigng at that point
- Please also provide the networkAddress of the device (you can find it in `database.db`)

sjorge, 2020-03-13:
Sadly I don't have an exact message ID from when it failed. On the upside, with latest dev (with the route request change you mentioned), I have been stable for 2 days without issue!

If it returns I will switch to debug logging and do a new capture... as for the devices...
bulb: ```{"id":3,"type":"Router","ieeeAddr":"0x000d6ffffe197fe9","nwkAddr":36186,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":512,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 CWS opal 600lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009"}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":0}},"lightingColorCtrl":{"attributes":{"currentX":30015,"currentY":26869,"colorMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20180410","swBuildId":"1.3.009","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1583959502134}```
repeater that is used between bulb and coordinator: ```{"id":2,"type":"Router","ieeeAddr":"0x000d6ffffe8e8d4f","nwkAddr":47816,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI signal repeater","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3,9,2821,4096,64636],"outClusterList":[25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI signal repeater","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190318","swBuildId":"2.2.005"}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190318","swBuildId":"2.2.005","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1,"configured":2},"lastSeen":1583959431751}```

sjorge, 2020-03-13:
The errors are back, I will try to capture it tomorrow as it is getting late

sjorge, 2020-03-14:
It's still happening but way less often, it's hard to try and capture it in a dump :(

```
zigbee2mqtt:info  2020-03-14 10:12:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 10:14:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 12:42:15: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 12:44:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 13:02:15: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 13:04:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 14:02:15: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 14:04:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 15:04:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 15:04:17: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 15:07:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 15:09:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 15:12:16: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 15:14:17: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
zigbee2mqtt:info  2020-03-14 15:14:20: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'offline'
zigbee2mqtt:info  2020-03-14 15:14:27: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb/availability', payload 'online'
```

sjorge, 2020-03-14:
I was unsuccesful in capturing one after 2h of trying :(
Lets close this for now as it is no longer happening every other ping, the pings followed the backoff and request route flow too right?

closed 2020-03-14 by sjorge