Koenkk/zigbee2mqtt #3064: linkquality for router keeps changing

mr2c12, 2020-03-04:
In my network, I have a CC2531 with an external antenna: "0x00124b001943099c"
The device is downstair with a lot of other devices (router or end devices).

Coordinator is CC2530+CC2591 1.2 without source routing.

Since my network has some stability problem, especially the "downstair part", I'm investigating.
Looking at z2m log, I found something unexpected: the linkquality for this devices is usually zero, but like twice an hour I got value like ~30 or ~60.
I pasted the log here: https://pastebin.com/NiUkVTFp

Why is this changing? Devices are not moving from their place. Does it mean that the network is changing over time but cannot find a fixed point?
Could it be the culprit for my troubles (sometimes I can get to a device, or it responds with a late of ~10 seconds)?

Addictional info:
z2m version: 1.9
network graph:
[image]

pixeldoc2000, 2020-04-04:
Hard to guess. Maybe this is due to interferences with other devices using the 2.4 GHz Band like WLAN, Bluetooth etc.

If the transmission is on the edge, even the present of a human body can be enough to disrupt the network untill it eventually recovers.

mr2c12, 2020-05-11:
Probably related to my wifi AP using a wireless channel that used to interfere with the zigbee network.

closed 2020-05-11 by mr2c12