Koenkk/zigbee2mqtt #3065: Need help with building own sensor

nielsfaber, 2020-03-04:
So Im starting a project for making an Air Quality sensor (how original), which will basically be an ESP32 with different sensors (CO2, PM2.5, O2, etc) in a box.
I would like to find out if it would be possible to use Zigbee (in favour of WiFI), for sending the readings periodically to the coordinator (raspberry pi).

My first thoughts:
- Buy an CC2530 (it seems [recommended)](https://www.zigbee2mqtt.io/devices/CC2530.ROUTER.html) for DIY
- Use CC programmer to flash the Z-stack FW into the CC2530
- Connect CC2530 to ESP32 through UART
- Define a format for the sensor readings to be send
- Follow the guide on [how to support new devices](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) to be able to process the data on the Coordinator side

Biggest unknowns for me are in the communication between ESP and CC2530. I don't know if I can simply send some string over UART and it would simply be sent to the coordinator, where it can be parsed into topic+payload structure, or there is much more to it.
I would like to know if someone else here has tried (and succeeded) a similar project. I understand this is not directly related to zigbee2mqtt, but I hope someone could point me in the right direction still.

ryanbeaton, 2020-03-04:
It’s not exactly what you’re doing but have you checked out:
https://github.com/formtapez/ZigUP

caveymon, 2020-03-04:
You could also look into the nRF52840 dongles/chipset. Pretty DIY, has zigbee stack all over it (and more), mbed support (makes it a bit easier to develop), and you can hook up all the sensors you could think of until you run out of gpio's.

closed 2020-05-10 by stale[bot]