Koenkk/zigbee2mqtt #3066: Availability for battery powered devices

twsI, 2020-03-05:
https://www.zigbee2mqtt.io/information/availability.html states, that this only includes non-battery powered devices. I know, this puts a lot of traffic on the network and might drain the battery, but I'd love to enable this for battery powered devices as well, just with different and slightly higher intervalls. What do you guys think of two separated values?

sjorge, 2020-03-05:
It actually sort of works with battery devices for a few months now.

It works under the assumtion that these devices will check in ~1x every 24h to send a battery state update. https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/deviceAvailability.js#L83

Koenkk, 2020-03-07:
This is already implemented, I've updated to docs a bit to be more clear about this.

closed 2020-03-07 by Koenkk