Koenkk/zigbee2mqtt #3067: zigbee2mqtt cannot be started after upgrade from 1.7.1 to any newer version

UlrichFrank, 2020-03-05:
# Bug Report
## What happened
Zigbee2mqtt cannot be started with any version newer than 1.7.1
I tried to setup from scratch (removed old database, device config) without success.
The zigbee device is a CC2530 connected via USB
Zigbee2mqtt docker image is used.
After downgrade to 1.7.1 it works like a charm again. (totally stable)
Do you have any idea?

log: https://pastebin.com/nafpMD8Q
config: https://pastebin.com/56cf5kNG

## Debug Info
zigbee2mqtt version: 1.11.0 (or any other version newer than 1.7.1) 
CC253X firmware version: CC2530_DEFAULT_20190608.zip, I also tried 3.0.x version a few weeks ago, without success

filoor, 2020-03-05:
Hi,

It looks it can connect to your mqtt broker.

Koenkk, 2020-03-05:
Can you try after removing the disable_led from your configuration?

UlrichFrank, 2020-03-05:
I‘ve seen that. Questions I have:
- difference between 1.7.1 and newer versions? Has the mqtt config changed?
(Broker is running and working)
- Only first Connection to Mqtt Broker fails, each time the container is restarted, MQTT connection can be established (no error message in log visible). Any thoughts about that?

UlrichFrank, 2020-03-05:
It works. 👍

closed 2020-03-05 by UlrichFrank