Koenkk/zigbee2mqtt #3069: IKEA LED1545G12 Remembering Power State on 2.3.007 firmware

kiall, 2020-03-05:
# Bug Report
## What happened

After an OTA update of an IKEA bulb to 2.3.007 (I'm unsure what the previous version was - likely very old), the bulb is now remembering it's power state - physically removing power, and turning it back on again, results in the bulb staying off.

Some people would call this a feature, but I'd consider it a bug :D Thus, I guess this is a feature request to support a configuration for this.

## What did you expect to happen

As far as I understand it, we should be able to configure this setting of the bulb. Z2M should be extended to support this configuration.

## How to reproduce it (minimal and precise)

OTA update a bulb to 2.3.077, cut power, restore power, verify light is still off.

## Debug Info
zigbee2mqtt version: 1.11.0-dev
CC253X firmware version: 20191106

sjorge, 2020-03-05:
If you check the r/trådfri subreddit, a lot of people complained and alot of people rejoiced about the new functionality.

Sadly it looks like this is a bug and was not the intended new behavior (I think the tradfri team said they will fix it in the next firmware). But not all update bulbs have this behavior. Mine don't (sadly as I don't personally would prefer this bahvior)

Ther is nothing z2m can do to configure this, as that needs support in the firmware on the bulb to work. On r/trådfri people had success with removing and joining the bulb again to get back the old behavior.

kiall, 2020-03-05:
Oh, that sucks. I figured they had added this as an option. Oh well, closing as there's nothing Z2M can do!

closed 2020-03-05 by kiall