Koenkk/zigbee2mqtt #3070: TRADFRI bulb E27 CWS opal 600lm might support color_temp

sjorge, 2020-03-05:
```
{
  "id": 3,
  "type": "Router",
  "ieeeAddr": "0x000d6ffffe197fe9",
  "nwkAddr": 36186,
  "manufId": 4476,
  "manufName": "IKEA of Sweden",
  "powerSource": "Mains (single phase)",
  "modelId": "TRADFRI bulb E27 CWS opal 600lm",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 49246,
      "epId": 1,
      "devId": 512,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        768,
        2821,
        4096
      ],
      "outClusterList": [
        5,
        25,
        32,
        4096
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "TRADFRI bulb E27 CWS opal 600lm",
            "manufacturerName": "IKEA of Sweden",
            "powerSource": 1,
            "zclVersion": 1,
            "appVersion": 17,
            "stackVersion": 87,
            "hwVersion": 1,
            "dateCode": "20180410",
            "swBuildId": "1.3.009"
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        },
        {
          "cluster": 8,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        },
        {
          "cluster": 768,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        }
      ]
    }
  },
  "appVersion": 17,
  "stackVersion": 87,
  "hwVersion": 1,
  "dateCode": "20180410",
  "swBuildId": "1.3.009",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": {
    "reporting": 1
  },
  "lastSeen": 1583411694494
}
```

I paired a spare Ikea E27 CWS bulb today to replace one that was badly behavion wrt ping commands.

Unfortunately I did not have debug on but it reported a color_temp value, I remember seeing that in the past too

 ```
zigbee2mqtt:info  2020-03-05 12:44:55: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light/bulb', payload '{"state":"OFF","linkquality":55,"brightness":254,"color":{"x":0.4597,"y":0.41},"update_available":false,"color_temp":370}'
```

I also unfortunately already change the color via color_xy, so now it looks like it stopped reporting color_temp :(

Looks like we are unable to set it though:
```
zigbee2mqtt:error 2020-03-05 12:45:32: No converter available for 'color_temp' (366)
```

It would be nice, if we could set it using color_temp too like we can with Hue bulbs.

Koenkk, 2020-03-06:
Can you update the toZigbee converter and check if it works? With mine it doesnt work (checked long time ago) but maybe it has been added in a recent OTA.

sjorge, 2020-03-06:
Sure, I can take a jab at it this weekend probably, still have some uncommit work I want to push first for Innr (moveToHue + moveToHueAndSaturation) though, been in testing for a few days now. (thanks to your options.applyRedFix trick that give me the implementation idea)

sjorge, 2020-03-06:
Odd that it publishes one color_mode and color_temp after joining, but afterwards I can only read colorMode... but not colorTemperature. The former being 1 so xy mode.
It does indeed appear impossinle to set a color_temp.

closed 2020-03-06 by sjorge