Koenkk/zigbee2mqtt #3073: Support for Aurora AOne Dual socket

jlpedrosa, 2020-03-06:
# Bug Report
Support for [AONE™ 13A SMART DOUBLE SOCKET](https://auroralighting.com/gb/ProductDetail/AU-A1ZBDSS) (AU-A1ZBDSS) ZigbeeId: 'DoubleSocket50AU' 

## What happened
The socket was paired correctly.
I followed the guide on how to add unsupported devices, but I was getting the same errors as: https://github.com/Koenkk/zigbee2mqtt/issues/3019

So I repeated the test using latest-dev docker image sha256:7b628d14e07581777883e38dbde2371935278a093ae56f4fb174b8e829776505

```
zigbee2mqtt:debug 2020-03-06 00:16:40: Received Zigbee message from 'double_socket', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4636,"imageType":5129,"fileVersion":20}
' from endpoint 1 with groupID 0
```
 No matter how many clicks I do on the socket or if I plug something into it, it won't apear new messages in the log. Just the ota ones.

Not sure if it cames with a dummy firmware that just allows update, if it won't emit messages on/off or if it does not provide power consumption.

 this are supposed to be the features:
- Red and Blue LED indicators that can be turned on/off to suit the end users needs
- The left and right side of the socket can be individually controlled and locked remotely using your smart home system

So at least we should be able to do each of those things for both sockets?

I tried adding to devices.js
```
 // Aurora
 {
    zigbeeModel: ['DoubleSocket50AU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'AU-A1ZBDSS', // Vendor model number, look on the device for a model number
    vendor: 'Aurora', // Vendor of the device (only used for documentation and startup logging)
    description: 'AONE 13A SMART DOUBLE SOCKET', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Who knows', // Actions this device supports (only used for documentation)
},
```
but the result after the message is: 
```
zigbee2mqtt:error 2020-03-06 00:31:07: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'filter' of undefined                                                                                  
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:105:52)                                                                                                                                               
    at Controller.callExtensionMethod (/app/lib/controller.js:340:44)                                                                                                                                                         
    at Controller.onZigbeeEvent (/app/lib/controller.js:230:14)                                                                                                                                                               
    at Zigbee.emit (events.js:311:20)                                                                                                                                                                                         
    at Controller.<anonymous> (/app/lib/zigbee.js:57:52)                                                                                                                                                                      
    at Controller.emit (events.js:311:20)                                                                                                                                                                                     
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:466:22)                                                                                                                        
    at Generator.next (<anonymous>)                                                                                                                                                                                           
    at /app/node_modules/zigbee-herdsman/dist/controller/controller.js:8:71                                                                                                                                                   
    at new Promise (<anonymous>))               
```
## What did you expect to happen
Messages in the log to trouble shoot and do a PR. 

## How to reproduce it (minimal and precise)
You'd need a device....

## Debug Info
zigbee2mqtt version: 7b628d14e07581777883e38dbde2371935278a093ae56f4fb174b8e829776505
CC253X firmware version: using board CC1352P-2

Koenkk, 2020-03-06:
First make sure you upgrade to the latest zigbee2mqtt dev branch. Now try with
```js
// Aurora
{
    zigbeeModel: ['DoubleSocket50AU'],
    model: 'AU-A1ZBDSS',
    vendor: 'Aurora',
    description: 'AONE 13A SMART DOUBLE SOCKET',
    supports: 'on/off',
    fromZigbee: [fz.on_off_multi_endpoint],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(1);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint2);
    },
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
},
```

After this:
- You should be able to control it by sending to `zigbee2mqtt/double_socket/left/set` (or `right`) payload `{"state": "ON"}`
- Get state changes when manual controlling manually.

jlpedrosa, 2020-03-07:
Hi !

I think there´s was a typo in the second enpoint. 
```
const endpoint2 = device.getEndpoint(1);
```
to
```
const endpoint2 = device.getEndpoint(2);
```

I added it and it is working! I was not able to find out how to configure reporting for power metering and the messages to control the socket light, it's dimmable.

this is the DB entry for that device I see it has two endpoints, but no clue how to generate the messages
```
{
    "id": 4,
    "type": "Router",
    "ieeeAddr": "0x00158d000256f890",
    "nwkAddr": 33106,
    "manufId": 4636,
    "manufName": "Aurora",
    "powerSource": "Mains (single phase)",
    "modelId": "DoubleSocket50AU",
    "epList": [
        1,
        2
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 9,
            "inClusterList": [
                6,
                5,
                4,
                0,
                3,
                2820,
                8
            ],
            "outClusterList": [
                25
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "DoubleSocket50AU",
                        "manufacturerName": "Aurora",
                        "powerSource": 1,
                        "zclVersion": 1,
                        "appVersion": 20,
                        "stackVersion": 2,
                        "hwVersion": 1
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "onOff": 1
                    }
                }
            },
            "binds": [
                {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b001ca1b6fd",
                    "endpointID": 1
                }
            ]
        },
        "2": {
            "profId": 260,
            "epId": 2,
            "devId": 9,
            "inClusterList": [
                6,
                5,
                4,
                0,
                3,
                2820
            ],
            "outClusterList": [],
            "clusters": {},
            "binds": [
                {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b001ca1b6fd",
                    "endpointID": 1
                }
            ]
        }
    },
    "appVersion": 20,
    "stackVersion": 2,
    "hwVersion": 1,
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": {
        "configured": 3
    },
    "lastSeen": 1583610966366
}
```

Any guidance how to configure the nexts steps? do you want me to try something else?

thanks!

Koenkk, 2020-03-09:
Please update to the latest zigbee2mqtt **dev branch** first, now try with the changes from: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1049/files

jlpedrosa, 2020-03-09:
Hi @Koenkk 

I´ve checked and it's not in latest-dev docker image, I got errors of undefined variables.
```
docker rm koenkk/zigbee2mqtt:latest-dev
docker pull koenkk/zigbee2mqtt:latest-dev
docker run -i -t koenkk/zigbee2mqtt:latest-dev --entrypoint /bin/sh
/app # grep -ri readEletricalMeasurementPowerConverterAttributes /app
/app # 
```
So I checked dev branch and rebuild the docker image to jlpedrosa/zigbee2mqtt:latest-dev.
and tried again, results:

1. "Malformed" JSON
fields names have curly braces inside
```
zigbee2mqtt:info  2020-03-09 22:31:54: MQTT publish: topic 'zigbee2mqtt/double_socket', payload '{"state":"ON","linkquality":0,"state_right":"OFF","state_left":"ON","power_left}":0,"state_left}":"OFF","state_right}":"OFF"}' 
```
2. Incorrect content:
a) Two states right and left.
b) Both sockets are OFF, and both socket leds (the ones that are dimmable) are on.
c) Only one power source (left), it seems to be left and the measurement seems a reasonable watts.
```
{
    "state": "ON",
    "linkquality": 0,
    "state_right": "OFF",
    "state_left": "ON",
    "power_left}": 0,
    "state_left}": "OFF",
    "state_right}": "OFF"
}
```
3. OTA issue 1
```
zigbee2mqtt:error 2020-03-09 22:24:56: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: CommandResponse 0x286d97000108897c/1 genOta.queryNextImageResponse({"status":152}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse
":true}) failed (Error: AREQ - AF - dataConfirm after 5000ms)                                                                                                                                                                               
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:359:23)                                                                                                                                    
    at Generator.throw (<anonymous>)                                                                                                                                                                                                        
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)) 
```
4. OTA Issue 2
```
zigbee2mqtt:debug 2020-03-09 22:27:25: Received Zigbee message from 'double_socket', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4636,"imageType":5129,"fileVersion":20}' from endpoin
t 1 with groupID 0                                                                                                                                                                                                                          
zigbee2mqtt:debug 2020-03-09 22:27:25: No converter available for 'AU-A1ZBDSS' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4636,"imageType":5129,"fileVersion":20}' 
```
Thanks! Let me know if you would like to do a shared coding session with access to my zigbee network.

Koenkk, 2020-03-10:
@jlpedrosa sorry build failed and therefore it was not published, this is fixed now.

First remove your `data/states.json`, and now update to the latest-dev and apply the change from https://github.com/Koenkk/zigbee-herdsman-converters/pull/1049/files

jlpedrosa, 2020-03-10:
Hi @Koenkk 

No problem! I ran it and I got this message:
```
zigbee2mqtt:error 2020-03-10 19:43:24: Failed to configure 'double_socket', attempt 1 (Error: Bind 0x00158d000256f890/1 genOnOff from '0x00124b001ca1b6fd/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)                                                                                  
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)                                                                                                                                                                                         
    at Generator.throw (<anonymous>)                                                                                                                                                                                                                                                             
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

So I tried to repair it and this is the log:
```                                                                                                                                                                                               
zigbee2mqtt:debug 2020-03-10 19:46:00: Device 'double_socket' announced itself                                                                                                                                                                                                                   
zigbee2mqtt:info  2020-03-10 19:46:00: Configuring 'double_socket'                                                                                                                                                                                                                               
zigbee2mqtt:debug 2020-03-10 19:46:04: Device 'door_sensor' announced itself                                                                                                                                                                                                                     
zigbee2mqtt:error 2020-03-10 19:46:10: Failed to configure 'double_socket', attempt 2 (Error: Bind 0x00158d000256f890/1 genOnOff from '0x00124b001ca1b6fd/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)                                                                                  
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)                                                                                                                                                                                         
    at Generator.throw (<anonymous>)                                                                                                                                                                                                                                                             
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)) 
```

And then also the usual OTA messages:
```
zigbee2mqtt:debug 2020-03-10 19:48:08: Received Zigbee message from 'double_socket', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4636,"imageType":5129,"fileVersion":20}' from endpoint 1 with groupID 0                                   
zigbee2mqtt:debug 2020-03-10 19:48:08: No converter available for 'AU-A1ZBDSS' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4636,"imageType":5129,"fileVersion":20}'                                                             
zigbee2mqtt:debug 2020-03-10 19:48:12: Saving state to file /app/data/state.json                                                                                                                                                                                                                 
zigbee2mqtt:error 2020-03-10 19:48:12: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: CommandResponse 0x00158d000256f890/1 genOta.queryNextImageResponse({"status":152}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Data request failed with erro
r: 'MAC no ack' (233))                                                                                                                                                                                                                                                                           
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:359:23)                                                                                                                                                                                         
    at Generator.throw (<anonymous>)                                                                                                                                                                                                                                                             
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
````

Koenkk, 2020-03-10:
And what happens when controlling now?

jlpedrosa, 2020-03-10:
Nothing, no input messages received at all about on/off or power consumption.

Koenkk, 2020-03-10:
When repairing, did you remove it first through: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

jlpedrosa, 2020-03-11:
Hi @Koenkk 
No I did not know I had to, so I end up sending that message and also deleting all files from the container that were not `configuration.yaml`, there I removed all the `devices:` and started the container again and re-pair the socket.

Results, a lot of improvements, with some bugs still:

0) Buttons are definitely correct.
1) (Minor) Until you push the buttons, the state published misses that state: ie:
```json
{"power_left":82,"linkquality":69,"power_right":82,"state_left":"OFF"}
```
And after pushing the right button:
```json
{"power_left":82,"linkquality":87,"power_right":82,"state_left":"ON","state_right":"ON"}'
```

2) Power measurement value sounds incorrect and linked to the left side only
 On the other attempt it seemed pretty accurate, 0 was 0, plugging a mac charger was going up and down between 40 and 60.

- Nothing plugged reports 82 ON BOTH
- Small bulb (labeled as 5.7w) on the left  reports 612-620 ON BOTH
- If I plug something on the left socket, I get the report on BOTH (from zigbee side, see log)
```
info  2020-03-11 20:38:20: MQTT publish: topic 'zigbee2mqtt/0x00158d000256f890', payload '{"power_left":82,"linkquality":93,"power_right":82,"state_left":"ON","state_right":"ON"}'
debug 2020-03-11 20:38:40: Received Zigbee message from '0x00158d000256f890', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":554}' from endpoint 2 with groupID 0
info  2020-03-11 20:38:40: MQTT publish: topic 'zigbee2mqtt/0x00158d000256f890', payload '{"power_left":82,"linkquality":93,"power_right":554,"state_left":"ON","state_right":"ON"}'
debug 2020-03-11 20:38:41: Received Zigbee message from '0x00158d000256f890', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":623}' from endpoint 1 with groupID 0
info  2020-03-11 20:38:41: MQTT publish: topic 'zigbee2mqtt/0x00158d000256f890', payload '{"power_left":623,"linkquality":96,"power_right":554,"state_left":"ON","state_right":"ON"}'
```
 - Pluging things on the right side does report any change.

3) We are still missing the dim of the lights of the socket, it may seem sily, but these guys are bright, and turning them off in a bedroom is really good idea. Or at max in the corridors for night minimal light (furniture collision avoidance)

Koenkk, 2020-03-12:
1. This is expected (this is how it works for all devices)
2. Could you:
    - Could you share your `database.db`?
    - Provide the logging when plugging a load on the right side (without something on the left)
    - Provide the logging when plugging a load on the left side (without something on the right)
3. Is this possible at all? Do you know if this example can be done via the original gateway?

jlpedrosa, 2020-03-12:
Hi! @Koenkk 
Thanks, Sure, let me go point by point.

1. Perfect.
2.1: Here you have
```
{
    "id": 2,
    "type": "Router",
    "ieeeAddr": "0x00158d000256f890",
    "nwkAddr": 14677,
    "manufId": 4636,
    "manufName": "Aurora",
    "powerSource": "Mains (single phase)",
    "modelId": "DoubleSocket50AU",
    "epList": [
        1,
        2
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 9,
            "inClusterList": [
                6,
                5,
                4,
                0,
                3,
                2820,
                8
            ],
            "outClusterList": [
                25
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "DoubleSocket50AU",
                        "manufacturerName": "Aurora",
                        "powerSource": 1,
                        "zclVersion": 1,
                        "appVersion": 20,
                        "stackVersion": 2,
                        "hwVersion": 1
                    }
                },
                "haElectricalMeasurement": {
                    "attributes": {
                        "activePower": 82
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "onOff": 0
                    }
                }
            },
            "binds": [
                {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b001ca1b6fd",
                    "endpointID": 1
                },
                {
                    "cluster": 2820,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b001ca1b6fd",
[... 78 lines cut ...]
```
NOTHING PLUGGED
22:12:00: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":71}' from endpoint 2 with groupID 0
22:12:00: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":82,"linkquality":114,"power_right":71,"state_left":"ON","state_right":"ON"}'
22:12:04: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":82}' from endpoint 2 with groupID 0
22:12:04: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":82,"linkquality":114,"power_right":82,"state_left":"ON","state_right":"ON"}'
PLUG SOMETHING ON THE LEFT
22:12:24: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":623}' from endpoint 2 with groupID 0
22:12:24: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":82,"linkquality":114,"power_right":623,"state_left":"ON","state_right":"ON"}'
22:12:25: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":612}' from endpoint 1 with groupID 0
22:12:25: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":612,"linkquality":114,"power_right":623,"state_left":"ON","state_right":"ON"}'
22:12:28: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":612}' from endpoint 2 with groupID 0
22:12:28: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":612,"linkquality":114,"power_right":612,"state_left":"ON","state_right":"ON"}'
22:12:32: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":623}' from endpoint 1 with groupID 0
22:12:32: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":623,"linkquality":117,"power_right":612,"state_left":"ON","state_right":"ON"}'
22:12:36: Received Zigbee message from 'double_socket_aurora', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":612}' from endpoint 1 with groupID 0
22:12:36: MQTT publish: topic 'zigbee2mqtt/double_socket_aurora', payload '{"power_left":612,"linkquality":117,"power_right":612,"state_left":"ON","state_right":"ON"}'
```
3. I never had the official hub or app, I bought this as a test, you have been availed by @kiall. Specs and brochure of the device claims so. I assume it is true.

4. Measured Values:
Values seems 100x.
Idle: reports 70 (0.7 seems correct to me)
Bulb labeled as 5.7w: reports 625 with the bulb. (6.2w seems correct for me as idle + bulb)
Macbook pro: reports between 2800 and 4200. (28-40w seems correct for me)

kiall, 2020-03-14:
> 1. I never had the official hub or app, I bought this as a test, you have been availed by @kiall.

Hey! Don't drag me into anything :D I just recommended the project to you :)

Koenkk, 2020-03-15:
I've updated the PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1049/files , please try with these changes.

2. It looks that this device can only report the combined power consumption? To confirm this please try to connect 2 known loads to both switches, if you would now plug 5W + 5W and `power` now shows `10`, this is true.
3. I don't know what to send in order to do this (this command is not standardardized within Zigbee), you will either need techincal documentation or sniff the traffic with the original gateway in order to figure this out.
4. Fixed

jlpedrosa, 2020-03-22:
Hi @Koenkk 

Sorry for the delay. I've seen the patch, that should fixed the 100x.
2 & 3: I'm trying to get their hub to be able to debug and sniff the traffic so we can set the right commands. I should be able to put the keys for the captured traffic that this hub/coordinator creates?

Thanks once again!

Koenkk, 2020-03-22:
@jlpedrosa if during that sniff you pair the device the network key is included in it. (transport key message) https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html#3-sniffing-traffic

jlpedrosa, 2020-04-30:
Hi, don't close this one, I'm still trying to get their hub to be able to sniff the traffic.

closed 2020-07-06 by stale[bot]

Sk3wby, 2021-02-24:
Hi, can anyone help me get this working on my HassOS VM install? Looks like it was almost there, i don't need to control the LEDs.

I currently have zigbee2mqtt add-on from https://github.com/zigbee2mqtt/hassio-zigbee2mqtt installed but can't get external_converters to work.  I have tried adding the fromZigbee.js and devices.js entries (both on their own and edited into the full files) but zigbee2mqtt fails to start.

Have also tried with the edge version but still the same, loads files but errors about not finding /utils/ or unexpected tokens.

Would realy like to get this socket working, any help would be appreciated.

thanks,