Koenkk/zigbee2mqtt #3075: WXCJKG13LM binding

Zolli, 2020-03-06:
# Bug Report
## What happened
I have a 6 gang aqara opple wall switch, i try to bind this to a group (group of lights), binding is successfull if i wakre up the device during the binding process and all three endpoints bindes sucessfully, but the switch does not control the group, not on-off or brightness. I try all combinations of keypresses (single/double/triple/hold)

The group has 3 lights
Lights: Ikea tradfri LED1732G11

## What did you expect to happen
After binding at leas on/off/brightness works

## How to reproduce it (minimal and precise)
Bind the switch to a group of lights

## Debug Info
zigbee2mqtt version: latest-dev (2020-03-03)
CC253X firmware version: -
CC1352P-2 firmware version: Z-Stack_3.x.0 (not sure, need to check)

lolorc, 2020-03-06:
Same here, interview successful but the switch doesn't control anything, z2m doesn't receive anything, unable to bind/unbind to/from Coordinator.

Zolli, 2020-03-06:
@lolorc Thats more strange, i see all events in logs just fine, i see these events from the 6 gang switch

button_X_singe
button_X_double
button_X_triple
button_X_holde
button_X_release
(X is 1-6)

I not try to unbind it from the coordinator, because i doesn't have the strange problem that the first and second button turn on/off all my light/switches. But i try to force unbind from the coordinator when i got home.

Zolli, 2020-03-06:
I tried all the binding combination i can think of, 

Removed from Coordinator
Bind it to a group - Not works
Unbind from the above mentioned group
Bind to a single bulb - Not works
Unbind from the bulb

lolorc, 2020-03-08:
ok, looks like we don't have the same issue, I get a interview_successful with my 2 WXCJKG13LM switches, but the switch doesn't send any data when pressing the buttons.
here is one database entry:
```
{"id":60,"type":"EndDevice","ieeeAddr":"0x04cf8cdf02134657","nwkAddr":52730,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b686opcn01","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,3,1],"outClusterList":[3,6,8,768],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.remote.b686opcn01","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":17,"stackVersion":2,"hwVersion":1,"dateCode":"20190730","swBuildId":"2019\u0000www."}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[3],"outClusterList":[6,3],"clusters":{},"binds":[]},"3":{"profId":260,"epId":3,"devId":259,"inClusterList":[18,3],"outClusterList":[6],"clusters":{},"binds":[]},"4":{"profId":260,"epId":4,"devId":259,"inClusterList":[18,3],"outClusterList":[6],"clusters":{},"binds":[]},"5":{"profId":260,"epId":5,"devId":259,"inClusterList":[18,3],"outClusterList":[6],"clusters":{},"binds":[]},"6":{"profId":260,"epId":6,"devId":259,"inClusterList":[18,3],"outClusterList":[6],"clusters":{},"binds":[]}},"appVersion":17,"stackVersion":2,"hwVersion":1,"dateCode":"20190730","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1583667808982}
```

I'll try to bind them to my xiaomi bridge later on the afternoon to see how they work and if there's some kind of firmware update.

lolorc, 2020-03-08:
I haven't been able to pair to my old aqara hub v3.
So for now, my opple switches useless, could any body provide instructions to use those devices with z2m ?

Zolli, 2020-03-08:
What version are you running? Could you try the latest Dev branch? Or use the latest-dev tag if you use docker.

lolorc, 2020-03-08:
i'm running lastest dev from march 6, I'm running it with a cc2652.
I've been curious to try with a cc2531 (zStack30x/20190425) and dev commit 2e958f8 from January 18 and it works as expected.
Will do more tests,

lolorc, 2020-03-09:
doesn't work with all routers off. will have to compare traces between cc2531 && cc2652.
any of you using cc2652 with those opple devices ?

Zolli, 2020-03-09:
I use the CC1352P-2, its basically tha same chip, as i mentioned it works fine, expect the binding.

Koenkk, 2020-03-09:
I've sniffed the traffic and saw it doesn't send any command used to control other devices, only attribute reports (so not strange this doesn't work)

[image]

@kirovilya has this something to do with: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L954 ?

lolorc, 2020-03-09:
I will compare dumps when pairing to cc2531 vs cc2652 tomorrow.
Just going to do one more try with cc2531 this evening, I'll configure the cc2531 with report: true.
since it's working with cc1352, I don't think it's a firmware issue, but I've seen wireshark was displaying a zigbee device profile, device announcement :
```
    Capability Information: 0x84
        .... ...0 = Alternate Coordinator: False
        .... ..0. = Full-Function Device: False
        .... .1.. = AC Power: True
        .... 0... = Rx On When Idle: False
        .0.. .... = Security Capability: False
        1... .... = Allocate Short Address: True
```

"AC Power: True" looks false since it's not a AC powered device.

lolorc, 2020-03-10:
hmm, correction,  "ac power: True" is send by the device, but powerSource is set to 3/battery in the database.db.
looking at the database.db I can see the entry for the same opple sensor is not the same:
cc2652:
```
{
   "zclVersion" : 3,
   "swBuildId" : "2019\u0000www.",
   "interviewCompleted" : true,
   "epList" : [
      1,
      2,
      3,
      4,
      5,
      6
   ],
   "powerSource" : "Battery",
   "id" : 60,
   "stackVersion" : 2,
   "dateCode" : "20190730",
   "ieeeAddr" : "0x04cf8cdf02134657",
   "nwkAddr" : 53361,
   "endpoints" : {
      "4" : {
         "devId" : 259,
         "epId" : 4,
         "profId" : 260,
         "binds" : [],
         "clusters" : {},
         "outClusterList" : [
            6
         ],
         "inClusterList" : [
            18,
            3
         ]
      },
      "6" : {
         "epId" : 6,
         "devId" : 259,
         "profId" : 260,
         "outClusterList" : [
            6
         ],
         "binds" : [],
         "clusters" : {},
         "inClusterList" : [
            18,
            3
         ]
      },
      "3" : {
         "devId" : 259,
         "profId" : 260,
         "epId" : 3,
         "binds" : [],
         "inClusterList" : [
            18,
            3
         ],
         "clusters" : {},
         "outClusterList" : [
            6
         ]
      },
      "2" : {
         "epId" : 2,
         "devId" : 259,
         "profId" : 260,
         "clusters" : {},
[... 203 lines cut ...]
      "3" : {
         "binds" : [],
         "epId" : 3,
         "outClusterList" : [],
         "clusters" : {
            "genMultistateInput" : {
               "attributes" : {
                  "presentValue" : 1
               }
            }
         },
         "inClusterList" : []
      }
   },
   "manufName" : "LUMI",
   "ieeeAddr" : "0x04cf8cdf02134657",
   "manufId" : 4151,
   "id" : 8,
   "powerSource" : "Battery",
   "modelId" : "lumi.remote.b686opcn01"
}
```

 z2m commit #407d329 (cc2652), z2m commit #4d54e50 (cc2531), so same herdsman* 
manufId (xioami on cc2652, eurotronic on cc2531) is not the same and endpoints are not the same

lolorc, 2020-03-10:
shall I open a different issue ? (mine is a pairing issue)

lolorc, 2020-03-12:
I've deleted my comments and have created a separate issue for my issue : https://github.com/Koenkk/zigbee2mqtt/issues/3111

closed 2020-05-20 by stale[bot]