Koenkk/zigbee2mqtt #3076: Mega23M12 ballast joins successfully but not able to control it afterwards

ErnstEeldert, 2020-03-06:
# Bug Report
## What happened
After the device is successfully interviewed, any commands sent to it fail with: 'No network route (205))'.

## What did you expect to happen
Since it has joined the network successfully, I would expect it to be responsive to commands from the network.

## How to reproduce it (minimal and precise)
1. Make the device join the network (by doing a factory reset).
2. Confirm it joins successfully.
3. Turn the device off using an MQTT message.

## Debug Info
zigbee2mqtt version: 1.10.0
CC26X2R1 firmware version: zStack3x0 / 20191106

Koenkk, 2020-03-06:
Make sure to check with zigbee2mqtt 1.11.0, it contains some improvements related to `205 no network route` errors.

ErnstEeldert, 2020-03-06:
Doh. I was pretty sure I was running the latest, but didn't docker-compose pull apparently. I'm on 1.11.0 now, and although I still see some network flakiness, at least this particular device behaves normally now. I will see if I can spot some patterns on the other devices. Closing for now.

ErnstEeldert, 2020-03-07:
It turns out that after a while the device becomes unresponsive again with the 205 error. I've uploaded all errors filtered from the logs here:

https://pastebin.com/S23yGgzP

Please let me know what else I can do to get to the bottom of this.

Koenkk, 2020-03-08:
Before this happens I see `Error: Data request failed with error: 'MAC no ack' (233))'` in the log, this would indicate a weak connection. Can you share your networkmap?

ErnstEeldert, 2020-03-09:
It's rather big :). The device is shown at the bottom left part. Graphviz is here: https://pastebin.com/raw/wRNpQHqS

[image]

Koenkk, 2020-03-09:
The LQI looks ok, could you provide the sniff log when trying to control the device? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ErnstEeldert, 2020-03-10:
Here it is. I've filtered the capture to zbee_nwk.addr==0x22c9.

[0x22c9_filtered_network_dump.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/4311748/0x22c9_filtered_network_dump.pcapng.zip)

Koenkk, 2020-03-10:
You use the non-default network key, please send it to me on telegram (@koenkk).

ErnstEeldert, 2020-03-13:
I did. Let me know if you need anything else!

Koenkk, 2020-03-14:
- Could you also send the unfiltered log, not only the messages to `0x22c9` are interesting, but also route requests send from the coordinator.
- Did you try to control the device while sniffing this? (on or off command e.g.).

ErnstEeldert, 2020-03-16:
Here you go. I've tried to turn on the device 3 times during this capture, all resulting in the same error.

[failing_device_capture.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/4336449/failing_device_capture.pcapng.zip)

Koenkk, 2020-03-16:
In this sniff I don't see the coordinator sending an on or off command to the Mega23M12, so that indicates the commands are not transmitted by the coordinator somehow.

Could you try reflashing your CC2652R with https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200312.zip while carefully following https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html (step 5 is important here). Repairing already paired devices is not necessary.

ErnstEeldert, 2020-03-21:
Sorry it took a bit longer, but here's the log while running the updated firmware. Problem seems to still be there.

[ledstrip_capture_new_firmware.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/4362699/ledstrip_capture_new_firmware.pcapng.zip)

Koenkk, 2020-03-21:
I'm still puzzled what's actually happening, but I noticed one very strange thing. `0x0000` is the address of the coordinator, sometimes messages come from a texas instrument device (= cc2652r):

[image]

and sometimes messages come from a sillicon labs device

[image]

Could you share your `database.db` so we can see what device this is?

ErnstEeldert, 2020-03-24:
Here you go: https://pastebin.com/VLeG2Zjd

Looking at the mac address of that device, it seems to be an IKEA bulb?

```
{"id":27,"type":"Router","ieeeAddr":"0x000b57fffe2a3ea8","nwkAddr":2980,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 opal 1000lm","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 opal 1000lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caabb74","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001caabb74","endpointID":1}]}},"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1584774010449}
```

Koenkk, 2020-03-24:
Yes, can you disconnect this device from mains, pair the Mega23M12 again, and try to reproduce the issue (please sniff it again).

ErnstEeldert, 2020-04-13:
So, finally got around digging into this a little deeper. The ballast wasn't responding for over a week now, stuck in its last on state. As soon as I disconnected the mentioned IKEA device from mains, the ballast was controllable again.

So I kept the IKEA device turned off, and re-paired the ballast. Powered on the IKEA device, and things are still functioning as expected. I will update in a few days to tell if this is still the case.

ErnstEeldert, 2020-04-22:
Closing this as re-pairing with the dodgy (buggy, faulty?) IKEA bulb turned off seemed to have cleared things for the ballast. Intermittently I still get errors on this device, but that's the usual fare on my zigbee network.

closed 2020-04-22 by ErnstEeldert