Koenkk/zigbee2mqtt #3077: Error while opening serialport

DaX59, 2020-03-06:
Hi,

I am a new user with a rather low level under Debian.
My configuration:
RPI3 +
RFlink (for 433Mhz)
CC2531 (for zigbee)

When I enter the command "npm start", I have this message:
```
DaX@raspberrypi:/opt/zigbee2mqtt$ sudo npm start

> zigbee2mqtt@1.11.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-06 10:43:29: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-06.10-43-29' filename: log.txt
zigbee2mqtt:info  2020-03-06 10:43:30: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)
zigbee2mqtt:info  2020-03-06 10:43:30: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-06 10:43:30: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-06 10:43:30: Failed to start zigbee
zigbee2mqtt:error 2020-03-06 10:43:30: Exiting...
zigbee2mqtt:error 2020-03-06 10:43:30: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-06T09_43_30_962Z-debug.log
```

I search on differents website and with the error serialport, I checked my port with this command :

```
DaX@raspberrypi:/opt/zigbee2mqtt$ ls -l /dev/ttyACM0
crw-rw---- 1 DaX dialout 166, 0 mars   6 10:47 /dev/ttyACM0
DaX@raspberrypi:/opt/zigbee2mqtt$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 mars   6 10:13 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0
```
Maybe because I have 2 USB key (CC2531 and Rflink)

My ports seem to be OK : 

```
DaX@raspberrypi:/opt/zigbee2mqtt$ test -w /dev/ttyACM0 && echo success || echo failure
success
DaX@raspberrypi:/opt/zigbee2mqtt$ test -w /dev/ttyUSB0 && echo success || echo failure
success
```

My configuration : 

```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyUSB.CC2531-01
devices:
  '0x00158d0004102bbd':
    friendly_name: 'Capteur_Temperature'
  '0x00158d00044c0c9d':
    friendly_name: 'Switch'
  '0x00158d000429b98c':
    friendly_name: 'Capteur_Porte'
```
Thanks for your support.

filoor, 2020-03-06:
Can you reboot your pi with booth usb already in it and run:
ls -l /dev/serial/by-id

you will see 2 lines of booth usb sticks
The one with Texas instumens is the right on
you will see something like this on the end of the line "->../../ttyACM0"

Replace you config with
port: /dev/ttyACM0
"ttyACM0 is an example but it should be the last word after the /"

DaX59, 2020-03-06:
I rebooted my time lot of times and when I run the command :
```
DaX@raspberrypi:~$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 mars   6 11:05 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0
```
Always the same issue :( Juste one USB (The RFlink)
Thanks for your help.

filoor, 2020-03-06:
What do you see if you only boot it with the cc2531?
what kind of adapter do you have?

DaX59, 2020-03-06:
I started the pi only with RFlink and when I run the command : 
DaX@raspberrypi:~$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 mars   6 23:17 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0

After i restarted the pi only with CC2531 and I had : 
```
DaX@raspberrypi:~$ ls -l /dev/serial/by-id
ls: impossible d'accéder à '/dev/serial/by-id': Aucun fichier ou dossier de ce type
```
So if I translate French to English : cannot access '/dev/serial/by-id/': No such file or directory

DaX59, 2020-03-07:
I run differents commands and yesterday, Domoticz found my sensors. I restarted and now, connection lost :(
Others commands : 
```
DaX@raspberrypi:~$ ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 mars   7 11:25 /dev/ttyUSB0
lrwxrwxrwx 1 root root         7 mars   7 00:25 /dev/ttyUSB.CC2531-01 -> ttyACM0
DaX@raspberrypi:~$ ls -l /dev/ttyACM*
crw-rw---- 1 root dialout 166, 0 mars   7 00:25 /dev/ttyACM0
```

DaX59, 2020-03-08:
Lot of new tests and result : 
Debug npm start
```
zigbee2mqtt:info  2020-03-08 09:40:09: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-08.09-40-09' filename: log.txt
zigbee2mqtt:debug 2020-03-08 09:40:10: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-08.08-54-57'
zigbee2mqtt:debug 2020-03-08 09:40:10: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-03-08 09:40:10: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)
zigbee2mqtt:info  2020-03-08 09:40:10: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-08 09:40:10: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2531-01"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2531-01"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB.CC2531-01 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
zigbee2mqtt:error 2020-03-08 09:40:10: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-08 09:40:10: Failed to start zigbee
zigbee2mqtt:error 2020-03-08 09:40:10: Exiting...
zigbee2mqtt:error 2020-03-08 09:40:10: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```
with the sudo : 
```
DaX@raspberrypi:/opt/zigbee2mqtt$ DEBUG=zigbee-herdsman* sudo npm start
[sudo] Mot de passe de DaX : 

> zigbee2mqtt@1.11.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-08 09:42:30: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-08.09-42-30' filename: log.txt
zigbee2mqtt:debug 2020-03-08 09:42:30: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-08.08-55-01'
zigbee2mqtt:debug 2020-03-08 09:42:31: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-03-08 09:42:31: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)
zigbee2mqtt:info  2020-03-08 09:42:31: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-08 09:42:31: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2531-01"}}'
zigbee2mqtt:error 2020-03-08 09:42:31: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-08 09:42:31: Failed to start zigbee
zigbee2mqtt:error 2020-03-08 09:42:31: Exiting...
zigbee2mqtt:error 2020-03-08 09:42:31: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-08T08_42_31_714Z-debug.log
```

DaX59, 2020-03-21:
Hello,
Nobody has a solution for me ?
Thanks

Koenkk, 2020-03-21:
When unplugging your rflink, do you see anything under `ls -l /dev/serial/by-id` ?

If not, possible issue are:
- CC2531 does not have the firmware on it (https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html)
- It's a CC2540 and not a CC2531 (https://www.zigbee2mqtt.io/information/FAQ.html#in-case-of-a-cc2530-or-cc2531-adapter-verify-that-dont-have-a-cc2540)
- The CC2531 is broken.

DaX59, 2020-03-22:
Thanks @Koenkk 
Before the unplug : 
`DaX@raspberrypi:~$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 mars  19 21:33 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0
`
After the same :(
My CC2531 is correctly flash with the last version and it's a CC2531 not 2540.
The CC2531 is not broken.
The only solution I found is launch npm start with sudo
```
cd /opt/zigbee2mqtt
sudo npm start
```
but everytime I restart my Pi, I must restart all.

Koenkk, 2020-03-22:
If it works with sudo you need to fix your permissions. (https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port)

DaX59, 2020-03-22:
Everything is OK.
DaX@raspberrypi:/opt/zigbee2mqtt$ test -w /dev/ttyACM0 && echo success || echo failure
success
DaX@raspberrypi:/opt/zigbee2mqtt$ test -w /dev/ttyUSB0 && echo success || echo failure
success

closed 2020-05-30 by stale[bot]

dweng0, 2021-12-13:
I have a similar issue to this, some hardware (cc2531 correctly flashed and working fine up until this morning), will report what the fix is

robbz23, 2022-10-12:
> test -w /dev/ttyUSB0 && echo success || echo failure

What was the issue?

hacor, 2022-10-20:
Maybe somewhat late to reply, but it might be handy to others. I think the problem has to do with the `naming` you try to do through the udev rule file. It seems (not sure yet) that docker doesn't follow symlinks for devices. So you attach an 'empty' device to the container.
In my case:

- udev rule:
```
ATTR{idVendor}=="0451", ATTR{idProduct}=="16a8", SYMLINK+="usb-zigbee" MODE="0666"
```

When I attach device `dev/usb-zigbee` to the docker container, I get error:

```
Zigbee2MQTT:info  2022-10-20 09:56:50: Logging to console and directory: '/app/data/log/2022-10-20.09-56-50' filename: log.txt
Zigbee2MQTT:info  2022-10-20 09:56:50: Starting Zigbee2MQTT version 1.25.2 (commit #3c5854fa)
Zigbee2MQTT:info  2022-10-20 09:56:50: Starting zigbee-herdsman (0.14.34)
Zigbee2MQTT:error 2022-10-20 09:56:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-20 09:56:51: Failed to start zigbee
Zigbee2MQTT:error 2022-10-20 09:56:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-20 09:56:51: Exiting...
Zigbee2MQTT:error 2022-10-20 09:56:51: Error: Error while opening serialport 'Error: Error: Not a tty setting custom baud rate of 115200'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
```

When I attach the same USB stick directly to the container using the `dev/ttyACM0` name, it works perfectly.

Hope this helps someone!

Hacor